DIGImend/digimend-kernel-drivers #303: Huion H640P Inspiroy not working on Ubuntu 19.04 (kernel 5.0)

betocostadev, 2019-08-15:
Hello everyone!

I'm trying to run the H640P using Digimend V9 but I've had no success. I also tried running Digimend V10, but still isn't working.
I'm running Kubuntu 19.04.

**Version**
Linux ddingo 5.0.0-25-generic #26-Ubuntu SMP Thu Aug 1 12:04:58 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

**USB**
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0a2a Intel Corp. 
Bus 001 Device 003: ID 13d3:5a07 IMC Networks 
Bus 001 Device 006: ID 256c:006d  
Bus 001 Device 002: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The tablet is the 256c:006d.
After installing digimend drivers, it still doesn't recognize the tablet. It doesn't appear on Gimp and Inkscape.
Is there a patch I can try to apply to see if it works?
Thanks

abmsohel, 2019-09-20:
I have a very similar problem. My system config is as follows. 

Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic

The tablet is the 256c:006d. 

Any help will be really helpful

codesnippy, 2019-09-25:
I have the same problem on Debian buster:
Linux version 4.19.0-6-amd64 (debian-kernel@lists.debian.org) (gcc version 8.3.0 (Debian 8.3.0-6)) #1 SMP Debian 4.19.67-2+deb10u1 (2019-09-20)

 Tablet ID: 256c:006d
 Tablet Product ID: Huion H640P

I thought this model was supported by Digimend V8.
But it seems Huion has changed the Device ID

Any help in how to get this working would be appreciated.

DasLebenIstSeltsam, 2019-11-30:
Huion H640P, Ubuntu 18.04, DIGImend v9 from deb package

I followed **Enabling Wacom X.org driver** section from **READ.ME** and that worked for me

abmsohel, 2019-12-20:
> Huion H640P, Ubuntu 18.04, DIGImend v9 from deb package
> 
> I followed **Enabling Wacom X.org driver** section from **READ.ME** and that worked for me

Hi there, I followed your suggestion but could not make it work. Probably because I am not a programmer, I am not understanding most of the steps. Would you be kind enough to discuss this with me over a skype call? I really need to use this tablet (its lying idle for 3 months and I cannot get a replacement in my country).

DasLebenIstSeltsam, 2019-12-21:
> Hi there, I followed your suggestion but could not make it work. Probably because I am not a programmer, I am not understanding most of the steps. Would you be kind enough to discuss this with me [...]

Well... My respond was a little bit too fast. It kinda worked. I could use pen as pointer but drawing area on tablet was messed up and i couldn't get past that. Probably need some tinkering. Long story short, you need to create text file and put in it few lines of configuration. So i guess you can handle ;).

I did following:
open terminal -> 'ctrl+alt+t'
'cd /etc/X11/xorg.conf.d'
'ls' to check if there is 50-tablet.conf file, if not this will create it 'sudo touch 50-tablet.conf'
'lsusb' to check <VID>:<PID> of your tablet
sudo vim 50-tablet.conf
press insert on keyboard and paste this with your variables

Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "<VID>:<PID>"
EndSection

press escape and type ':wq!' (write and quit)
than restart your pc and you should be one step closer to making it work
-----------------------------------------------------------------------------------
Also i found this config for H610 model but maybe it will work for H640 as well (or some mix of these two configuration). Basically you do the same thing but file name is 52 instead of 50 and content is a little bit different.

https://github.com/DIGImend/digimend-kernel-drivers/issues/26#issuecomment-192761796

abmsohel, 2019-12-26:
Thanks man. I actually found another solution in youtube, and it works :) 

The link is posted below.

https://www.youtube.com/watch?v=1jHEGahVIoc