DIGImend/digimend-kernel-drivers #305: Gaomon S56K regression

jaylinee, 2019-08-16:
My S56K stopped working after commit cb4dfa4382603b03eb3c4e26b7e3b8f049e9a425 got merged into the mainline kernel (at torvalds/linux@2c3a88c64cb62cc59010359dbfc97f734e98d683).

After messing around a bit I found out that the driver finds the v2 pen parameters and successfully creates the v2 buttonpad parameters, however they don't seem to work correctly with my tablet.

Forcing the driver to probe the v1 pen parameters solves my issue. My firmware version is `HUION_T156_160726`

giovEra, 2020-05-29:
@jaylinee 
How did you force the driver  to probe the v1 pen parameters?
(I have the same problem)
Thanks

Weznon, 2020-05-30:
@giovEra, I put up my "fix" for this at https://github.com/Weznon/digimend-kernel-drivers.

I'm not sure how to fix the issue properly, but I just deleted the section of code related to probing for the v2 pen parameters and it works for me, kernel version `5.6.15-arch1-1`. Hope this helps.

israel24961, 2020-12-05:
I just wanted to say that @Weznon found the solution on his  https://github.com/Weznon/digimend-kernel-drivers
Thanks a lot.
I want to point out something that changes

In the Fixed version:
```
sudo usbhid-dump -es -m 256c:006e

001:007:000:STREAM            1607195151.244136
 07 80 04 0B FE 32 00 00

001:007:000:STREAM             1607195151.246407
 07 80 04 0B FA 32 00 00
```
In the master Master branch, see the first  3 bytes are different. (The fourth changes with the buttons)
**_0A C_*** ** ** ...
```
sudo usbhid-dump -es -m 256c:006e

001:007:000:STREAM            1607195151.244136
 0A C0 04 0B FE 32 00 00

001:007:000:STREAM             1607195151.246407
 0A C0 04 0B FA 32 00 00
```

Might some developer answer why?
Thanks for your time.

scrouthtv, 2020-12-23:
Can reproduce this with my Gaomon S56K.
`v10` branch on ec5c16d as well as release `v10` do not work.
Weznon's fork works without problems for me.

scrouthtv, 2021-02-10:
I am using Weznon's fork and my tablet stopped working some time ago. I never really cared about it until I fixed it today. For anyone else having this issue, my problem was that I installed the module manually using `make install`. This has to be redone after each kernel update.

Alternatively, use dkms: cd into the tree, run `make clean` and then `sudo dkms_install`, and the module should persist kernel updates.

z0tedd, 2022-11-02:
> @giovEra, I put up my "fix" for this at https://github.com/Weznon/digimend-kernel-drivers.
> 
> I'm not sure how to fix the issue properly, but I just deleted the section of code related to probing for the v2 pen parameters and it works for me, kernel version `5.6.15-arch1-1`. Hope this helps.

Thank you very much, I have tried a lot of solutions, but only yours helped.