DIGImend/digimend-kernel-drivers #308: Huion H950P has a new product ID

sornen, 2019-08-25:
Recent purchase of Huion H950P did not work. The product ID has changed and the device was being assigned to hid-generic.

The device ID 256c:006d whereas previously it was ID 256c:006e.

After changing this in hids-ids.h in the driver source
`#define USB_DEVICE_ID_HUION_TABLET	0x006d`
and recompiling
```
make clean
make
sudo make install
```
and rebooting the PC, it works.

mauriciomarinho, 2019-09-18:
@sornen Thank you. This fixed the issue after updating firmware on my H1060P. Should be noted that changing 0x006e to 0x006d can cause an error because the key already exists. I just swap them, not sure if it is the best way of doing it.

Lvaskz, 2019-12-21:
> The device ID 256c:006d whereas previously it was ID 256c:006e.

Hi I have the same problem, How can I fix it, I have no idea. Some help ?

Lvaskz, 2020-01-03:
May be, can some one make some appimage package, flatpak or snap for divers and settings?

kedodrill, 2020-02-13:
Is this still an issue?
I saw that this commit was pushed: https://github.com/DIGImend/digimend-kernel-drivers/commit/89e4d3fe977119fa30ca72a23d51a89e61d4a798#diff-c67b231c379204ac6146dea2ef7517d5

Seems like it added another variable there for 0x006d...does that work fine now?

kedodrill, 2020-02-14:
Just an update, got my H950P in today, right off the bat without digimend things worked including levels of pressure, but the buttons did not. I installed digimend by installing from source (not using dkms) and pulling directly from master and have had no issues.

Lvaskz, 2020-02-14:
> Is this still an issue?
> I saw that this commit was pushed: [89e4d3f#diff-c67b231c379204ac6146dea2ef7517d5](https://github.com/DIGImend/digimend-kernel-drivers/commit/89e4d3fe977119fa30ca72a23d51a89e61d4a798#diff-c67b231c379204ac6146dea2ef7517d5)
> 
> Seems like it added another variable there for 0x006d...does that work fine now?

it seems now is working.
  

 

> but the buttons did not.

It is a shame, but some one is try to solved, the problem is has a long long time, and we don't know if will solved.

ajboni, 2020-03-14:
> Just an update, got my H950P in today, right off the bat without digimend things worked including levels of pressure, but the buttons did not. I installed digimend by installing from source (not using dkms) and pulling directly from master and have had no issues.

Hey thanks for the report! How was your experience so far? Im planning to buy a tablet but want to make sure it works on Linux, is 950 or the intuos S. 

Any problemas so far?

Thanks!

kedodrill, 2020-03-16:
@ajboni Everything works really well as far as I can tell. Again, even without the digimend driver everything seemed to work fine except for the buttons. Once I got digimend working the buttons started working as well. I haven't messed around with button mapping, but I assume that's possible somehow.

ajboni, 2020-03-25:
I can confirm I had to do the same procedure as kedodrill. 

Tried the .deb at first but it would not work. 
After cloning the repo and building from source it works flawlessly.

jrw77, 2020-08-03:
I am having a difficulty with the new version (006d id): the x-values range from 0 to 65535, but apparently my setup only accepts values up to 32767 (2^15-^). So I can only use the left half of the tablet or only access the left half of the screen, depending on the setting I can control with xsetwacom. Does anyone have this problem? What is your result for 
`xsetwacom --get 9 Area`

ajboni, 2020-08-03:
> I am having a difficulty with the new version (006d id): the x-values range from 0 to 65535, but apparently my setup only accepts values up to 32767 (2^15-1). So I can only use the left half of the tablet or only access the left half of the screen, depending on the setting I can control with xsetwacom. Does anyone have this problem? What is your result for
> `xsetwacom --get 9 Area`

xsetwacom --get 13 Area                                                                                                                                                       12:46:55
0 0 44200 27600

jlj77, 2020-08-04:
Yeah, me too:
`> xsetwacom --get 10 area`
`0 0 44200 27600`

spbnick, 2020-08-04:
@ajboni, @jlj77, please try using the latest development version of the driver. Your symptoms show the kernel driver is not handling the tablet. The latest version can help.

ajboni, 2020-08-04:
> @ajboni, @jlj77, please try using the latest development version of the driver. Your symptoms show the kernel driver is not handling the tablet. The latest version can help.

I'm not having any issues! Just showed that info for @jrw77 . It's working perfectly on my end. 
Will upgrade the driver anyway and report back tonight though.

arichiardi, 2020-08-29:
I received the Huion 610p v2 (256c:006d), compiled the kernel driver and I _can confirm_ the tablet works. I haven't tried soft and hard button, I wonder how do I test?

joergfoertsch, 2020-09-01:
> I am having a difficulty with the new version (006d id): the x-values range from 0 to 65535, but apparently my setup only accepts values up to 32767 (2^15-1). So I can only use the left half of the [...]

I do have the same issue on an Debian system with kernel 4.9.0-13-amd64 with the :006d HUION H950P
The standard setting for Area is 0 0 32767 32767 which results in only half the tablet being mapped to my full screen.
If I type ... set Area 0 0 65535 32767 the following get returns "0 0 65535 32767". However still only half of the tablet is usable, as now half of my tablet are correctly mapped to half of my screen but I can not move my curser into the other screen-half using the tablet.
Maybe this should go into a new bug report, I just stumbled across this comment through google.

jrw77, 2020-12-26:
Don´t forget to put the Huion 950p Tablet into PC mode!
Button 1 and 5 simultaneously put the tablet into PC mode. In smartphone mode, only the left half is usable.

phispi, 2025-12-27:
I also had the problem that only the left halve of the drawing area was used. I did a firmware update of the Huion H950P (from version 200718 to 240919) and afterwards, the full area was usable. (I assume with the old firmware, the tablet automatically switched to Android mode with newer kernels.) The 8 buttons left to the drawing area are still not doing anything though.

phispi, 2025-12-30:
My bad: The buttons *do* work with the Huion H690p. In Debian, beside the `digimend-dkms` package, I needed to install the `linux-headers-amd64` (which somehow got uninstalled during the update without me noticing).

```
> xsetwacom --list
HUION Huion Tablet_H950P Touch Strip pad        id: 17  type: PAD       
HUION Huion Tablet_H950P Dial pad       id: 18  type: PAD       
HUION Huion Tablet_H950P Pen stylus     id: 19  type: STYLUS    
HUION Huion Tablet_H950P Pad pad        id: 20  type: PAD       
HUION Huion Tablet_H950P stylus         id: 21  type: STYLUS    
HUION Huion Tablet_H950P eraser         id: 22  type: ERASER    
HUION Huion Tablet_H950P Group Buttons pad      id: 23  type: PAD       
```

My kernel is 6.12.57 and the `digimend-kms` version is 13-4.

To make it short Huion H690p with the newest firmware is fully supported. I guess this ticket can be closed.