DIGImend/digimend-kernel-drivers #310: Tilt on Kamvas 22 Pro

FraYoshi, 2019-08-31:
Hi,
Has been a while since the tilt implementation but my tablet still don't see it..
I upgraded it via windows, it is the pre-tilt version) and can confirm it works because detected running `# usbhid-dump -es -m 256c:006e` but when testing `xinput test 14` it only report x,y and pressure..

OS info: 5.2.11.arch1-1

FraYoshi, 2019-11-25:
Okay,
I tested with the git version of the Digimend kernel drivers and tilt is responding!

[3 and 4]
Still, doesn't work with the Linux kernel's version...

So this issue is more related to that.
Feel free to close it if no longer relevant.

crocket, 2020-03-11:
Is it the 2019 version of kamvas pro 22? How well does it work?