DIGImend/digimend-kernel-drivers #312: button click not working on pen stylus on Huion WH1409?

biscuitlad, 2019-09-23:
Hi

Many many thanks for the wonderful work with the drivers. I've got the WH1409 working pretty well with xsetwacom mapping the various buttons. But there is one bug I can't fix.

Right click with the mouse in Krita brings up the colour selector wheel. It stays on the screen while you select your colour / brush / etc.

No matter what I do, when I map the pen stylus upper button to button 3, clicking it brings up the colour wheel but releasing the button means the colour wheel disappears before you can select your colour!

It's very odd and I don't know how to go about fixing it. Tried lots of variations with xsetwacom, but nothing is able to keep the colour wheel on the screen after clicking the stylus button. However, there is something glitchy about it, as sometimes when you move the stylus and click at the same time you can get it to remain on your screen. But it's nearly random, and I can't work out the steps to do it consistently.

Is this a bug, or something unique to the pen stylus of WH1409? 

I'm using the lines given by David Revoy in his blog post example (who doesn't mention this issue):

xsetwacom --set "$tabletstylus" Button 2 2 
xsetwacom --set "$tabletstylus" Button 3 3

Mapping works fine, it's the click behaviour that's not reproducing a genuine right mouse click. Or perhaps it's a right mouse click followed by something else? Something must be forcing the colour wheel to unload again on releasing the button.

DarkAngelCaptain, 2019-09-26:
The Stylus behaviour can be fixed by going into Krita configuration and going to tablet settings and clicking the tick box for Use Mouse events for right and middle clicks (Workaround for convertible devices needs restart).

biscuitlad, 2019-09-30:
That's great - many thanks for this. I'd missed that trick.  :-)

DarkAngelCaptain, 2019-09-30:
No problem. :)