DIGImend/digimend-kernel-drivers #314: Huion H1161 Support

DarkAngelCaptain, 2019-09-25:
The tablet was released this year so digimend doesn't support it yet. Here is the information I got from probing the tablet.

[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3652172/probe.txt)

https://www.huiontablet.com/all-products/graphic-tablets/huion-h1161.html

DarkAngelCaptain, 2019-09-25:
Here is the pen data and buttons etc.
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3653346/descriptors.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3653335/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3653337/pen_tilt.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3653340/pen_buttons.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3653343/frame_buttons.txt)
[soft_buttons_and_scroll.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3653344/soft_buttons_and_scroll.txt)

DarkAngelCaptain, 2019-09-26:
Installed master driver since version 9 doesn't work. The driver from the current master code lets me draw but the buttons do not respond on the tablet. The buttons on the pen do work.

DarkAngelCaptain, 2019-09-26:
soft scroll on the left side of the tablet is acting like I have pressed the second from bottom key on the tablet. So if I set the key to Save As the scroll does the same thing when I move my finger near the top of the scroll area.

opl-, 2019-11-01:
The USB device ID is `256c:006d` which matches the ID added to support the HS64 tablet in 66605efc.

According to `xinput test`, the tablet pressure, tilt, and hard buttons work using that commit. Alternatively it seemed that the v9 release also works after [mapping the device ID to the wacom driver](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/README.md#enabling-wacom-xorg-driver), though I haven't tested it too extensively with it.

The buttons were mapped using the Wacom tablet finder utility from `kde-config-tablet` and the generated config is included below. 

What **doesn't work** is the touch bar. As noted by DarkAngelCaptain, the driver behaves as if some sections of the touch bar were one of the hardware buttons while others get ignored. Dump included below.

The soft buttons also don't seem to work, though I assume that this is a feature handled by the software.

<details>
<summary>Touch bar traffic dump</summary>

```
drag top to bottom start
08 f0 01 01 00 01 00 00 00 00 1b e8
08 f0 01 01 00 02 00 00 00 00 1b e8
08 f0 01 01 00 03 00 00 00 00 1b e8
08 f0 01 01 00 04 00 00 00 00 1b e8
08 f0 01 01 00 05 00 00 00 00 1b e8
08 f0 01 01 00 06 00 00 00 00 1b e8
08 f0 01 01 00 07 00 00 00 00 1b e8
08 f0 01 01 00 08 00 00 00 00 1b e8
08 f0 01 01 00 00 00 00 00 00 1b e8

drag bottom to top start
08 f0 01 01 00 08 00 00 00 00 1b e8
08 f0 01 01 00 07 00 00 00 00 1b e8
08 f0 01 01 00 06 00 00 00 00 1b e8
08 f0 01 01 00 05 00 00 00 00 1b e8
08 f0 01 01 00 04 00 00 00 00 1b e8
08 f0 01 01 00 03 00 00 00 00 1b e8
08 f0 01 01 00 02 00 00 00 00 1b e8
08 f0 01 01 00 01 00 00 00 00 1b e8
08 f0 01 01 00 00 00 00 00 00 1b e8

```

</details>

<details>
<summary>
KDE tablet information (I added it to <code>/usr/share/wacomtablet/data/huion_devicelist</code>, but the utility output it to <code>~/.config/tabletdblocalrc</code>)</summary>

```
[006D]
model=H1161
name=Huion Inspiroy H1161
layout=unknown
hwbutton1=1
hwbutton2=2
hwbutton3=3
hwbutton4=8
hwbutton5=9
hwbutton6=10
hwbutton7=11
hwbutton8=12
hwbutton9=13
hwbutton10=14
padbuttons=10
statusleds=0
touchring=no
touchstripl=yes
touchstripr=no
wheel=no
```

</details>

JShorthouse, 2019-11-20:
I have also just received this tablet so adding my +1.

It was extremely simple to get this tablet working by building from master and then adding an xorg entry, so huge props for all the work you've done.

In addition to what @opl- has posted here is the output of `input test` for the touch bar. For some reason it is generating both motion and button press events, which confirms the behaviour @DarkAngelCaptain was seeing. I also tried just tapping on the scrollbar but this generates both motion and touch events too.

<details><summary>xinput test output</summary>

```
# Sliding from top to bottom over 2s
button press   13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0 
button press   14 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button press   13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0 
button release 14 a[3]=0 a[4]=0 a[5]=0 
button press   15 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button press   13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0 
button press   14 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button press   13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0 
button release 14 a[3]=0 a[4]=0 a[5]=0 
button release 15 a[3]=0 a[4]=0 a[5]=0 
button press   16 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 16 a[3]=0 a[4]=0 a[5]=0

# Sliding from bottom to top over 2s
button press   16 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button press   13 a[3]=0 a[4]=0 a[5]=0 
button press   14 a[3]=0 a[4]=0 a[5]=0 
button press   15 a[3]=0 a[4]=0 a[5]=0 
button release 16 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button press   13 a[3]=0 a[4]=0 a[5]=0 
button release 14 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button press   13 a[3]=0 a[4]=0 a[5]=0 
button press   14 a[3]=0 a[4]=0 a[5]=0 
button release 15 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button press   13 a[3]=0 a[4]=0 a[5]=0 
button release 14 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=

# Tapping the bottom of the scrollbar 3 times
button press   16 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 16 a[3]=0 a[4]=0 a[5]=0 
button press   16 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 16 a[3]=0 a[4]=0 a[5]=0 
button press   16 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 16 a[3]=0 a[4]=0 a[5]=0

# Tapping the top of the scrollbar 3 times
button press   13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0 
button press   14 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 14 a[3]=0 a[4]=0 a[5]=0 
button press   13 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 13 a[3]=0 a[4]=0 a[5]=0
```

</details>

Carles567, 2019-11-25:
> I have also just received this tablet so adding my +1.
> 
> It was extremely simple to get this tablet working by building from master and then adding an xorg entry, so huge props for all the work you've done.
> 
> ```
Hi, could you be more specific about what you did? I just received this tablet and I have never used any wacom nor huion tablet before. What exactly did you do to get it to work on linux? What di you mean you builit it from master? How can I do that myself? what did you add to xorg and how? Does that mean it will not work on Wayland?

JShorthouse, 2019-11-25:
@Carles567 Clone this git repo, `cd` into the cloned directory and then follow the instructions under "Installing from source with DKMS" in the readme.
I don't use wayland so I can't help you there, but doing those steps should allow your tablet to work at a basic level. The xorg stuff is only needed for `xsetwacom` stuff (i.e. mapping tablet buttons, mapping the tablet to a single monitor).
The xorg config file I added was following the steps under "Enabling Wacom X.org driver".

Carles567, 2019-11-25:
@JShorthouse Thanks for your help! Much appreaciated!

m-majetic, 2020-03-31:
I just got my "HS1161" tablet and it works like a charm with the latest master. Thank you all !

It _looks_ like the new "HS611" tablet uses a similar slider to the "HS1161" one, although I can't confirm it.
It would be beneficial to get it working for future tablets as well if this becomes a pattern.

So, one more thumbs up on this issue !

Let's get this piece of tech working completely.

spbnick, 2020-04-05:
Hi everyone, the latest master is likely to support this tablet as well as possible so far. Please try and report any remaining issues you see. Please remember to set "Suppress" to 0 using xsetwacom for the Touch Ring device (touch strip in your case) to get a decent speed out of it (this is to be documented in README.md as mentioned in #370).

spbnick, 2020-04-05:
Regarding soft-buttons, they are a marketing gimmick and are not worth supporting on this, or any other tablet. You have to look at the tablet to use them, and it is more efficient to simply use a panel with buttons on the screen - at least you don't have to take your eyes off work. Plus you get more useful area to draw on without them.

JShorthouse, 2020-04-07:
Thank you very much @spbnick, the touchbar is working well now.
I personally found that even with Supress set to 0 the speed was still too slow so I ended up setting each scroll event to trigger two scrolls which gives a comfortable speed for web browsing. For anyone else who wants to try.

```
xsetwacom set "HUION Huion Tablet Touch Ring pad" "Suppress" "0"  
xsetwacom set "HUION Huion Tablet Touch Ring pad" "AbsWheelDown" "button 4 4"
xsetwacom set "HUION Huion Tablet Touch Ring pad" "AbsWheelUp" "button 5 5"  
```

Also there seems to be a bug where two stylus devices are appearing in xsetwacom. I have tried rebooting and replugging the tablet but they always both appear.

```
~ 
→ xsetwacom --list         
HUION Huion Tablet Dial pad     	id: 12	type: PAD       
HUION Huion Tablet Touch Ring pad	id: 13	type: PAD       
HUION Huion Tablet Pad pad      	id: 14	type: PAD       
HUION Huion Tablet Pen stylus   	id: 15	type: STYLUS    
HUION Huion Tablet Pen stylus   	id: 23	type: STYLUS    
HUION Huion Tablet Pen eraser   	id: 24	type: ERASER    

~ 
→ xsetwacom -s --get 15 all
xsetwacom set "15" "Area" "0 0 55880 31433"
xsetwacom set "15" "Button" "1" "button +1 "
xsetwacom set "15" "Button" "2" "button +2 "
xsetwacom set "15" "Button" "3" "button +3 "
xsetwacom set "15" "ToolDebugLevel" "0"
xsetwacom set "15" "TabletDebugLevel" "0"
xsetwacom set "15" "Suppress" "2"
xsetwacom set "15" "RawSample" "4"
xsetwacom set "15" "PressureCurve" "0 0 100 100"
xsetwacom set "15" "Mode" "Absolute"
xsetwacom set "15" "TabletPCButton" "off"
xsetwacom set "15" "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
xsetwacom set "15" "Gesture" "off"
xsetwacom set "15" "ZoomDistance" "0"
xsetwacom set "15" "ScrollDistance" "0"
xsetwacom set "15" "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
xsetwacom set "15" "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
xsetwacom set "15" "Threshold" "26"
xsetwacom set "15" "BindToSerial" "0"
xsetwacom set "15" "PressureRecalibration" "on"

~ 
→ xsetwacom -s --get 23 all
xsetwacom set "23" "Area" "0 0 55880 31433"
xsetwacom set "23" "Button" "1" "button +1 "
xsetwacom set "23" "Button" "2" "button +2 "
xsetwacom set "23" "Button" "3" "button +3 "
xsetwacom set "23" "ToolDebugLevel" "0"
xsetwacom set "23" "TabletDebugLevel" "0"
xsetwacom set "23" "Suppress" "2"
xsetwacom set "23" "RawSample" "4"
xsetwacom set "23" "PressureCurve" "0 0 100 100"
xsetwacom set "23" "Mode" "Absolute"
xsetwacom set "23" "TabletPCButton" "off"
xsetwacom set "23" "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
xsetwacom set "23" "Gesture" "off"
xsetwacom set "23" "ZoomDistance" "0"
xsetwacom set "23" "ScrollDistance" "0"
xsetwacom set "23" "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
xsetwacom set "23" "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
xsetwacom set "23" "Threshold" "26"
xsetwacom set "23" "BindToSerial" "0"
xsetwacom set "23" "PressureRecalibration" "on"
```

One of these is a "dummy" stylus, setting the parameters on it have no effect.
Not a big deal but it breaks my xsetwacom script that relies on using the unique device names.

m-majetic, 2020-04-08:
@JShorthouse This is what I get when I input "xsetwacom --list" 

```
HUION Huion Tablet Pen stylus   	id: 9	type: STYLUS    
HUION Huion Tablet Pad pad      	id: 10	type: PAD       
HUION Huion Tablet Touch Ring pad	id: 11	type: PAD       
HUION Huion Tablet Dial pad     	id: 12	type: PAD 
```

What distro are you on ?
I'm using Debian 10 with i3wm.

spbnick, 2020-04-08:
@JShorthouse, thank you for the detailed report! I think I know where the dummy "stylus" is coming from and will need to think on how to handle that best, but it should be fixed. I'll try to take a look at it on the weekend.

@m-majetic, the difference in your `xsetwacom --list` and @JShorthouse likely comes down to the version of the Wacom drivers.

spbnick, 2020-04-13:
Alright, I've made some more improvements, and now the dummy "pen" device should be gone and the touch strip might work a little smoother. Please check the [latest master](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip) and report how it's working for you.

m-majetic, 2020-04-13:
The tablet still works great.
The touch strip is now in the correct orientation for me ( it was inverted before ) and it is smoother.

Thank you ! :smiley:

JShorthouse, 2020-04-13:
@spbnick Thanks, the dummy devices have gone now. Can't say I can notice much difference in the scrolling but it's nice that the default is now in the direction that my brain expects it to be (you'd think that being a "touch" strip my brain would want to use "natural scrolling" but for some reason my brain treats it like a scrollwheel, and it seems that m-majestic had the same experience).

spbnick, 2020-04-14:
Thank you for testing, @m-majetic and @JShorthouse! I'll poke a little more at the Wacom driver to see if I can have the touch strip recognized exactly as such and supported better, but so far I suspect it might not work.

spbnick, 2020-04-26:
NOTE: The latest development version configures the system to set "Suppress" to zero for the touch strip device automatically.

cionx, 2020-04-28:
I also have a Huion H1161, but the soft scroll doesn’t seem to work for me. I’m using the latest version (d36d2ca), installed on ArchLinux via the [Arch User Repository][1]. I have enabled the wacom drivers [as described in the repository][2]. Everything else seems to work and can be configured using `xsetwacom`. My installed version of xf86-input-wacom is 0.39.0.

I get the folllowing output from `xsetwacom --list`:

    $ xsetwacom --list      
    HUION Huion Tablet Pen stylus           id: 15  type: STYLUS    
    HUION Huion Tablet Touch Strip pad      id: 16  type: PAD       
    HUION Huion Tablet Pad pad              id: 17  type: PAD       
    HUION Huion Tablet Dial pad             id: 18  type: PAD 

This agrees with the output of `xinput list`. But when I try `xinput test 16` (or `xinput test` for any of these four devices) there is no reaction to using the touch strip. Setting "Suppress" to zero didn’t change anything. I have attached the various outputs of `xsetwacom --get` below.

However, the LED of the tablet lights up while I’m using the touch strip and I’m getting some output from `sudo usbhid-dump -es -m 256c:006d`. I have attached this output below. (I’m first going from the top to the bottom, and then from the bottom to the top.)

[xsetwacomget.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4542422/xsetwacomget.txt)
[soft_scroll.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4542273/soft_scroll.txt)

[1]: https://aur.archlinux.org/packages/digimend-kernel-drivers-dkms-git/
[2]: https://github.com/DIGImend/digimend-kernel-drivers#enabling-wacom-xorg-driver

spbnick, 2020-04-28:
@cionx, if you're using Gnome, see https://github.com/DIGImend/digimend-kernel-drivers/#touch-ringstrip-scrolling-doesnt-work-in-gnome

If not, please post the output of `xinput test` for the "Touch Strip pad" device.

"Suppress" should be set to zero automatically with the latest development version.

cionx, 2020-04-29:
I’m using KDE, not Gnome. (I tried to follow [the link][1] given on the front page to learn more about the problem with Gnome, but get a 404 page.) I also tried it under fluxbox to see if this makes a difference, but it doesn’t seem to.

The command `xinput test "HUION Huion Tablet Touch Strip pad"` doesn’t print any output when I use the touch scrollbar, move the pen, press the pen on the tablet, press the buttons of the pen, or press the buttons on the tablet. So there’s no output I could post here.

I fixed my problem with "Suppress" not being automatically set to zero: the ArchLinux package doesn’t install the file `xorg.conf`, aka `/usr/share/X11/xorg.d/50-digimend.conf`. After installing this file (e.g. via `sudo make xorg_conf_install`), "Suppress" is automatically set to zero for the device "Touch Strip pad".

I have also noticed the following things:

- If both `/usr/share/X11/xorg.d/50-digimend.conf` and `/etc/X11/xorg.conf.d/50-tablet.conf` are present then I get both the device "Touch Strip pad" and also "Dial pad". `xinput test` does not give any output for any of the two devices, as described above.
- If only `/usr/share/X11/xorg.d/50-digimend.conf` is present then the output of `xsetwacom --list` has the device "Touch Strip pad", but not the device "Dial pad". (So the output consists of only three devices.) But `xinput list` still shows "HUION Huion Tablet Dial". There is again no output from `xinput test` on any of these two devices.
- Suppose that I don’t install the drivers from this git-repository and instead use the one coming with my currently installed kernel (5.6.7), but still specify (e.g. via `/etc/X11/xorg.conf/50-tablet.conf`) that I want to use the wacom drivers. Then I get neither the "Touch Strip pad" device, nor "Dial pad". Both `xsetwacom --list` and `xinput list` show only the devices "Pen stylus" and "Pad pad". But if I now use `xinput test` on "Pad pad" then I do actually get some output for the touch scrollbar. It gives both motion and button press events, as described in https://github.com/DIGImend/digimend-kernel-drivers/issues/314#issuecomment-556274882.  
(Normally `xinput test` shows no output for "Pad pad" when the touch scrollbar is used. This is the first time I managed to get any kind of output from `xinput test` from using the touch scrollbar.)

[1]: https://github.com/DIGImend/digimend-kernel-drivers/blob/master/gnome_touch_scroll_issue

spbnick, 2020-04-29:
@cionx, that's a lot to figure out, but first I'd like you to make sure you're using the [latest development version of the drivers](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip), not another package. I'm not responsible for the ArchLinux packaging, and, unfortunately, don't have the time to debug any of its issues. Please also remove the `50-tablet.conf` file at least while we're troubleshooting this.

Once you did that, please post the output of `xinput list` and `xsetwacom list`, plus the contents of `/var/log/Xorg.0.log`, if you have it, or the same content extracted from the journal, if not. Please also post the output of `cat /sys/module/hid_uclogic/version`.

I fixed the Gnome issue link now. However, it's Gnome-specific, and KDE should work without a problem.

cionx, 2020-04-29:
I had some hiccups installing the drivers: After I did `sudo make dkms_install` and then rebooted, the wrong modules were loaded, even though the correct ones were available.
```
$ ls /usr/lib/modules/5.6.7-arch1-1/extra 
hid-kye.ko.xz  hid-polostar.ko.xz  hid-uclogic.ko.xz  hid-viewsonic.ko.xz

$ modinfo hid-uclogic | grep filename
filename:       /lib/modules/5.6.7-arch1-1/kernel/drivers/hid/hid-uclogic.ko.xz
```
 This problem didn’t occur when I used `sudo make` and `sudo make install` instead.
```
$ modinfo hid-uclogic | grep filename
filename:       /lib/modules/5.6.7-arch1-1/extra/hid-uclogic.ko.x
```
I guess the problem is that `sudo make dkms_install` doesn’t install the `depmod.conf` file. I managed to get the correct modules loaded by using any of the following two methods of installation:
- `sudo make dkms_install depmod_conf_install` and then `sudo depmod -a`
- `sudo make depmod_conf_install dkms_install`

I decided to go with the second option. I’ve attached the the installation output below in case that it helps. (The module `hid-polostar.ko.xz` isn’t being replaced because it is allegedly already contained in kernel with the correct version. I don’t know how to check if this makes sense.)

I’ve also attached `Xorg.0.log` below. It wasn’t located in `/var/log` but `~/.local/share/xorg`. I’ve checked that it’s the correct file by reconnecting the tablet, as can be seen at the end of the file.

The output of `xsetwacom list` is as follows:
```
$ xsetwacom list                                               
HUION Huion Tablet Pad pad              id: 15  type: PAD       
HUION Huion Tablet Touch Strip pad      id: 16  type: PAD       
HUION Huion Tablet Pen stylus           id: 23  type: STYLUS 
```
(The numbering changed from 15, 16, 17 to 23, 15, 16 when I reconnected the tablet.)
The output of `xinput list` ist attached below.
The content of `/sys/module/hid_uclogic/version` is as follows:
```
$ cat /sys/module/hid_uclogic/version
10
```

I’ve tried `xinput test` with the three devices shown by `xsetwacom list`, but they still behave as before (Pad and Pen behave as expected, and Touch Strip shows no reaction).

[installation.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4552025/installation.txt)
[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/4552034/Xorg.0.log)
[xinputlist.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4552076/xinputlist.txt)

spbnick, 2020-04-29:
Thank you, I'm yet to go through other issues, but DKMS behavior is troubling. Both on Debian and on Fedora it takes care of depmod configuration. Can you check what ArchLinux version is supposed to do?

spbnick, 2020-05-18:
I just merged #394, which adds depmod configuration installation/uninstallation for dkms targets. Shouldn't be a problem anymore. Closing. Reopen if you see any more issues.

closed 2020-05-18 by spbnick