DIGImend/digimend-kernel-drivers #316: Huion H430p

NateRafa, 2019-10-06:
hello I installed the driver and restarted the pc and the tablet didn't work

Distro: Deepin

plembo, 2019-11-05:
Basically same here. On Ubuntu Bionic (18.04.03 LTS) with the 4.15.0-66-generic kernel. Tried v9 driver dkms deb. After reboot xinput --list didn't show the driver, although dmesg said it was installed.

NateRafa, 2019-11-05:
> Basically same here. On Ubuntu Bionic (18.04.03 LTS) with the 4.15.0-66-generic kernel. Tried v9 driver dkms deb. After reboot xinput --list didn't show the driver, although lsmod does. Also tried with the 5.0.0-32-generic hwe kernel, but can't see the driver in lsmod after reboot. Maybe some ubuntu/debian default config weirdness? Above my competence level.

on ubuntu 19.10 and fedora 31 is working, I tested it here

plembo, 2019-11-05:
Found my problem turned out to be in my xorg config. The default 50-digimend.conf under /usr/share/X11/xorg.conf.d had the standard Huion usbid 256c:006e, but lsusb showed my tablet (with the mfr field blank) as 256c:006d. I created an additional 50-huion-430p.conf file using my device's usbid, restarted X and voila!

Madac1, 2019-12-18:
> Thanks Nate! That gave me to incentive to dig deeper. Found my problem was in the xorg config. The 50-digimend.conf installed by the package under /usr/share/X11/xorg.conf.d had the standard Huion usbid 256c:006e, but lsusb showed my tablet (with the mfr name blank) as 256c:006d. I created an additional 50-huion-430p.conf file using my device's usbid, restarted X and voila! This was on kernel 4.15.0-66-generic, but I'm thinking it will also work on 5.0.0 when I try it.

hey @plembo , could you show how to create the additional config file for the 256c:006d, i bought the same tablet and lsusb is showing the a "d" instead of a "c"

plembo, 2019-12-18:
Just use your favorite editor (mine is vim) _as sudo root_ to create /usr/share/X11/xorg.conf.d/50-huion-430p.conf:

```bash
$ sudo vi /usr/share/X11/xorg.conf.d/50-huion-430p.conf
```
Here's the finished product on my machine:

```bash
Section "InputClass"
    Identifier "Huion H430P Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "256c:006d"
EndSection
```
Note the "MatchUSBID" has the usb id of my actual hardware, found through lsusb:

```bash
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:5409 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 046d:c33a Logitech, Inc. 
Bus 001 Device 005: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 010: ID 256c:006d  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
You'll notice that the Huion does not display a label. Very inconsiderate of them. But given what we've all learned about their products, "256c:006d" is clearly the right one because "256c" is Huion's vendor id. If you run dmesg you'll find the same info, albeit not as legible:

```bash
$ dmesg
...
[253535.584690] usb 1-10: new full-speed USB device number 10 using xhci_hcd
[253535.734580] usb 1-10: New USB device found, idVendor=256c, idProduct=006d
[253535.734585] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[253535.734589] usb 1-10: Product: Huion Tablet
[253535.734592] usb 1-10: Manufacturer: HUION
[253535.738096] hid-generic 0003:256C:006D.000C: hiddev3,hidraw6: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:14.0-10/input0
...
```

NateRafa, 2020-01-06:
> Just use your favorite editor (mine is vim) _as sudo root_ to create /usr/share/X11/xorg.conf.d/50-huion-430p.conf:
> 
> ```shell
> $ sudo vi /usr/share/X11/xorg.conf.d/50-huion-430p.conf
> ```
> 
> Here's the finished product on my machine:
> 
> ```shell
> Section "InputClass"
>     Identifier "Huion H430P Tablet"
>     Driver "wacom"
>     MatchDevicePath "/dev/input/event*"
>     MatchUSBID "256c:006d"
> EndSection
> ```
> 
> Note the "MatchUSBID" has the usb id of my actual hardware, found through lsusb:
> 
> ```shell
> $ lsusb
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 001 Device 004: ID 0bda:5409 Realtek Semiconductor Corp. 
> Bus 001 Device 002: ID 046d:c33a Logitech, Inc. 
> Bus 001 Device 005: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
> Bus 001 Device 010: ID 256c:006d  
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> ```
> 
> You'll notice that the Huion does not display a label. Very inconsiderate of them. But given what we've all learned about their products, "256c:006d" is clearly the right one because "256c" is Huion's vendor id. If you run dmesg you'll find the same info:
> 
> ```shell
> $ dmesg
> ...
> [253535.584690] usb 1-10: new full-speed USB device number 10 using xhci_hcd
> [253535.734580] usb 1-10: New USB device found, idVendor=256c, idProduct=006d
> [253535.734585] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [253535.734589] usb 1-10: Product: Huion Tablet
> [253535.734592] usb 1-10: Manufacturer: HUION
> [253535.738096] hid-generic 0003:256C:006D.000C: hiddev3,hidraw6: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:14.0-10/input0
> ...
> ```

did not work