DIGImend/digimend-kernel-drivers #317: Huion 1409 Kernel 5.2.11-1-default

leoplaw, 2019-10-06:
My Huion 1409 was working out of the box with Kernel 4.20.6-1-default (OpenSuse Tumbleweed).
I recently upgraded the entire system and now have Kernel 5.2.11-1-default. The Huion 1409 table no longer works.

So, I pulled Digimend V9, compiled and installed, still no joy. I then pulled the latest update, compiled and installed, still broken.

I thoroughly went over all of the instructions.

It seems that the X.org config needs the identifier to be "PenTablet" and not "TABLET".

Section "InputClass"
    Identifier "InputTable"

with this correctly set, xinput reports the following pointers

PenTablet  Pen stylus                     id=9    [slave  pointer  (2)]
PenTablet  Pad pad                        id=10   [slave  pointer  (2)]
PenTablet  Consumer Control        id=12   [slave  pointer  (2)]

and Keyboards

PenTablet  Keyboard                       id=11   [slave  keyboard (3)]
PenTablet  System Control                 id=13   [slave  keyboard (3)]

However, the tablet still does not work.
xsetwacom --list
PenTablet  Pen stylus                   id: 9   type: STYLUS    
PenTablet  Pad pad                      id: 10  type: PAD   

lsusb
Bus 002 Device 003: ID 256c:006e  

I also followed the instructions to remove the kernel modules.
sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic

and then
dksm install digimend/9 --force

reboot, still nothing works.

After much experimenting I came to the conclusion the issue is with the module hid-uclogic and removed hid-uclogic with modprobe and then specifically installed the compiled V9 resulting in the following error.

insmod /lib/modules/5.2.11-1-default/extra/hid-uclogic.ko
insmod: ERROR: could not insert module /lib/modules/5.2.11-1-default/extra/hid-uclogic.ko: Unknown symbol in module

At this point I don't know what further to do, and have resorted to booting from Kernel 4.20.6-1-default again to make the Huion usable again.

I can provide whatever output you need to solve the issue.

_Originally posted by @leoplaw in https://github.com/DIGImend/digimend-kernel-drivers/issues/217#issuecomment-538751581_

leucome, 2019-10-19:
I had some issue that  may be related. 
It's with a WH1409 (8192) and Manjaro kernel 5.2.21.   

The tablet works  strait out with 5 buttons mapped to mouse buttons.   I was not able to find which files configure the button mapping so I tried to install digimend.  And whatever I do it seem to keep running with the default driver. So if the default driver was not working for some reason probably that   using digimend would not help as it seem to keep using the default driver anyway.  Or something like that.  

It's like digimend driver is completely ignored.  Even if the Wacom configuration tool think everything is working.

Do you have an Huion WH1409  (2048)   or the (8192).   Just in case, it turn out to be relevant.

leoplaw, 2019-10-19:
I have checked the device itself, I also dug out the manuals and I can't tell you any more than it is a Huion WH1409. What are the other numbers?

leucome, 2019-10-20:
I tried and get the opposite effect.  My tablet wont work on on 4.20 kernel and only work with the newer kernel. 

Model is indicated on the back sticker  Model WH1409(8192)    If the name is WH1409  it's probably the the original 2048.  The number is the pen level of pressure. 
There is 3 model. 
The original WH1409  2048 level of pressure. Pen  P68 battery or P80 rechargeable.
WH1409( 8192)  level of pressure.  Pen  PE150 reachable.
WH1409 V2  (8192) level of pressure with tilt. Pen  PW500 Battery-free.

The 3 tablet are visually identical but I'm pretty sure they send data differently to the driver.

leoplaw, 2019-10-20:
My WH1409 is then the 2048, as that is the spec I ordered at the time.

I'm going to try out a later Kernel, but keep 4.20.6-1-default just in case 5.3.6 does not work.

Narice, 2020-03-14:
I get absolutely no pointer mouvement when using the WH1409(original) tablet. I installed digimend and rebooted multiple times but nothing works
it is listed in the xsetwacom --list and when dumping data with `sudo usbhid-dump -es -m 256c:006e` data is actually sent when using the tablet, but doing a xinput test on the pad or stylus doesn't give anything.
Do you have anything new ?

leoplaw, 2020-03-15:
I can confirm the same, the tablet is listed, and also dumps data, beyond that, nothing else.

Narice, 2020-03-15:
I resolved the problem on my part, the tablet only works when in wireless and only in wireless. It does not work if you let it plugged in.
I hope it will work for you too!
(kernel 5.5.8)

leoplaw, 2020-03-15:
Interesting, I'll try that.

leoplaw, 2020-03-15:
@Narice, YES! I can confirm also that the WH1409 tablet works on wireless.
Thank you very much for the tip!

Narice, 2020-03-16:
@leoplaw really happy that I could help!
It was really annoying to discover that it worked really well from the beginning xDDD

For other scripts and technics that you might use, here is a great article that goes over the setup process for a Huion WH1409: https://www.davidrevoy.com/article331/setup-huion-giano-wh1409-tablet-on-linux-mint-18-1-ubuntu-16-04
All buttons can be accessed and used on this table btw.
I'm telling all this here in case another person has problems setup-ing their tablet.
Have a nice day and enjoy!

mikesbytes, 2020-05-16:
I can confirm that this issue is still present in kernel 5.6.13. I noticed that the output of `xinput --list` is different when the tablet is in wireless mode as opposed to wired.

I would like for wired support to work properly again as my tablet lives on my desk with my desktop and that wireless adapter gets really hot for some reason.

Narice, 2020-05-17:
I think the wireless adapter is designed for the temperature it goes up to (or I hope so x)) as it's mentioned on the back of it, so you can plug it on your desktop and forget about it.
But yeah, that would be really nice to have the wired mode working.
-------- Message d'origine --------

mikesbytes, 2021-08-23:
Has there been any progress on this issue? The battery in my wh1409 crapped out, so now it unusable as it will not connect to the wireless adapter when plugged in.

Narice, 2021-08-24:
@mikesbytes I don't think so.
@spbnick, was there any progress on this, or is it possible to resolve this issue?

hugolpz, 2023-03-10:
Just bough a Dell with a clean Ubuntu 20.04.5 LTS x86_64 .

Same on H640P. When plugged via usb, light turns on, when pen hovering then light gets brighter (this with or without driver installed).

```bash
# Listing usb returns no "tablet", no "tablet"
sudo lsusb
# Listing hardware returns no "tablet", no "tablet"
sudo lshw > ~/Documents/lshw.md
```

System info:
```bash
hugo@urfist-lap-0003:~$ neofetch --stdout
hugo@urfist-lap-0003 
-------------------- 
OS: Ubuntu 20.04.5 LTS x86_64 
Host: Precision 7670 
Kernel: 5.15.0-67-generic 
Uptime: 37 mins 
Packages: 1921 (dpkg), 20 (snap) 
Shell: bash 5.0.17 
Resolution: 1920x1200, 1920x1200 
DE: GNOME 
WM: Mutter 
WM Theme: Adwaita 
Theme: Yaru [GTK2/3] 
Icons: Yaru [GTK2/3] 
Terminal: gnome-terminal 
CPU: 12th Gen Intel i7-12850HX (24) @ 4.700GHz 
GPU: NVIDIA 01:00.0 NVIDIA Corporation Device 25ba 
GPU: Intel Device 4688 
Memory: 3614MiB / 15683MiB 
```