DIGImend/digimend-kernel-drivers #318: Huion 1060Plus not working on kernel 5.3

Akezhar, 2019-10-15:
The tablet works fine on kernel 5.2 but I can't seem to make it work on 5.3.
`lsusb` shows the tablet but `xinput list`, `xsetwacom list`, and `uclogic-probe` don't give any info about it.
If there's anything I can do to help with troubleshooting please let me know.

YuudaiNoboru, 2019-10-31:
I have the same problem with the Huion 1060 Plus Tablet, using Ubuntu 19.10 with the 5.3.0-19-generic Kernel.

earldbjr, 2019-11-05:
Same issue here, had to revert a machine to keep the tablet working.

alihakki, 2019-11-07:
i use fedora 31, i dont install any digimend packages, H1060p works well but lsusb dont sees name of tablet :(
Linux 5.3.7-301.fc31.x86_64
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 011 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 010 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 2109:2813 VIA Labs, Inc. USB2.0 Hub             
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 045e:0768 Microsoft Corp. Sidewinder X4
Bus 004 Device 002: ID 09da:f213 A4Tech Co., Ltd. USB Device
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 002: ID 256c:006e  
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

xinput --list                            
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad                         id=9    [slave  pointer  (2)]
⎜   ↳ COMPANY USB Device                        id=10   [slave  pointer  (2)]
⎜   ↳ Microsoft Microsoft® SiderWinderTM X4 Keyboard Consumer Control   id=12   [slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pen Pen (0)                 id=15   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ HID 256c:006e Pen                         id=8    [slave  keyboard (3)]
    ↳ Microsoft Microsoft® SiderWinderTM X4 Keyboard    id=11   [slave  keyboard (3)]
    ↳ Microsoft Microsoft® SiderWinderTM X4 Keyboard    id=13   [slave  keyboard (3)]
    ↳ Microsoft Microsoft® SiderWinderTM X4 Keyboard Consumer Control   id=14   [slave  keyboard (3)]

earldbjr, 2019-11-07:
> i use fedora 31, i dont install any digimend packages, H1060p works well but lsusb dont sees name of tablet :(
> Linux 5.3.7-301.fc31.x86_64
> Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root [...]

It appears to be id=9,15,8
And 
Bus 008 Device 002: ID 256c:006e
Vendor id 256c is huion.

Have you tried typing "sudo mod probe hid-uclogic" in your terminal?

You can also try running "evtest" and see if it lists and/or responds to input.

alihakki, 2019-11-07:
i didn't used any command or any settings. 
To make sure, I downloaded the Fedora 31 KDE spin. (Fedora-KDE-Live-x86_64-31-1.9.iso)
I could use the tablet when fedora-live-kde-spin started.

but i need help, how to configure those button. O_O

 [liveuser@localhost-live ~]$ lsusb
Bus 003 Device 002: ID 0781:5567 SanDisk Corp. Cruzer Blade
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 011 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 010 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 004: ID 256c:006e  
Bus 002 Device 003: ID 03f0:002a HP, Inc LaserJet P1102
Bus 002 Device 002: ID 2109:2813 VIA Labs, Inc. USB2.0 Hub             
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 045e:0768 Microsoft Corp. Sidewinder X4
Bus 004 Device 002: ID 09da:f213 A4Tech Co., Ltd. USB Device
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
 
 
 [liveuser@localhost-live ~]$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Power Button
/dev/input/event1:      Power Button
/dev/input/event2:      COMPANY USB Device
/dev/input/event3:      Microsoft Microsoft® SiderWinderTM X4 Keyboard
/dev/input/event4:      HID 256c:006e Pen
/dev/input/event5:      HID 256c:006e Pad
/dev/input/event6:      Microsoft Microsoft® SiderWinderTM X4 Keyboard Consumer Control
/dev/input/event7:      Microsoft Microsoft® SiderWinderTM X4 Keyboard
/dev/input/event8:      HDA ATI HDMI HDMI/DP,pcm=3
/dev/input/event9:      HDA ATI HDMI HDMI/DP,pcm=7
/dev/input/event10:     HDA ATI HDMI HDMI/DP,pcm=8
/dev/input/event11:     HDA ATI HDMI HDMI/DP,pcm=9
/dev/input/event12:     HDA ATI HDMI HDMI/DP,pcm=10
/dev/input/event13:     HDA ATI SB Front Mic
/dev/input/event14:     HDA ATI SB Rear Mic
/dev/input/event15:     HDA ATI SB Line
/dev/input/event16:     HDA ATI SB Line Out Front
/dev/input/event17:     HDA ATI SB Line Out Surround
/dev/input/event18:     HDA ATI SB Line Out CLFE
/dev/input/event19:     HDA ATI SB Line Out Side
/dev/input/event20:     HDA ATI SB Front Headphone
Select the device event number [0-20]: 5
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "HID 256c:006e Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 331 (BTN_STYLUS)
[... 112 lines cut ...]
Event: time 1573143235.710172, -------------- SYN_REPORT ------------
Event: time 1573143236.168102, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1573143236.168102, type 1 (EV_KEY), code 264 (BTN_8), value 1
Event: time 1573143236.168102, -------------- SYN_REPORT ------------
Event: time 1573143236.298082, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1573143236.298082, type 1 (EV_KEY), code 264 (BTN_8), value 0
Event: time 1573143236.298082, -------------- SYN_REPORT ------------
Event: time 1573143236.788004, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1573143236.788004, type 1 (EV_KEY), code 265 (BTN_9), value 1
Event: time 1573143236.788004, -------------- SYN_REPORT ------------
Event: time 1573143236.951980, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1573143236.951980, type 1 (EV_KEY), code 265 (BTN_9), value 0
Event: time 1573143236.951980, -------------- SYN_REPORT ------------
Event: time 1573143237.505896, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1573143237.505896, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1573143237.505896, -------------- SYN_REPORT ------------
Event: time 1573143237.669872, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1573143237.669872, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1573143237.669872, -------------- SYN_REPORT ------------
Event: time 1573143238.125802, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1573143238.125802, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
Event: time 1573143238.125802, -------------- SYN_REPORT ------------
Event: time 1573143238.355767, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1573143238.355767, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
Event: time 1573143238.355767, -------------- SYN_REPORT ------------

alihakki, 2019-11-07:
> 
> It appears to be id=9,15,8
> And
> Bus 008 Device 002: ID 256c:006e
> Vendor id 256c is huion.
> 
> Have you tried typing "sudo mod probe hid-uclogic" in your terminal?
> 
> You can also try running "evtest" and see if it lists and/or responds to input.

i didn't used any command or any settings.
To make sure, I downloaded the Fedora 31 KDE spin. (Fedora-KDE-Live-x86_64-31-1.9.iso)
I could use the tablet when fedora-live-kde-spin started.

but i need help, how to configure those button. O_O

[liveuser@localhost-live ~]$ lsusb
Bus 003 Device 002: ID 0781:5567 SanDisk Corp. Cruzer Blade
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 011 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 010 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 004: ID 256c:006e
Bus 002 Device 003: ID 03f0:002a HP, Inc LaserJet P1102
Bus 002 Device 002: ID 2109:2813 VIA Labs, Inc. USB2.0 Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 045e:0768 Microsoft Corp. Sidewinder X4
Bus 004 Device 002: ID 09da:f213 A4Tech Co., Ltd. USB Device
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

[liveuser@localhost-live ~]$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0: Power Button
/dev/input/event1: Power Button
/dev/input/event2: COMPANY USB Device
/dev/input/event3: Microsoft Microsoft® SiderWinderTM X4 Keyboard
/dev/input/event4: HID 256c:006e Pen
/dev/input/event5: HID 256c:006e Pad
/dev/input/event6: Microsoft Microsoft® SiderWinderTM X4 Keyboard Consumer Control
/dev/input/event7: Microsoft Microsoft® SiderWinderTM X4 Keyboard
/dev/input/event8: HDA ATI HDMI HDMI/DP,pcm=3
/dev/input/event9: HDA ATI HDMI HDMI/DP,pcm=7
/dev/input/event10: HDA ATI HDMI HDMI/DP,pcm=8
/dev/input/event11: HDA ATI HDMI HDMI/DP,pcm=9
/dev/input/event12: HDA ATI HDMI HDMI/DP,pcm=10
/dev/input/event13: HDA ATI SB Front Mic
/dev/input/event14: HDA ATI SB Rear Mic
/dev/input/event15: HDA ATI SB Line
/dev/input/event16: HDA ATI SB Line Out Front
/dev/input/event17: HDA ATI SB Line Out Surround
/dev/input/event18: HDA ATI SB Line Out CLFE
/dev/input/event19: HDA ATI SB Line Out Side
/dev/input/event20: HDA ATI SB Front Headphone
Select the device event number [0-20]: 5
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "HID 256c:006e Pad"
Supported events:
Event type 0 (EV_SYN)
Event type 1 (EV_KEY)
Event code 256 (BTN_0)
Event code 257 (BTN_1)
Event code 258 (BTN_2)
Event code 259 (BTN_3)
[... 121 lines cut ...]
Event: time 1573143235.710172, -------------- SYN_REPORT ------------
Event: time 1573143236.168102, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1573143236.168102, type 1 (EV_KEY), code 264 (BTN_8), value 1
Event: time 1573143236.168102, -------------- SYN_REPORT ------------
Event: time 1573143236.298082, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1573143236.298082, type 1 (EV_KEY), code 264 (BTN_8), value 0
Event: time 1573143236.298082, -------------- SYN_REPORT ------------
Event: time 1573143236.788004, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1573143236.788004, type 1 (EV_KEY), code 265 (BTN_9), value 1
Event: time 1573143236.788004, -------------- SYN_REPORT ------------
Event: time 1573143236.951980, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1573143236.951980, type 1 (EV_KEY), code 265 (BTN_9), value 0
Event: time 1573143236.951980, -------------- SYN_REPORT ------------
Event: time 1573143237.505896, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1573143237.505896, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1573143237.505896, -------------- SYN_REPORT ------------
Event: time 1573143237.669872, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1573143237.669872, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1573143237.669872, -------------- SYN_REPORT ------------
Event: time 1573143238.125802, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1573143238.125802, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
Event: time 1573143238.125802, -------------- SYN_REPORT ------------
Event: time 1573143238.355767, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1573143238.355767, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
Event: time 1573143238.355767, -------------- SYN_REPORT ------------

Akezhar, 2019-11-07:
Please keep the issue on topic, if you are not using the digimend driver why are you asking for help here?
I'm sure your distro maintainer has a forum for help.

alihakki, 2019-11-07:
i wrote kernel version too. if you delete digimend driver it can be works. maybe need upgrade kernel too.
and digimend driver cant works with usb3 too, but new kernel works perfectlee with usb 3 (which bruce lee is perfect :D)
and this("but i need help, how to configure those button. O_O") was a  metaphor.  i dont need help realy. i dont use any button still because im an amateur :D

alihakki, 2019-11-07:
Dude delete digimend driver. 
i test it for you with kubuntu-19.10 (kubuntu-19.10-desktop-amd64.iso)
when kubuntu live start, tablet  works perfect.

kubuntu@kubuntu:~$ lsusb
Bus 003 Device 002: ID 0781:5567 SanDisk Corp. Cruzer Blade
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 011 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 010 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 004: ID 256c:006e  
Bus 002 Device 003: ID 03f0:002a HP, Inc LaserJet P1102
Bus 002 Device 002: ID 2109:2813 VIA Labs, Inc. USB2.0 Hub             
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 045e:0768 Microsoft Corp. Sidewinder X4
Bus 004 Device 002: ID 09da:f213 A4Tech Co., Ltd. USB Device
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

kubuntu@kubuntu:~$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Power Button
/dev/input/event1:      Power Button
/dev/input/event2:      COMPANY USB Device
/dev/input/event3:      HDA ATI HDMI HDMI/DP,pcm=3
/dev/input/event4:      Microsoft Microsoft® SiderWinderTM X4 Keyboard
/dev/input/event5:      Microsoft Microsoft® SiderWinderTM X4 Keyboard Consumer Control
/dev/input/event6:      Microsoft Microsoft® SiderWinderTM X4 Keyboard
/dev/input/event7:      HID 256c:006e Pen
/dev/input/event8:      HID 256c:006e Pad
/dev/input/event9:      HDA ATI HDMI HDMI/DP,pcm=7
/dev/input/event10:     HDA ATI HDMI HDMI/DP,pcm=8
/dev/input/event11:     HDA ATI HDMI HDMI/DP,pcm=9
/dev/input/event12:     HDA ATI HDMI HDMI/DP,pcm=10
/dev/input/event13:     HDA ATI SB Front Mic
/dev/input/event14:     HDA ATI SB Rear Mic
/dev/input/event15:     HDA ATI SB Line
/dev/input/event16:     HDA ATI SB Line Out Front
/dev/input/event17:     HDA ATI SB Line Out Surround
/dev/input/event18:     HDA ATI SB Line Out CLFE
/dev/input/event19:     HDA ATI SB Line Out Side
/dev/input/event20:     HDA ATI SB Front Headphone
Select the device event number [0-20]: 8
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "HID 256c:006e Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
[... 58 lines cut ...]
Event: time 1573149802.216059, -------------- SYN_REPORT ------------
Event: time 1573149802.737897, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1573149802.737897, type 1 (EV_KEY), code 264 (BTN_8), value 1
Event: time 1573149802.737897, -------------- SYN_REPORT ------------
Event: time 1573149802.901850, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1573149802.901850, type 1 (EV_KEY), code 264 (BTN_8), value 0
Event: time 1573149802.901850, -------------- SYN_REPORT ------------
Event: time 1573149803.259741, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1573149803.259741, type 1 (EV_KEY), code 265 (BTN_9), value 1
Event: time 1573149803.259741, -------------- SYN_REPORT ------------
Event: time 1573149803.423692, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1573149803.423692, type 1 (EV_KEY), code 265 (BTN_9), value 0
Event: time 1573149803.423692, -------------- SYN_REPORT ------------
Event: time 1573149803.977472, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1573149803.977472, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1573149803.977472, -------------- SYN_REPORT ------------
Event: time 1573149804.141472, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1573149804.141472, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1573149804.141472, -------------- SYN_REPORT ------------
Event: time 1573149804.501380, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1573149804.501380, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
Event: time 1573149804.501380, -------------- SYN_REPORT ------------
Event: time 1573149804.697309, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1573149804.697309, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
Event: time 1573149804.697309, -------------- SYN_REPORT ------------

YuudaiNoboru, 2019-11-07:
> Dude, delete digimend driver.
> i test it for you with kubuntu-19.10 (kubuntu-19.10-desktop-amd64.iso)
> when kubuntu live start, tablet works perfect.
> 
> and if you do button settings file share me too pls. thx, i dont know which configure is good for start.
> 
> kubuntu@kubuntu:~$ lsusb
> Bus 003 Device 002: ID 0781:5567 SanDisk Corp. Cruzer Blade
> Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 011 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 010 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 002 Device 004: ID 256c:006e
> Bus 002 Device 003: ID 03f0:002a HP, Inc LaserJet P1102
> Bus 002 Device 002: ID 2109:2813 VIA Labs, Inc. USB2.0 Hub
> Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 004 Device 003: ID 045e:0768 Microsoft Corp. Sidewinder X4
> Bus 004 Device 002: ID 09da:f213 A4Tech Co., Ltd. USB Device
> Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> 
> kubuntu@kubuntu:~$ sudo evtest
> No device specified, trying to scan all of /dev/input/event*
> Available devices:
> /dev/input/event0: Power Button
> /dev/input/event1: Power Button
> /dev/input/event2: COMPANY USB Device
> /dev/input/event3: HDA ATI HDMI HDMI/DP,pcm=3
> /dev/input/event4: Microsoft Microsoft® SiderWinderTM X4 Keyboard
> /dev/input/event5: Microsoft Microsoft® SiderWinderTM X4 Keyboard Consumer Control
> /dev/input/event6: Microsoft Microsoft® SiderWinderTM X4 Keyboard
> /dev/input/event7: HID 256c:006e Pen
> /dev/input/event8: HID 256c:006e Pad
> /dev/input/event9: HDA ATI HDMI HDMI/DP,pcm=7
> /dev/input/event10: HDA ATI HDMI HDMI/DP,pcm=8
> /dev/input/event11: HDA ATI HDMI HDMI/DP,pcm=9
> /dev/input/event12: HDA ATI HDMI HDMI/DP,pcm=10
> /dev/input/event13: HDA ATI SB Front Mic
> /dev/input/event14: HDA ATI SB Rear Mic
> /dev/input/event15: HDA ATI SB Line
> /dev/input/event16: HDA ATI SB Line Out Front
> /dev/input/event17: HDA ATI SB Line Out Surround
> /dev/input/event18: HDA ATI SB Line Out CLFE
> /dev/input/event19: HDA ATI SB Line Out Side
> /dev/input/event20: HDA ATI SB Front Headphone
> Select the device event number [0-20]: 8
> Input driver version is 1.0.1
> Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
> Input device name: "HID 256c:006e Pad"
> Supported events:
> Event type 0 (EV_SYN)
> Event type 1 (EV_KEY)
> Event code 256 (BTN_0)
> Event code 257 (BTN_1)
> Event code 258 (BTN_2)
> Event code 259 (BTN_3)
> Event code 260 (BTN_4)
> Event code 261 (BTN_5)
> Event code 262 (BTN_6)
> Event code 263 (BTN_7)
> Event code 264 (BTN_8)
> Event code 265 (BTN_9)
> Event code 304 (BTN_SOUTH)
> Event code 305 (BTN_EAST)
> Event code 331 (BTN_STYLUS)
> Event type 3 (EV_ABS)
[... 62 lines cut ...]
> Event: time 1573149802.737897, type 1 (EV_KEY), code 264 (BTN_8), value 1
> Event: time 1573149802.737897, -------------- SYN_REPORT ------------
> Event: time 1573149802.901850, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
> Event: time 1573149802.901850, type 1 (EV_KEY), code 264 (BTN_8), value 0
> Event: time 1573149802.901850, -------------- SYN_REPORT ------------
> Event: time 1573149803.259741, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
> Event: time 1573149803.259741, type 1 (EV_KEY), code 265 (BTN_9), value 1
> Event: time 1573149803.259741, -------------- SYN_REPORT ------------
> Event: time 1573149803.423692, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
> Event: time 1573149803.423692, type 1 (EV_KEY), code 265 (BTN_9), value 0
> Event: time 1573149803.423692, -------------- SYN_REPORT ------------
> Event: time 1573149803.977472, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
> Event: time 1573149803.977472, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
> Event: time 1573149803.977472, -------------- SYN_REPORT ------------
> Event: time 1573149804.141472, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
> Event: time 1573149804.141472, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
> Event: time 1573149804.141472, -------------- SYN_REPORT ------------
> Event: time 1573149804.501380, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
> Event: time 1573149804.501380, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
> Event: time 1573149804.501380, -------------- SYN_REPORT ------------
> Event: time 1573149804.697309, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
> Event: time 1573149804.697309, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
> Event: time 1573149804.697309, -------------- SYN_REPORT ------------

I think when I used Ubuntu 19.10 I installed the drive even before testing tablet. I will try to test without the drive. Did you just plug in the tablet and it worked? No additional setup or installation?

alihakki, 2019-11-07:
yes i download  (ubuntu)kubuntu-19.10-desktop-amd64.iso
run live iso,  no need any config or settings or install, it will run.

note. this tablet a little dumb if plug unplug plug usb rapidly, it dont work some time. unplug it and wait 2 or 3 minuts and plug it again.

Majomi, 2019-11-10:
There seems some support for the tablett on Wayland, but on Xorg i cant get it to work.
I tried it on Arch Linux on `linux (5.3.x)` and `linux-lts (4.19.x)` on neither of those kernels it worked (only on 5.2.x)

On 4.19 Kernel it gets registered as Tablett but wont register any inputs.

YuudaiNoboru, 2019-11-10:
> There seems some support for the tablett on Wayland, but on Xorg i cant get it to work.
> I tried it on Arch Linux on `linux (5.3.x)` and `linux-lts (4.19.x)` on neither of those kernels it worked (only on 5.2.x)
> 
> On 4.19 Kernel it gets registered as Tablett but wont register any inputs.

I tested Kubuntu 19.10, but it didn't work, maybe because I'm using Xorg. I was only able to use the tablet on Ubuntu 18.04 anyway, I will expect some support, maybe for Ubuntu 20.04.

jakub-kaminski, 2019-12-05:
Hi Everyone,
I am using kernel `5.3.0-22-generic` on Pop_OS (like Ubuntu) 18.04, and `HUION New 1060 Plus`. As opposed to the Author of this thread, I can detect the tablet using `xinput list`, `xsetwacom list`. However, it does not work at all. Nothing happens when trying to use buttons or pen (fully charged). Also, devices listed are:

```
Wacom Pen and multitouch sensor Finger touch    id: 15  type: TOUCH
Wacom Pen and multitouch sensor Pen stylus      id: 16  type: STYLUS
Wacom Pen and multitouch sensor Pen eraser      id: 24  type: ERASER
```
I do not see the `Pad` device. I have the most recent, 9th version of digimend installed.
Is there any way to make it working?

Using lsusb I get: `056a:5193` for device UID.

What else should I consider to add to this post to make this issue better solvable?
How can I help you with sharing my config? (for those who are unable even see the tablet) 
Thank you for your time!
Regards, Jakub

earldbjr, 2019-12-09:
Has anyone gotten this to work?
I reinstalled from source to try debian version 10, but the errors kicked back by usb-devices remains:

`T:  Bus=03 Lev=04 Prnt=08 Port=00 Cnt=01 Dev#=  9 Spd=12  MxCh= 0
D:  Ver= 1.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=256c ProdID=006e Rev=30.00
C:  #Ifs= 0 Cfg#= 0 Atr= MxPwr=
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bInterfaceNumber': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bAlternateSetting': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bNumEndpoints': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bInterfaceClass': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bInterfaceSubClass': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bInterfaceProtocol': No such file or directory
/usr/bin/usb-devices: 76: printf: (none): expected numeric value
I:  If#=0x Alt= 0 #EPs= 0 Cls=() Sub= Prot= Driver=
`

Akezhar, 2019-12-29:
The problem is still persisting even with kernel 5.4 and 5.5; for some reason the tablet doesn't get recognised at all, and I don't know how else to troubleshoot the problem.
If anyone has any clue it would be greatly appreciated.

earldbjr, 2019-12-29:
> The problem is still persisting even with kernel 5.4 and 5.5; for some reason the tablet doesn't get recognised at all, and I don't know how else to troubleshoot the problem.
> If anyone has any clue it would be greatly appreciated.

I finally solved it, at least for my problem. Disable Usb3 in the bios (called xHCI in my bios, may be eHCI in yours).

I did that, rebooted, everything worked 100% with no further fiddling.

Akezhar, 2019-12-30:
Sadly I don't have that option, I can only let the mobo handle the xHCI if the OS can't do it but that's not helping the situation.
Thanks for sharing though.

Akezhar, 2020-01-17:
Doing more research I stumbled upon [this](https://bbs.archlinux.org/viewtopic.php?id=249783) thread, I'm not sure what to make of it, but it seems like it might be a problem with the kernel and usb driver.
Hopefully this can help with figuring out a solution.

Mstrodl, 2020-02-15:
Archlinux has updated `linux-lts` past version 5.2... Any suggestions for getting an old enough kernel now?

menaechmi, 2020-02-21:
If you are comfortable patching the kernel, I have made a patch for 5.5 that removes all calls to the tt_buffer_clear and lets the tablet work on current stable kernels:

[Huion1060P-kernel-patch](https://github.com/menaechmi/Huion1060P-kernel-patch)

While its definitely a bit like using a band-aid to cover a bullethole, more study would be required to actually figure out the problem and I just wanted to use my tablet again.

Mstrodl, 2020-02-22:
@menaechmi Interestingly, it seems as though compiling the module separately doesn't fix the issue. I'm using Arch Linux and made [a package](https://github.com/Mstrodl/xhci-fixy) to build only the xhci* drivers with this patch and it seems like it made no difference? I still get error -32

fernandodandrea, 2020-04-06:
I have a H1060P and it seems it's not the same model you other people are using, as it is reported as `xinput list` shows it as `HID 256c:006d Pen Pen (0)`, instead of `256c:006e`.

Anyway, my problem is: the pen did work, even before using the digimend drivers, including pressure on Krita. But no KDE tool or command allows me to access the hardware to configure it, and I'm left-handed. This means I need to be able to turn the tablet upside-down, as Windows config tools allow me to do.

I guess I'll have to check what happens if I change the right port do USB2.0 mode.

AlexanderBartash, 2020-04-28:
Hey guys. I am using `Kubuntu 20.04` and `Huion New 1060 Plus` and I managed to make it work but with an extra step which is **sending system into sleep `systemctl suspend` and waking it up**. For some reason it makes the system detect the tablet. System **reboot does not do that**, I have tried it multiple times. There is something special happens during system wake up routine.

Shortly the steps are:
1) Connect your tablet to USB.
2) Install `digimend-kernel-drivers` (I have been using the latest version from master d36d2cac7db997238508e163156ffd6b139c730e and installed it using `sudo make dkms_install` with secure boot disabled)
3) Reboot your system.
4) After reboot tablet won't work.
5) Execute in terminal `systemctl suspend`
6) Wake up your PC and tablet will be detected by system and configuration will be available in Wacom Tablet Settings

System information:
1) I have secure boot disabled, I do not want to deal with it at the moment, later I might try to enable it.
2) Kernel `5.4.0-26-generic`
```
$ sudo uname -a
[sudo] password for sasha: 
Linux kubuntu 5.4.0-26-generic #30-Ubuntu SMP Mon Apr 20 16:58:30 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
```
Operating System: Kubuntu 20.04
KDE Plasma Version: 5.18.4
KDE Frameworks Version: 5.68.0
Qt Version: 5.12.8
Kernel Version: 5.4.0-26-generic
OS Type: 64-bit
Processors: 8 × Intel® Core™ i7-4790 CPU @ 3.60GHz
Memory: 31.3 GiB of RAM
```
3) I have tried USB 2.0 and USB 3.0 and any be used because both work with the tablet.

When I connect the tablet to USB 2.0 or USB 3.0 it does **not** work. Installing `digimend-kernel-drivers` 9.0 or later from master does not do any difference. I have tried installing form source and using DKMS and using Debian package.
But if you do `systemctl suspend` then wake it up the system will see your tablet even without `digimend-kernel-drivers` but without drivers configuration in Wacom Tablet Settings is not available. 

But if you install `digimend-kernel-drivers` you will also be able to configure it like on the below screenshot.

All of the below has been captured after doing `systemctl suspend` and waking it up.

[image]
```
$ xinput --list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Telink Wireless Receiver Mouse            id=10   [slave  pointer  (2)]
⎜   ↳ Telink Wireless Receiver Consumer Control id=11   [slave  pointer  (2)]
⎜   ↳ COUGAR Vantar COUGAR Vantar Consumer Control      id=14   [slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pen stylus                  id=24   [slave  pointer  (2)] <<<<<<<<<<<<<<<<<<
⎜   ↳ HID 256c:006e Touch Strip pad             id=25   [slave  pointer  (2)] <<<<<<<<<<<<<<<<<<
⎜   ↳ HID 256c:006e Pad pad                     id=26   [slave  pointer  (2)] <<<<<<<<<<<<<<<<<<
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Yubico YubiKey OTP+FIDO+CCID              id=8    [slave  keyboard (3)]
    ↳ Telink Wireless Receiver System Control   id=9    [slave  keyboard (3)]
    ↳ Telink Wireless Receiver                  id=12   [slave  keyboard (3)]
    ↳ COUGAR Vantar COUGAR Vantar               id=13   [slave  keyboard (3)]
    ↳ COUGAR Vantar COUGAR Vantar System Control        id=15   [slave  keyboard (3)]
    ↳ COUGAR Vantar COUGAR Vantar Keyboard      id=16   [slave  keyboard (3)]
    ↳ Blue Microphones Yeti Stereo Microphone Consumer Control  id=17   [slave  keyboard (3)]
    ↳ ASUSTeK ASUS XONAR U5                     id=18   [slave  keyboard (3)]
    ↳ Logitech BRIO                             id=19   [slave  keyboard (3)]
    ↳ Logitech BRIO Consumer Control            id=20   [slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                        id=21   [slave  keyboard (3)]
    ↳ Telink Wireless Receiver Consumer Control id=22   [slave  keyboard (3)]
    ↳ COUGAR Vantar COUGAR Vantar Consumer Control      id=23   [slave  keyboard (3)]
```

```
$ sudo lsusb 
[sudo] password for sasha: 
[... 59 lines cut ...]
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 306 (BTN_C)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
```

AlexanderBartash, 2020-04-28:
@spbnick Hi. Do you have any ideas why sending system into sleep via `systemctl suspend and` waking it up makes the tablet work? Maybe it is possible to do a code fix. I I can help somehow by providing more information I'd be glad to help.

spbnick, 2020-04-28:
@AlexanderBartash, can't think of anything beside something in the USB stack. There's https://github.com/DIGImend/digimend-kernel-drivers/blob/master/hid-uclogic-core.c#L264-L277 of course, but it should be no different to just unloading and reloading the module, for example, or rebinding the tablet to hid-generic and back to hid-uclogic.

mirenradia, 2020-04-28:
> Hey guys. I am using `Kubuntu 20.04` and `Huion New 1060 Plus` and I managed to make it work but with an extra step which is **sending system into sleep via `systemctl suspend` and waking it up**. For some reason it makes the system detect the tablet. System **reboot does not do that**, I have tried it multiple times. There is something special happens during system wake up routine.

For what it's worth, I was unable to reproduce this behaviour with the Huion New 1060 Plus on Ubuntu 18.04 (KDE Plasma 5.12.9) with the `5.4.28-050428-generic` kernel and release v9 of `digimend-dkms`.

AlexanderBartash, 2020-04-28:
Hi. Did you follow the steps in the exact order as I described? Like
connecting your tablet, rebooting etc

mirenradia, 2020-04-28:
> Hi. Did you follow the steps in the exact order as I described? Like connecting your tablet, rebooting etc
Almost exactly (I did check as well before rebooting just in case) but I had already installed `digimend-dkms` v9.0 so I skipped that step.

lleraj, 2020-04-29:
I've uploaded a [patched](https://bugzilla.kernel.org/show_bug.cgi?id=207065#c7) kernel to the [aur](https://aur.archlinux.org/packages/linux-xhci/). It's all working without installing installing the drivers at all.

Akezhar, 2020-04-29:
> I've uploaded a [patched](https://bugzilla.kernel.org/show_bug.cgi?id=207065#c7) kernel to the [aur](https://aur.archlinux.org/packages/linux-xhci/). It's all working without installing installing the drivers at all.

I've just build the patched kernel and can confirm that it is working, thanks a lot.
I suppose we'll need new patched kernels for as long as this doesn't get fixed in the main kernel?
Sorry if this seems obvious but it's the first time I use a patched kernel.

lleraj, 2020-05-01:
The patches needed for the tablett to work should be implemented in linux-5.6.8 which is now available on arch. So the aur package is not longer needed
[Changelog](https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.6.8)

Cheers

piegamesde, 2020-05-01:
I can confirm that it works again with the new kernel. Even though the display mapping is broken (it always uses all my monitors), but that may be a configuration issue somewhere else.

Akezhar, 2020-05-02:
Another confirmation that the tablet works with kernel 5.6.8.
Thanks to everyone that worked on a fix for this.
The issue is resolved so I'm closing it.

closed 2020-05-02 by Akezhar