DIGImend/digimend-kernel-drivers #321: Support for Genius EasyPen m406

TPb1HDELj, 2019-11-02:
Hello! 
Add support for the topic, please.

[digimendPage](https://digimend.github.io/tablets/KYE_EasyPen_M406/)
[3dnews](https://3dnews.ru/596016)

[descritpors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3800843/descritpors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3800844/frame_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3800845/frame_dials.txt)
[hid_report_descritors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3800846/hid_report_descritors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3800847/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3800848/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3800849/pen_pressure.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3800850/probe.txt)

TPb1HDELj, 2019-11-02:
the mouse is lost

Christian2006-Dex, 2020-03-09:
What Linux and version do you use? I have a similar problem.

MrPancakes39, 2021-09-24:
I have same problem, I use Ubuntu 200.04

It doesn't seem to be supported yet, will it ever get support? and how to help?

nic3-14159, 2021-09-24:
Hello! Correct, this tablet isn't supported yet. I'm mostly familiar with uclogic based tablets, but looking at the kye code and the dumps, it looks like it should be fairly simple to add supoort for.

I'm a bit busy right now, but I'll see if I can add support soon.

@Christian2006-Dex @MrPancakes39 By the way, if either of you has the mouse that comes with this tablet, could you also post raw input samples for that? Instructions for doing that are [here](https://digimend.github.io/support/howto/trbl/diagnostics/), under the section "Collect raw input samples" > Mouse

MrPancakes39, 2021-09-24:
Oh thanks for the response! Unfortunately my tablet didn't come with a mouse so I can't help with that... sorry.
By the way, thanks for writing drivers you're a life saver!

nic3-14159, 2021-09-25:
Hello all! Could you test the following branch and report if it works? I noticed that the pen input samples didn't seem to be reporting pressure, but I did see that the existing DIGImend code sends an initialization message for some of the EasyPen tablets, which seems like it might enable the pressure. I just assumed this tablet operates similarly to those tablets and just copied and tweaked that code, so I'm not 100% sure if this will actually match the reports the tablet sends.

https://github.com/nic3-14159/digimend-kernel-drivers/tree/easypen-m406

Also, I did a bit of digging and it seems like the mouse was just another one of Genius's products that came packaged with the tablet as a bonus. I think some tablets had mice that only worked on top of the tablet, but since the one packaged with the M406 is a standalone product, it should just work as a normal USB mouse without the tablet, and thus retrieving input samples for it doesn't actually need to be done.

MrPancakes39, 2021-09-25:
Hello again @nic3-14159, I installed the drivers from source and the tablet now works. I tested the pressure sensitivity in Krita (since it supports this tablet) and it works too! Thank you for your help and support ♥♥

nic3-14159, 2021-09-25:
@MrPancakes39 Great! The tablet buttons are currently hard mapped to Redo, Undo, Zoom in, and Zoom out application buttons, as that's what was done for many of the other Genius tablets, probably because the buttons are labeled with specific functions on the tablet itself. However, application support for those seems a little flaky. You should be able to remap them to whatever shortcut you want using a tool like [key-mapper](https://github.com/sezanzeb/key-mapper/) though.

If you don't mind, could you also post raw input samples while using the driver I wrote? That way there is some reference for the input reports after performing the initialization procedure.

MrPancakes39, 2021-09-25:
@nic3-14159 I forgot to test the buttons but (for me) they don't seem to work, I'm okay with that thou.
And for raw input samples I'll send them sure thing.

MrPancakes39, 2021-12-11:
Hello again @nic3-14159, I didn't forget about the raw input samples only been super busy with university loads.
:heart: Thanks again, you're a life saver! :heart:
(Note: My pen's tip is slightly broken, so I tried my best with pen_tilt sample);

raw input samples:
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7696845/frame_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7696846/frame_dials.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7696847/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7696849/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7696850/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7696851/pen_tilt.txt)
(Note: I don't have the mouse for mouse samples; Also really nice documentation 👍🏻;)

MrPancakes39, 2021-12-17:
> You should be able to remap them to whatever shortcut you want using a tool like [key-mapper](https://github.com/sezanzeb/key-mapper/) though.

Thanks also @nic3-14159 for this suggestion.