DIGImend/digimend-kernel-drivers #322: Huion new 1060 plus and soft keys

Patlol, 2019-11-14:
Hello, 
My Huion new 1060 plus tablet with your latest debian package (V9) works perfectly on my linux mint 19.1 Mate 64 bit (kernel 4.15.0-20-generic). Thank you very much for all this work that saves the lives of many people! :)
But is there a way to use and configure the 16 soft keys? I can't find anything on that. When I put the stylus on it I have my cursor at the top of my screen but nothing happens, they are apparently not assigned or activated.
Thank you.

JShorthouse, 2019-11-25:
As far as I know the soft keys are implemented entirely in software. If you wanted to replicate their functionality you could find / create a program that overlays 16 invisible buttons at the top of your monitor. I don't know if a program like that already exists or not.

Patlol, 2019-12-01:
Thank you for your answer. 
Unfortunately I have no idea how to program this! Or where to find it.

spbnick, 2020-05-17:
The soft buttons are essentially just marketing gimmick, and are not worth the trouble. You would have to look at the tablet to use them, taking your attention from your work on the screen. You might as well just put a panel with buttons on your screen where you like it and use those instead. There's a reason why Wacom tablets don't have them.

closed 2020-05-17 by spbnick