DIGImend/digimend-kernel-drivers #323: Cannot map buttons for H640P on debian

ghistes, 2019-11-15:
Hi,

I apologize in advance as this issue is probably just me being stupid, but I cannot figure it out..

I've got an Huion H640P that I would like to use with debian testing.

It seems to work out of the box on my system as far as the basic functionality is concerned but I also want to use the buttons.

So I installed  digimend-dkms_9_all.deb and created a file in /etc/X11/xorg.conf.d with this content:

Section "InputClass"
        Identifier "Huion"
        MatchUSBID "256c:006e"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection

So now I get this:

/home/mh > xinput --list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ HID 046a:0023                             id=9    [slave  pointer  (2)]
⎜   ↳ Logitech USB Optical Mouse                id=10   [slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pen stylus                  id=11   [slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad pad                     id=12   [slave  pointer  (2)]
⎜   ↳ Advanced Silicon S.A CoolTouch(TM) System id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ HID 046a:0023                             id=8    [slave  keyboard (3)]
    ↳ HID 046a:0023                             id=14   [slave  keyboard (3)]

/home/mh > xsetwacom --list
HID 256c:006e Pen stylus                id: 11  type: STYLUS
HID 256c:006e Pad pad                   id: 12  type: PAD

However when I try to map the buttons I get an error:

/home/mh > xsetwacom --set "Pad pad" Button 1 "key a"
Cannot find device 'Pad pad'.

I have tried a lot of values instead of  "Pad pad", but nothing seems to work.

What am I doing wrong here?

Many thanks!

ghistes, 2019-11-15:
I finally figured it out...

xsetwacom --set "HID 256c:006e Pad pad" Button 1 "key a"

works.

Sorry.

I would be interested if there was a way to make this "HID 256c:006e Pad pad" nicer.

6etacat, 2019-11-27:
Please close this issue.

closed 2019-12-03 by ghistes