DIGImend/digimend-kernel-drivers #326: unable to compile kernel driver for kernel 5.4.1

gianlucapettinello, 2019-12-02:
After upgrading kernel to 5.4.1 I get an error when dkms compiles digimend kernel driver. See attached make.log for more info. OS: Archlinux
[make.log](https://github.com/DIGImend/digimend-kernel-drivers/files/3913499/make.log)

FraYoshi, 2019-12-10:
Same for kernel 5.4.2-1 (archlinux) dkms build.

```
Creating symlink /var/lib/dkms/digimend/10/source ->
                 /usr/src/digimend-10

DKMS: add completed.
dkms build digimend/10

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.4.2-arch1-1 KVERSION=5.4.2-arch1-1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.2-arch1-1 (x86_64)
Consult /var/lib/dkms/digimend/10/build/make.log for more information.
make: *** [Makefile:74: dkms_modules_install] Error 10
```

Make Log in detail:
[make.log](https://github.com/DIGImend/digimend-kernel-drivers/files/3943718/make.log)

smt923, 2019-12-18:
Also an issue on 5.4.3, same error as above

gonX, 2019-12-19:
See if #331 fixes it for you, thanks

smt923, 2019-12-19:
making the change in the PR allowed it to build for me on 5.4, thanks!

FraYoshi, 2019-12-20:
Solved also for me in 5.4.2-arch1-1, thanks!

spbnick, 2020-01-06:
Thanks everybody for the reports and @gonX for the fix! Merged now.

PR #331 (2020-01-06): Fix build on Linux kernel >= 5.4
5.4 seems to mandate the use of `M` in addition to `SUBDIRS` when building from kernel sources.

This PR fixes DKMS build not working with 5.4 kernels.

Fixes #326
closed 2020-01-06 by spbnick