DIGImend/digimend-kernel-drivers #328: digimend-dkms_9_all.deb install: HUION H430P not working

pelgrimforever, 2019-12-15:
OS: Ubuntu 18.04.3 LTS
processor: Intel® Core™ i7 CPU 920 @ 2.67GHz × 8
GNOME: 3.28.2
64-bit

installed the V9 release found on:
https://github.com/DIGImend/digimend-kernel-drivers/releases

.deb file installs without any error, tabled is still nog recognized.

output of
cat /proc/bus/input/devices
`I: Bus=0003 Vendor=256c Product=006d Version=0111
N: Name="HUION Huion Tablet"
P: Phys=usb-0000:00:1d.2-1/input1
S: Sysfs=/devices/pci0000:00/0000:00:1d.2/usb8/8-1/8-1:1.1/0003:256C:006D.0009/input/input19
U: Uniq=
H: Handlers=mouse1 event5 
B: PROP=0
B: EV=1b
B: KEY=c03 0 0 0 0 0
B: ABS=1000003
B: MSC=10`

Madac1, 2019-12-18:
Hey, same here, bough the Huion Inspiroy H430P from aliexpress, works fine on windows 10 and on my android 9.0 phone. But it seems to not work on my laptop with linux mint 19.2.
The tablet its supposed to be 256c:006e, but my laptop reads it as 256c:006d (same as you).
On the DIGIment website shows that the 256c:006d model (that correspond to the Huion HS610 model) is unsupported atm.

SirAd, 2019-12-19:
Hi,

I have sam issue. See dmesg snippet:

```
[  538.995953] usb 1-4: new full-speed USB device number 7 using xhci_hcd
[  539.146769] usb 1-4: New USB device found, idVendor=256c, idProduct=006d
[  539.146777] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  539.146782] usb 1-4: Product: Huion Tablet
[  539.146787] usb 1-4: Manufacturer: HUION
[  539.150265] hid-generic 0003:256C:006D.0004: hiddev2,hidraw3: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:14.0-4/input0
[  539.177808] input: HUION Huion Tablet as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:256C:006D.0005/input/input23
[  539.236593] hid-generic 0003:256C:006D.0005: input,hidraw4: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:14.0-4/input
```

I've changed modified source code in deb package

```
#define USB_VENDOR_ID_HUION             0x256c
#define USB_DEVICE_ID_HUION_TABLET      0x006d
```

and work as a charm.

Madac1, 2019-12-19:
Hey @pelgrimforever , check https://github.com/DIGImend/digimend-kernel-drivers/issues/316#issuecomment-567206661 

> Hi,
>
> I have sam issue. See dmesg snippet:
>
> ```
> [  538.995953] usb 1-4: new full-speed USB device number 7 using xhci_hcd
> [  539.146769] usb 1-4: New USB device found, idVendor=256c, idProduct=006d [...]

I'll test this out, my tablet is now working but I cant manage to get the tablet buttons to work, and one of the pen buttons doesn't work, thanks you a lot for helping <3

pelgrimforever, 2019-12-21:
Hi,

can you tell me how to do this code change ?
Didn't know this was possible in a deb package.

deansheather, 2019-12-21:
Related: #332 for the buttons not working

pelgrimforever, 2019-12-22:
ok, did some digging online.
this makes HUION H430P work on my pc.

https://github.com/DIGImend/digimend-kernel-drivers/releases
download digimend-dkms_9_all.deb
dpkg-deb -R digimend-dkms_9_all.deb digimend9
gedit digimend9/usr/share/X11/xorg.conf.d

Section "InputClass"
	Identifier "Huion tablets with Wacom driver"
	MatchUSBID "256c:006d"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
EndSection

sudo dpkg-deb -b digimend9 digimend9.deb
install digimend9.deb

Check installation
~$ xsetwacom list dev
HUION Huion Tablet eraser       	id: 12	type: ERASER    
HUION Huion Tablet stylus       	id: 11	type: STYLUS 

set buttons:

get modifiers list:
xsetwacom --list modifiers

xsetwacom --set 'HUION Huion Tablet stylus' Button 1 "button +1"
xsetwacom --set 'HUION Huion Tablet stylus' Button 2 "button +2"
xsetwacom --set 'HUION Huion Tablet stylus' Button 3 "button +3"

TODO:
- make the buttons on the tabled work, for now only the buttons on the pen work.
- limit the tablet to 1 screen. This worked for my wacom, but not for HUION.

SirAd, 2019-12-23:
Hi,
seems that problem is already fix in the code, but not in rel 9 deb package. I suggest to download the source code, build deb package and install it. 
It work for me without any modifications:
 
```
~$ git clone https://github.com/DIGImend/digimend-kernel-drivers
~$ cd digimend-kernel-drivers
~$ dpkg-buildpackage -b -uc
~$ sudo dpkg -i ../digimend-dkms_10_all.deb
```

pelgrimforever, 2019-12-24:
ok, if I only knew that 2 days ago :)

SirAd, thank you, that works indeed.
This is a much shorter installation process, with hopefully better code,
but it doesn't solve the last problems I have.

Output from ~$ xsetwacom list dev
HUION Huion Tablet Pen stylus   	id: 8	type: STYLUS    
HUION Huion Tablet Pad pad      	id: 9	type: PAD

So these make my stylus buttons work:
xsetwacom --set 'HUION Huion Tablet Pen stylus' Button 1 "button +1"
xsetwacom --set 'HUION Huion Tablet Pen stylus' Button 2 "button +2"
xsetwacom --set 'HUION Huion Tablet Pen stylus' Button 3 "button +3"

**But these don't make any of my pad buttons work**:
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 8 "button +1"
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 9 "button +1"
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 10 "button +1"
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 11 "button +1"
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 12 "button +1"

**Also still not possible to limit the tablet to 1 screen.**

Anickonoff, 2020-02-05:
> Also still not possible to limit the tablet to 1 screen.

To limit the tablet on one screen, the commands
`xsetwacom set "HUION Huion Tablet stylus" MapToOutput next`
or
`xsetwacom set "HUION Huion Tablet stylus" MapToOutput 855x572+0+0`
work fine.
The command
`xsetwacom set "HUION Huion Tablet stylus" MapToOutput DVI-D-0`
did not work.

dublinx333, 2020-09-05:
dpkg-buildpackage -b -uc  command results in 
"error: failed to sign .dsc file"