DIGImend/digimend-kernel-drivers #329: Gaomon 1060PRO xsetwacom --list showing Stylus and Eraser

sbstratos79, 2019-12-17:
I am very new to linux so I am not sure what other useful info I could provide for figuring this out but I tried the latest development version and the Huion HS610 version which is apparently very similar to Gaomon 1060PRO. They both have the same id in `lsusb` i.e. `256c:006d`.

I did a clean install by doing `make clean`, `sudo make dkms_uninstall`, `sudo make uninstall`, `make`, `sudo make dkms_install`. I tried installing it on kernel 5.4.2-1, but for some reason it kept giving me errors so I installed it on 4.19.88-1, where the installation was successful. But every time I do `xsetwacom --list` it shows 
`HUION Huion Tablet stylus       	id: 15	type: STYLUS    
HUION Huion Tablet eraser       	id: 21	type: ERASER` as the output

The tablet works in Krita out of the box but I would like to be able to customize the stylus and tablet buttons for other applications like Blender etc.

sbstratos79, 2019-12-23:
Doing a "uclogic probe" and "uclogic decode" did the job. Now the tablet is detected as a pad by both xinput and xsetwacom.

closed 2019-12-23 by sbstratos79

DraconYale, 2019-12-25:
Sorry, I tried using `sudo uclogic-probe BUS_NUM DEV_ADDR | uclogic-decode` but it didn't work.
Did you use other commands after that?

sbstratos79, 2019-12-26:
That's not how I did it. I did it the way it's described [here](https://github.com/DIGImend/uclogic-tools).
First I did a `lsusb` to check my tablet's `Bus` and `Device` number. Then I did `sudo uclogic-probe x y | uclogic-decode` where 'x' is the `Bus` number and 'y' is the `Device` number.

sbstratos79, 2019-12-26:
Maybe do a clean install again. I used the dkms method for installation. Also, make sure you restart your computer after doing the uclogic thing. A simple hotplug of the tablet didn't work for me. I had to restart the pc.

DraconYale, 2019-12-26:
I tried the clean install with the uclogic commands, but it didn't change much. I'm using a Huion H640P, maybe this method doesn't work because it's another tablet? I followed the guides in the readme and on the DIGImend website (editing some .conf files in `etc/X11/xorg.conf.d`) but it keeps recognize it as "stylus" and "eraser"

sbstratos79, 2019-12-26:
Huion H640P should be supported out of the box by digimend according to the driver page. In my experience, editing the conf files doesn't help much if the device is recognised as an eraser. Could your share the uclogic-decode output so I can confirm for you whether it's similar to the output I got. I can't use my computer atm otherwise I'd have shared my own output. Tbh even if it doesn't look similar, I can't help you much because I don't have much experience with this. But atleast we can cross check our results.