DIGImend/digimend-kernel-drivers #330: Huion 1060P not working since update to Eoan / kernel 5.3

earldbjr, 2019-12-19:
Hi everyone, I've scoured the internet and every forum post on every site I could find, but I'm still unable to get my 1060p working and could use some assistance. I got everything working fine on 19.4, but since the update to 19.10 it simply doesn't work.

I've tried installing version 8 and 9 dkms, no dice. I've compiled from source, nada. I've tried editing udev rules, tried every test I could find, still nothing.

On plugging in dmesg gives me this:
-----------------------------------
`[45649.530531] usb 3-10.1.2.1: new full-speed USB device number 24 using xhci_hcd
[45649.661421] usb 3-10.1.2.1: New USB device found, idVendor=256c, idProduct=006e, bcdDevice= 0.00
[45649.661426] usb 3-10.1.2.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[45649.667627] usb 3-10.1.2.1: can't set config #1, error -32
`

sudo lsusb -vvv gives this:
-----------------------------------
Bus 003 Device 012: ID 256c:006e  
Device Descriptor:

  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x256c 
  idProduct          0x006e 
  bcdDevice           30.00
  iManufacturer           5 (error)
  iProduct                6 (error)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              480mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
            bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      83
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2

usb-devices:
------------------------------------------
T:  Bus=03 Lev=04 Prnt=23 Port=00 Cnt=01 Dev#= 24 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=256c ProdID=006e Rev=00.00
C:  #Ifs= 0 Cfg#= 0 Atr= MxPwr=
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bInterfaceNumber': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bAlternateSetting': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bNumEndpoints': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bInterfaceClass': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bInterfaceSubClass': No such file or directory
cat: '/sys/bus/usb/devices/usb3/3-10/3-10.1/3-10.1.2/3-10.1.2.1/3-*:?.*/bInterfaceProtocol': No such file or directory
/usr/bin/usb-devices: 76: printf: (none): expected numeric value
I:  If#=0x Alt= 0 #EPs= 0 Cls=() Sub= Prot= Driver=

sudo usbhid-dump 256c:006e
---------------------------
Failed to find HID interfaces: Entity not found

No relevant devices listed on xinput nor evtest

I try to refrain from asking questions here in case I'm just missing something stupid, but the first several pages of results on google are all purple links for anything I can think of to search.

Thanks in advance for any insight.

earldbjr, 2019-12-20:
After what has to be a hundred hours of troubleshooting, I found that everything started working again after turning off xHCI in the BIOS.

Something about this tablet and USB3 just doesn't get along.

mkmarkera, 2019-12-31:
earldbjr, I don't know where is xHCI in my system. Anyway, my boot is not secure, so I don't know what else could I set? I didn't get problems in Ubuntu, but now in Mint, something is wrong. Give more feedback. 'cause I want to draw. :D

earldbjr, 2020-01-01:
> earldbjr, I don't know where is xHCI in my system. Anyway, my boot is not secure, so I don't know what else could I set? I didn't get problems in Ubuntu, but now in Mint, something is wrong. Give more feedback. 'cause I want to draw. :D My tablet is 1060 PLUS.

I speak as a layman who found a fix through persistence rather than an expert who simply fixed it.
From what I read, it's the system that handles USB3.0. If you can find any reference to USB3 and turn it off you may well be doing the same thing.

Short of that, maybe look into disabling USB3 in linux for the time being? It may have the same effect! My search stopped short of googling how to do that.

mkmarkera, 2020-01-01:
> I speak as a layman who found a fix through persistence rather than an expert who simply fixed it.
> From what I read, it's the system that handles USB3.0. If you can find any reference to USB3 and [...]

Did I understand correctly that you want me to NOT use USB3.0 (USB 3.0 (type A)) for anything? Cause I don't understand how to turn off physical port?!

earldbjr, 2020-01-01:
A USB3 port becomes a USB2 port if the hci isn't enabled. I still have use
of all my ports, just at USB2 speeds.

timtoo, 2020-01-30:
Still not working as of kernel 5.4.15 (manjaro). Thanks for figuring out the work-around.

earldbjr, 2020-01-30:
> Still not working as of kernel 5.4.15 (manjaro). Thanks for figuring out the work-around.

For what it's worth the exact commit that causes the problem has been found. Now it's a matter of deciding what the fix is.

plemelin, 2020-01-30:
> > Still not working as of kernel 5.4.15 (manjaro). Thanks for figuring out the work-around.
> 
> For what it's worth the exact commit that causes the problem has been found. Now it's a matter of deciding what the fix is.

Is it a commit in the digimend-kernel or a commit in the kernel?

I'm encountering this issue and I tried to disabled xHCI from the OS (https://www.systutorials.com/241533/how-to-force-a-usb-3-0-port-to-work-in-usb-2-0-mode-in-linux/) without success. I don't seem to have an option in my "Latitude E5470".

earldbjr, 2020-01-30:
> > > Still not working as of kernel 5.4.15 (manjaro). Thanks for figuring out the work-around.
> > 
> > 
> > For what it's worth the exact commit that causes the problem has been found. Now it's a matter of deciding what the fix is.
> 
> Is it a commit in the digimend-kernel-driver or a commit in the kernel?
> 
> I'm encountering this issue and I tried to disabled xHCI from the OS (https://www.systutorials.com/241533/how-to-force-a-usb-3-0-port-to-work-in-usb-2-0-mode-in-linux/) without success. I don't seem to have an option in my "Latitude E5470".

Its in the kernel. A new procedure was added (it was a buffer flush or ready flag or something, can't recall exactly) that has to be called when the device is connecting.

plemelin, 2020-01-31:
This is a dupe of #318

ThePreviousOne, 2020-03-04:
Seeing as this a duplicate, I 'll take this opportunity to hijack.
First I'd like to say STOP CALLING YOUR '1060Plus' A 1060P

Second I'd like to report that my 1060Pro (its the old one) is also not working.
The kernel driver doesn't register the tablet at all, and digimend driver (v10) it only appears under lsusb

Mstrodl, 2020-03-05:
@ThePreviousOne What does `dmesg` give after you've plugged the tablet in? Any interesting errors? Are you running on xhci? If yes, try disabling that to see if it fixes it

fernandodandrea, 2020-03-07:
I'm running kernel 5.5.7-1 Manjaro. My dmesg says this when I plug the H1060P in: https://pastebin.com/ybFpVUX4

Mstrodl, 2020-03-08:
@fernandodandrea If I'm not mistaken, this is probably unrelated as your device is actually picked up by the HID drivers and added properly. Create a new issue.

ThePreviousOne, 2020-03-08:
That's about what I'm getting too, though I'm not at my computer to check for a few days

Sent from ProtonMail mobile

-------- Original Message --------

jimmy00784, 2020-04-14:
Any update on this? I am unable to disable USB 3 on my Dell Inspiron. There was an article about turning USB Debug to Enabled in Bios would have the same effect, but I still see that xhci_hcd is being picked up for the HUION 1060PLUS
```
[  669.441779] usb 2-2: new high-speed USB device number 18 using xhci_hcd
[  669.458986] usb 2-2: New USB device found, idVendor=058f, idProduct=6254, bcdDevice= 1.00
[  669.458991] usb 2-2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  669.458994] usb 2-2: Product: USB2.0Hub
[  669.461020] hub 2-2:1.0: USB hub found
[  669.461096] hub 2-2:1.0: 4 ports detected
[  669.758462] usb 2-2.1: new full-speed USB device number 19 using xhci_hcd
[  669.876140] usb 2-2.1: New USB device found, idVendor=256c, idProduct=006e, bcdDevice= 0.00
[  669.876148] usb 2-2.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  669.883567] usb 2-2.1: can't set config #1, error -32
[  669.971659] usb 2-2.4: new high-speed USB device number 20 using xhci_hcd
```
It's kind of sad that the tablet works without necessarily installing drivers on Windows 10 but not on linux anymore.
```
$ uname -a
Linux karim-inspiron 5.6.3-arch1-1 #1 SMP PREEMPT Wed, 08 Apr 2020 07:47:16 +0000 x86_64 GNU/Linux
```
```
$ xinput 
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ DLL0674:00 06CB:75DB Mouse                id=12   [slave  pointer  (2)]
⎜   ↳ DLL0674:00 06CB:75DB Touchpad             id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Power Button                              id=8    [slave  keyboard (3)]
    ↳ Sleep Button                              id=9    [slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate           id=10   [slave  keyboard (3)]
    ↳ Intel Virtual Button driver               id=11   [slave  keyboard (3)]
    ↳ Dell WMI hotkeys                          id=14   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=15   [slave  keyboard (3)]
    ↳ DELL Wireless hotkeys                     id=16   [slave  keyboard (3)]

```
```
$ lsusb
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 008: ID 04f3:2031 Elan Microelectronics Corp. Touchscreen
Bus 002 Device 007: ID 8087:0a2a Intel Corp. 
Bus 002 Device 005: ID 0c45:64cb Microdia Integrated_Webcam_HD
Bus 002 Device 003: ID 0483:91d1 STMicroelectronics Sensor Hub
Bus 002 Device 020: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 002 Device 019: ID 256c:006e  
Bus 002 Device 018: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

jimmy00784, 2020-04-15:
On Archlinux, I was able to get it to work by downgrading the linux-lts kernel to version 4.19.101. This kernel was compiled with gcc 9.2.0, so gcc and dependencies also had to be downgraded from the latest 9.3.0 to 9.2.0. This allowed digimend and other modules to be rebuilt for linux-lts.
At this time, I've added linux-lts and linux-lts-headers to IngorePkg in pacman.conf. But that is less than ideal.

Mstrodl, 2020-04-16:
@jimmy00784 I've done the same

jnahelou, 2020-04-28:
I had the same error (`` 'can't set config # 1, error -32```) on my Dell XPS.
I tried a 5.6 kernel (got 5.3) using a kubuntu live CD without success .. So I decided to apply the 2 patches mentioned on an other thread/site  (https://bugzilla.kernel.org/show_bug.cgi?id=206897#c8).
After 1 hour compilation, my new 5.6.7 kernel found my huion 1060plus :1st_place_medal: 
Then I installed DIGImend drivers from sources (master) and now everything works!
Thanks all!