DIGImend/digimend-kernel-drivers #334: Huion H640P stylus/eraser and workspace problem

DraconYale, 2019-12-25:
Hi, after installing the digimend drivers following the readme and editing some .conf files, I managed to make the tablet work, Krita recognizes pressure really well, but I have two problems.

1. `xsetwacom list` print these infos:

```
HID 256c:006d stylus            	id: 16	type: STYLUS  
HID 256c:006d eraser            	id: 20	type: ERASER
```
While the guide has "PAD" instead of "ERASER". From what I understood, "PAD" is needed to customize the tablet's buttons.

2. It looks like that half of tablet "represents" my screen (in horizontal). For example, if I draw a line from left to right, I will reach the end of my screen at the mid of the tablet. The vertical axes are fine though.
I tried using the `xsetwacom set` command but it didn't work.

I'll post below the `xinput list` and `xrandr` output:

```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB Gaming Mouse                        	id=8	[slave  pointer  (2)]
⎜   ↳ USB Gaming Mouse Consumer Control       	id=11	[slave  pointer  (2)]
⎜   ↳ USB Keyboard Consumer Control           	id=13	[slave  pointer  (2)]
⎜   ↳ HID 256c:006d stylus                    	id=16	[slave  pointer  (2)]
⎜   ↳ HID 256c:006d eraser                    	id=20	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ USB Gaming Mouse Keyboard               	id=9	[slave  keyboard (3)]
    ↳ USB Gaming Mouse System Control         	id=10	[slave  keyboard (3)]
    ↳ USB Keyboard                            	id=12	[slave  keyboard (3)]
    ↳ USB Keyboard System Control             	id=14	[slave  keyboard (3)]
    ↳ USB Keyboard                            	id=15	[slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                      	id=17	[slave  keyboard (3)]
    ↳ USB Gaming Mouse Consumer Control       	id=18	[slave  keyboard (3)]
    ↳ USB Keyboard Consumer Control           	id=19	[slave  keyboard (3)]
```
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
DisplayPort-0 disconnected (normal left inverted right x axis y axis)
HDMI-A-0 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 521mm x 293mm
   1920x1080     60.00 +  71.91*   50.00    59.94  
   1680x1050     59.88  
   1600x900      60.00  
   1280x1024     60.02  
   1440x900      59.90  
   1280x800      59.91  
   1280x720      60.00    50.00    59.94  
   1024x768      70.07    60.00  
   800x600       72.19    60.32    56.25  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       72.81    66.67    60.00    59.94  
   720x400       70.08  
DVI-D-0 disconnected (normal left inverted right x axis y axis)
```

DraconYale, 2019-12-25:
EDIT: I solved the second problem: the tablet was simply in "phone mode" and to switch to "computer mode" you need to press the first and the fifth frame buttons. Don't know why it worked only now.

DraconYale, 2019-12-26:
I solved the problem installing the development version instead of the v9 of the driver.
You will need `50-tablet.conf` in `/etc/X11/xorg.conf.d` anyway though.
Now `xsetwacom set` works and I can customize my tablet's frame buttons.

closed 2019-12-26 by DraconYale

sbstratos79, 2019-12-26:
Glad to know your tablet's working now. I did suggest to install the latest development version. XD

DraconYale, 2019-12-26:
Sorry I forgot to inform you about it. Thank you for the help anyway, I thought it was a problem on my side and not of the driver's version!