DIGImend/digimend-kernel-drivers #335: Huion H610Pro V2

emccollu, 2019-12-26:
I just got one for christmas, plugged it in and the light came on when I moved the pen but nothing happened, same after installing this, finally got it working after reading a million methods. 
If it helps anyone, here's what might work if you have the same issue.
Plug in the tablet, open a terminal, type lsusb.
One of the lines should say Bus and a number, Device and a number, then either "ID 256c:006e" or "ID 256c:006d"
If its 256c:006e, then that's not the issue I had.
But if it's 256c:006d, the rest of this might work for you. Here's the steps that got mine working;
- Unplug the tablet
- Uninstall any digimend version you might have installed already, reboot.
- Go to the releases pages on this github, get the Source Code for version 8, zip or tar file.
- Go to the file in your computer and right click on it, choose extract here
- Go to the extracted folder and find the hid-ids.h file, it should open in any text editor
- On Line 9 it says: "#define USB_DEVICE_ID_HUION_TABLET 0x006e"
- Change that e to a d at the end, save it.
- Go back to the folder you extracted, right click on it and choose open in terminal
- Enter "make clean", then "make", then "sudo make install" and Reboot when it's done.
- Plug the tablet back in.

After that, hopefully it's working for you. Apparently they just changed that one letter in huion products. If it doesn't work, get a wacom like I'm doing next time cause I'm sick of supporting companies that can't be bothered to supply linux drivers.

rpavlik, 2020-02-13:
Looks like the workaround you found is already in the master branch, hopefully there's a new release soon.

Did you get this message in journalctl/dmesg? I'm seeing it with my H610pro v2, not sure if it is just noise or actually indicates an issue.

```
Feb 13 13:59:31 ryzenshine hid-rebind[22547]: rebinding /devices/pci0000:00/0000:00:07.1/0000:0b:00.3/usb3/3-1/3-1.4/3-1.4.3/3-1.4.3:1.1/0003:256C:006D.001C to "uclogic" driver
Feb 13 13:59:31 ryzenshine kernel: uclogic 0003:256C:006D.001C: interface is invalid, ignoring
```

I'm on Debian Buster with the backports kernel 5.4.something.

Veek, 2021-11-03:
I'm using a Gaomon S56K - same message you guys are getting

> [ 2367.308180] uclogic 0003:256C:006E.000D: interface is invalid, ignoring
> [ 2367.345597] uclogic 0003:256C:006E.000D: interface is invalid, ignoring
> [ 2411.161869] hid_uclogic:uclogic_probe:224: uclogic 0003:256C:006E.000E: parameters:
> [ 2411.161874] hid_uclogic:uclogic_params_hid_dbg:116: uclogic 0003:256C:006E.000E: .invalid = true
> [ 2411.161877] hid_uclogic:uclogic_params_hid_dbg:118: uclogic 0003:256C:006E.000E: .desc_ptr = 0000000000000000
> [ 2411.161879] hid_uclogic:uclogic_params_hid_dbg

JoseExposito, 2024-05-03:
Hi!

Closing this issue as the original report (product ID 0x006d) was fixed a while ago.

About @Veek's issue, it should be fixed by [DIGImend v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13). Please, if you have a chance, give it a try and report any problems found in a new bug report.

Thanks!

closed 2024-05-03 by JoseExposito