DIGImend/digimend-kernel-drivers #338: Huion Q11K V2 on Ubuntu 16.04

nexar, 2020-01-03:
I've taken delivery of a Huion Q11K v2 and just plugged it in to Ubuntu 16.04 and it is not being recognised at all. i.e. Can't move cursor with pen.

I've been reading the comments on : 

Huion Q11K not detected #76

but have not tried anything from there because all of that seems to relate to Q11K v1.  

I'm attaching an lsusb -v -s 002:006 (ID 256C:006e) and uclogic-probe outputs.

I've left a message with konachan700 asking whether his Q11K v1 driver can be updated for Q11K v2.  

However if possible I would like to use a digimend set of drivers for ongoing support.  On issue #76 towards the end of the comments there is a mention that kernel 5+ has built in support for the Q11K v1.  However there is also another comment saying that it did not work for that user.  I'm currently unable to move to a kernel 5 version of Ubuntu because I need to support older systems.

Although I have been using Ubuntu for a while now I'm not a sysdev but am comfortable following instructions.  Please help / advise if possible.

Many thanks in advance.

[lsusb.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4018914/lsusb.txt)
[uclogic.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4018915/uclogic.txt)

Fatih20, 2020-01-14:
I don't think supporting an OS version that only has 1 year left in support is the priority of the maintainer, not gonna lie.

spbnick, 2020-01-14:
FWIW, Huion [sent](https://www.patreon.com/posts/resuming-work-33141466) me a sample of Q11K v2 yesterday and it might arrive this month. Afterwards I should be able to implement support for it, if I have time and/or Huion pays.

The digimend-kernel-drivers should support any kernel starting with v3.5 and one of the aims of the package is to support older kernels (within reason). Ubuntu 16.04 qualifies for that very well.

Fatih20, 2020-01-14:
@spbnick Huion just released a slew of newer tablets, are you not going to fix that first?

spbnick, 2020-01-14:
I see a lot of requests for HS610 and Q11K V2 at the moment, so they're a priority. Otherwise I'll probably work on the other models they sent me (see link above) next.

Fatih20, 2020-01-14:
@spbnick You get paid by Huion? Shame that they don't put Linux support as official.

spbnick, 2020-01-14:
@Fatih20, sometimes. They had a plan to either link to digimend-kernel-drivers, or to host a download. Not sure what came of it. Otherwise, Linux users is not a big market, but they're doing the most for it, after Wacom.

nexar, 2020-01-15:
@spbnick Thanks very much for responding.  I have currently returned my tablet :-( but will keep a lookout for it on the Tablets page to see when it's supported.  In the mean time where can I read up on how to write drivers for these things.  I'm obviously completely new at this but just interested in understanding what exactly is involved.

spbnick, 2020-01-15:
I don't have a good write-up, but you could see my [talk on generic graphics tablets in Linux](https://archive.fosdem.org/2018/schedule/event/graphic_tablets/), and see [some live streams, where I do it](https://www.youtube.com/channel/UCioG-vCm2WB6nSz0Y0JTXoA/videos).

nexar, 2020-01-15:
Thanks @spbnick

spbnick, 2020-01-15:
Also check out some user-space tablet drivers floating around in the issues here. Those are a good-enough way to do it and is much easier to approach.

nexar, 2020-01-18:
@spbnick I've sent you an email.

lepinko, 2020-02-08:
Hello, I would like to add my voice to the Q11K v2 chorus. Thank you for your efforts!

spbnick, 2020-03-15:
I've got a Q11K v2 from Huion, tested it, and it works with latest master. You can download the [zip archive of current development version](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip). The support will be included in the next release.

closed 2020-03-15 by spbnick

nexar, 2020-06-12:
Hi I'm using the Q11K V2 with Ubuntu 16.04 and Krita.  The tablet is recognised along with the pressure levels.  

I want to know how I can change / set the buttons on both the tablet and the pen.

I connected the tablet to a Windows PC and used Huion's config program to change the values of the buttons however when I restart in Ubuntu the changed values are not recognised.

Thanks for your continued help.

valent1draw, 2021-01-25:
May be a solution! https://github.com/DIGImend/digimend-kernel-drivers/issues/510