DIGImend/digimend-kernel-drivers #340: XP-Pen Deco 01 V2 Support

sinsixin, 2020-01-06:
I have an XP-PEN Deco 01 V2 https://www.xp-pen.com/product/461.html on a desktop with Arch Linux kernel 5.4.8

Currently, it has an odd rotation on different boots. Drawing up makes the cursor go right. Drawing right makes the cursor go down. Down->Left. Left->Up. Unplugging then replugging the tablet makes sure the problem occurs, even if it was rotated correctly on boot. Pressure sensitivity and tilt work out of the box.

I've made a new issue because the USB Product ID seems to be different from the regular Art Deco 01.

Tablet Diagnostics:
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4028039/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4028040/hid_report_descriptors.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4028042/probe.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4028044/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4028045/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4028061/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4028062/pen_buttons.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4028063/frame_buttons.txt)

For now I've been having the xorg wacom driver handle mapping the tablet area and fixing rotation with an xsetwacom script.

noob8650, 2020-01-07:
I would like report an issue too: I am using Kernel 5.3.18_1 on Void GNU/Linux, for me too, the upper stylus button is not assignable under xf86-input-wacom.
I am using the newest (V9 I think) digimend kernel drivers, installed via dkms on one system and via a Debian package on Ubuntu.
The rotation is correct, and I can even set the Scaling/Proportions via the kcm-wacomtablet settings module.
What doesn't work however, are the frame buttons.
I can see them, and they register, when using the xf86-input-evdev driver, but under xf86-input-wacom they just won't show up.
Should I open a new issue for this?
As far as I understand xf86-input-wacom is the way to go, and

noob8650, 2020-01-18:
I guess this is redundant with sinxsin's files in a way, but here it is already ready to be integrated into the digimend.github.io remote repository, so I guess it's an improvement.
Here are the information files, pictures and diagnostics of the XP-Pen_Deco_01_v2_revb which has another vid:pid than the already  existing model of the same name.
[XP-Pen_Deco_01_v2_revb.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/4080413/XP-Pen_Deco_01_v2_revb.tar.gz)

I included a few comments in the files, I hope it doesn't disrupt anything, but as far as I could tell, when I got it running once in a local repo, it all displayed correctly.

jestarray, 2020-02-18:
thinking of purchasing this tablet, on linux im mainly just going to jot down math equations(maybe learn to draw later), but is this working? should i purchase something else? do the drivers on their site work?

jestarray, 2020-02-23:
alright so it works if you run and configure the official driver application: https://www.xp-pen.com/download-440.html
sadly it doesn't run on wayland(sway), only x11

ihewitt, 2020-04-06:
Just playing with one of these whilst in lockdown. Adding the device arrays into rdesc.c and mine's also working nicely with tilt even. 
Although at the moment also not having luck getting the top button to work.
I'll clean up my changes and submit it (I couldn't see an existing fork/pull request?)

ihewitt, 2020-04-07:
Simple in the end, just needed a mapping in /etc/udev/hwdb.d/10-xppen.hwdb
`evdev:input:b0003v28BDp0905*
 KEYBOARD_KEY_d0045=332
`
Not sure about frame buttons, with evdev mappings them to 0x100->0x107 they map to mouse buttons, shift<-, shift->, etc but don't seem re-mappable through the xwacom gui.

ihewitt, 2020-04-08:
Ok buttons and tips working ok with the combination of
[10-xppen.hwdb.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4451272/10-xppen.hwdb.txt) in /etc/udev/hwdb.d and 
[xp-pen-dec01.tablet.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4451273/xp-pen-dec01.tablet.txt) in /usr/share/libwacom

jestarray, 2020-04-11:
@ihewitt will your patch work make the tablet work on wayland?!

ihewitt, 2020-04-12:
If it simply requires a working kernel module rather than some sort of X hook interface then I'd assume so, but I have no experience of wayland personally.

MalcJPM, 2020-05-12:
Hi @ihewitt - great work with this!
Sorry to pick your brains, the pen works , but my frame buttons do not.

I have placed your `10-xppen.hwdb`in `/etc/udev/hwdb.d`and `xp-pen-dec01.tablet`in `/usr/share/libwacom`but I am still at a loss.

I believe I have enabled the Wacom driver by editing
`/etc/X11/xorg.conf.d/50-tablet.conf`

```
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "28bd:0905"
EndSection
```
`xsetwacom list` is successful, so I think that has worked
```
[mrm@mrmp50 ~]$ xsetwacom list
UGTABLET 10 inch PenTablet Mouse touch  id: 9   type: TOUCH     
UGTABLET 10 inch PenTablet Mouse pad    id: 10  type: PAD       
UGTABLET 10 inch PenTablet Keyboard pad id: 11  type: PAD       
UGTABLET 10 inch PenTablet stylus       id: 17  type: STYLUS   
```
Any pointers would be greatly appreciated - thanks!

ihewitt, 2020-05-13:
Hi, sorry a bit busy at the moment - just wanted to check, you're also running with either the version from https://github.com/ihewitt/digimend-kernel-drivers or with the changes from https://github.com/DIGImend/digimend-kernel-drivers/pull/375 applied?

MalcJPM, 2020-05-14:
@ihewitt - No problem!

You're right to ask that - I was tinkering yesterday, and realised my error there - so I uninstalled everything and installed your repo.

Just so I'm clear - do I need to have the `/etc/X11/xorg.conf.d/50-tablet.conf`file as I listed?

I've no replaced that, and the keys are working now in evtest... and the Wacom Tablet Finder now registers the button presses. However, the Wacom Graphic Tablet GUI no longer shows me the tab to alter their actions.

I see that you mention not being able to set through the gui - but I was unsure if this was fixed through your later edits?

I'm going to have a play with xsetwacom now - fingers crossed!

MalcJPM, 2020-05-14:
Right ho - I think I've got this, cheers for your work!
Not sure if this was a superfluous step, but I used the tablet finder GUI to map through the buttons?

I can set the frame buttons now through `xsetwacom`- not the GUI.

`xsetwacom list`shows:
```
[mrm@mrmp50 ~]$ xsetwacom list
UGTABLET 10 inch PenTablet Keyboard pad id: 9   type: PAD       
UGTABLET 10 inch PenTablet Pen stylus   id: 10  type: STYLUS    
[mrm@mrmp50 ~]$ 
```

For example I would use:
`xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" button 1 key ctrl z`
to set button 1 as ctrl-z

I'm just reading up on making these binds persistent through xorg.conf.

As I say, thank you for working on this. This arrangement is far more preferable to the XP-Pen linux driver.

ihewitt, 2020-05-14:
superb, yeah I'm not 100% sure if the button mappings should be different to work without the hwdb stuff, need an expert to chime in. :)
On my setup I'm just using the kde tablet config app, which just does that xsetwacom mapping under the hood on startup.

MalcJPM, 2020-05-14:
Learning as I go on this... I ended up using a script in `/etc/profile.d` to run the `xsetwacom`commands...

This lead to my mouse not responding, but it was persisting the bindings. Not sure what was going on there. I restarted for a second time and everything seemed to be correct.

Since the keybindings were working, I had a look at the GUI app and the Express Buttons are there now, and using them does result in a persistent change - hurrah!

However, what do you make of buttons 2 and 3 swapping? `xsetwacom` appears to get it correct - it is the bindings from the GUI that are swapped. It's easy enough to change the bindings, but thought you may have an idea!

[mrm@mrmp50 ~]$ xsetwacom -s get 14 all
Property 'Wacom Tablet Area' does not exist on device.
xsetwacom set "14" "Button" "1" "key +1 -1 "
**xsetwacom set "14" "Button" "2" "key +3 -3 "
xsetwacom set "14" "Button" "3" "key +2 -2 "**
xsetwacom set "14" "Button" "8" "key +4 -4 "
xsetwacom set "14" "Button" "9" "key +5 -5 "
xsetwacom set "14" "Button" "10" "key +6 -6 "
xsetwacom set "14" "Button" "11" "key +7 -7 "
xsetwacom set "14" "Button" "12" "key +8 -8 "

[image]

ihewitt, 2020-05-15:
ahh, with simply the following in xp-pen-dec01.tablet (ignoring previous tinkering I had):
[Buttons]
Left=A;B;C;D;E;F;G;H;
EvdevCodes=0x100;0x101;0x102;0x103;0x104;0x105;0x106;0x107

then its sequential:
xsetwacom -s get 9 all
Property 'Wacom Tablet Area' does not exist on device.
xsetwacom set "9" "Button" "1" "key +a -a "
xsetwacom set "9" "Button" "2" "key +b -b "
xsetwacom set "9" "Button" "3" "key +c -c "
xsetwacom set "9" "Button" "8" "key +d -d "
xsetwacom set "9" "Button" "9" "key +e -e "
xsetwacom set "9" "Button" "10" "key +f -f "
xsetwacom set "9" "Button" "11" "key +g -g "
xsetwacom set "9" "Button" "12" "key +h -h "

MalcJPM, 2020-05-21:
Stunner! Sorry - been in school mode (teacher), so no time for tinkering! That works a treat. Thank you!

cbm755, 2020-08-16:
@ihewitt @MalcJPM and others: many thanks for this, I've been using it to start to get my new Deco 01 v2 going.  My dept bought a crate of these and we have quite a few GNU/Linux users!

So far:
  * On Fedora 32 (OOTB on Wayland) I had no upper pen-button.  But the 8 buttons on the tablet did generate keycodes (e.g., in `xev`).
  * Switched to X11, installed @ihewitt's branch of DIGImend, and copied the `.hwdb` and `.tablet` files
       - now I have upper pen button working (great!)
       - 8 buttons don't seem to do anything

04ELY, 2020-11-18:
I have the product, xp-pen proprietary driver working but no aspect:ratio setting is avaliable and  Tablet stops working when I close program until I unplug-plug. In xsetwacom, aspect ratio is adjustable but upper pen key makes nothing and express keys not working as "Unsupported offset into 'Wacom Button Actions' property." 
here is files:
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5562066/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5562068/hid_report_descriptors.txt)
probe.txt is empty idk why
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5562071/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5562073/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5562074/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5562075/pen_buttons.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5562076/frame_buttons.txt)

WPettersson, 2020-11-25:
I've been playing with my Deco 01 V2, and the buttons. I was wondering, how is everyone (who is not using proprietary drivers) mapping the buttons to specific actions? Are you doing it per-program? Are you setting say the top button to "Ctrl+Z" for undo in the `10-xppen.hwdb` file? My current solution is to map the top 4 buttons to `prog{1,2,3,4}` and the lower four to `F1{3,4,5,6}` in there, and then map those to specific actions via (hopefully) some XFCE tool so I can customize it without having to re-run `system-hwdb update` and `udevadm trigger` all the time. Annoyingly, it seems udev only supports prog1 through prog4, so I couldn't just use those, but at the same time, F20 and above are also mapped to mic mute/speaker mute/volume adjustments, so I couldn't just use `F{12..20}` either.

colorizer, 2020-12-22:
> @ihewitt @MalcJPM and others: many thanks for this, I've been using it to start to get my new Deco 01 v2 going. My dept bought a crate of these and we have quite a few GNU/Linux users!
>
> So far: [...]

I'm also facing the same issue in Fedora. It used to work on Kubuntu .

WPettersson, 2020-12-22:
My current setup (with both stylus buttons working, and the 8 pad buttons mapped, is through the following:
* Use ihewitt's branch of digimend
* libwacom and xf86-input-wacom are *not* installed, instead I have libinput and xf86-input-libinput installed
* No changes to anything under /etc/X11/xorg.conf.d/
* [/etc/udev/hwdb.d/10-xppen.hwdb](https://gist.github.com/WPettersson/9cc0b1a44db7ee527fed38af401af128)

The 10-xppen.hwdb file maps the 8 buttons (top to bottom when buttons on right) to keycodes prog1, prog2, prog3, prog4, then f14, f15, f16, f17. These can then be remapped using xmodmap, or just used directly. I use XFCE4 hotkeys and xdotool to send ctrl+z on f17, and ctrl+y on f16, for easy undo and redo, and then use xmodmap to remap f14/f15 to up/down and prog3/prog4 to PgUp/PgDown.

BloodyMess, 2021-01-06:
> My current setup (with both stylus buttons working and the 8 pad buttons mapped) is through the following:
>
>     * Use ihewitt's branch of digimend
>
>     * libwacom and xf86-input-wacom are _not_ [...]

Are you using a release from ihewitt's fork of digimend? Or are you installing from the latest source?

WPettersson, 2021-01-06:
> 
> Are you using a release from ihewitt's fork of digimend? Or are you installing from the latest source in his fork?
     
I'm using the latest source, pulled from master.

colorizer, 2021-01-09:
I haven't used libinput previously. Is it possible to map the tablet to secondary screen in libinput?

WPettersson, 2021-01-09:
> I haven't used libinput previously. Is it possible to map the tablet to secondary screen in libinput?

Yes, for example, I use `xinput map-to-output "UGTABLET 10 inch PenTablet Pen Pen (0)"  "HDMI-0"` to map my tablet to only one of my monitors. The double `Pen` in the name is odd, and I think something to do with libinput, but I haven't really tried to narrow it down at all.

04ELY, 2021-01-30:
Hi, XP-PEN released a new beta driver. 
But install.sh script is a bit missing, you have to manually set utility(`/usr/lib/pentablet/pentablet`) as executable(sudo cmod +x _file_) and add sh to the beginning of Exec= line(`Exec=sh /usr/lib/pentablet/pentablet.sh`) in .desktop file(`/usr/share/applications/xppentablet.desktop`) to be shown in menu but it works a lot better than the old official one.
one-click aspect ratio fix in tablet settings is nice i think. And more importantly you dont have to keep the window open, it minimizes to tray.
But tilt is still not working and there is no application-specific settings on both official nor beta driver. (tilt is working with digimend kernel drivers)
And all "saved" settings lost when you unplug tablet or log off
And last, I don't know if this is related to DE but, mouse cursor is gone when i re-plug my tablet until i open applications menu or activities with super key on keyboard.

I mean, Can we take a look at this and get upper button on pen & express keys working?
And this was one of the little but important reasons that i cant switch to linux in 2019. And i was an absolute-beginner to linux at that times.
I switched in Q4 2020 lately and just ignoring this. But I want to help if I can, somehow.

liltingle, 2021-03-20:
Hello, I have problems when configuring with xsetwacom set "UGTABLET 10 inch PenTablet Pen stylus" button 3 button 2, as with other commands, including some for the buttons, the problem is that the pressing of each button is not continuous, for example in Blender when having the middle button pressed the camera should rotate, when I leave the default settings, it works normal, but when I change the settings there are failures, the truth is I don't know why, if the first time it worked normal, everything went out of control when I created this Bash, to automate the configuration. 

> #!/bin/bash
xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" Button 1 "key alt"
xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" Button 2 key "z"
xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" Button 3 key "ctrl c"
xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" Button 8 key "ctrl v"
xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" Button 9 "key backspace"
xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" Button 10 "key f"
xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" Button 11 "key Shift"
xsetwacom set "UGTABLET 10 inch PenTablet Keyboard pad" Button 12 "key ctrl"
xsetwacom set  "14"  Button 3  Button 2
xsetwacom set  "14"  Button 2  Button 3
echo 
exit 0

I think that maybe changing the keys directly from the 10-xppen.hwdb file might not cause problems, but I don't understand which key you mean by which.

> evdev:input:b0003v28BDp0905*
 KEYBOARD_KEY_d0045=332
 KEYBOARD_KEY_700e0=fn
 KEYBOARD_KEY_700e2=fn
 KEYBOARD_KEY_70015=256
 KEYBOARD_KEY_70018=257
 KEYBOARD_KEY_7001c=258
 KEYBOARD_KEY_7003c=259
 KEYBOARD_KEY_7002d=260
 KEYBOARD_KEY_70026=261
 KEYBOARD_KEY_70067=262
 KEYBOARD_KEY_70066=263

liltingle, 2021-03-20:
> Simple in the end, just needed a mapping in /etc/udev/hwdb.d/10-xppen.hwdb
> `evdev:input:b0003v28BDp0905*`
> `  #KEYBOARD_KEY_d0045=332`
>
> Not sure about frame buttons, with evdev mappings them to [...]

how do I reverse the middle click on 2 and the right click on 3?

> Simple in the end, just needed a mapping in /etc/udev/hwdb.d/10-xppen.hwdb
> `evdev:input:b0003v28BDp0905*`
> ` KEYBOARD_KEY_d0045=332`
>

liltingle, 2021-03-20:
hello, for my configuration I tried the simplest possible method, I made an excutable.sh and set it to start with the system.

> #!/bin/bash

xsetwacom --list
sleep 0.1
xsetwacom -s get 13 all
sleep 0.1
xsetwacom set "13" "Button" "1" "key alt"
sleep 0.1
xsetwacom set "13" "Button" "2" "key +z -z"
sleep 0.1
xsetwacom set "13" "Button" "3"  key "key ctrl c"
sleep 0.1
xsetwacom set "13" "Button" "8"  key "key ctrl v"
sleep 0.1
xsetwacom set "13" "Button" "9"  key "key +backspace  -backspace"
sleep 0.1
xsetwacom set "13" "Button" "10"  key "key +f -f"
sleep 0.1
xsetwacom set "13" "Button" "11"  key "key +shift -shift"
sleep 0.1
xsetwacom set "13" "Button" "11"  key "key shift"
sleep 0.1
xsetwacom set "13" "Button" "12"  key "key +ctrl -ctrl "
sleep 0.1
xsetwacom set "13" "Button" "12"  key "key ctrl "
sleep 0.1
xsetwacom -s get 14 all
sleep 0.1
xsetwacom set "14" "Button" "2" "button +3 "
sleep 0.1
xsetwacom set "14" "Button" "3" "button +2 "
exit

04ELY, 2021-05-03:
So what can we do right now? Other than collecting same diagnosis over and over again. I am confused.

Saegifu, 2021-09-04:
Need some help. Wacom drivers recognize the tablet, everything works fine as it looks like in GUI. However, when I change anything (pressure, express keys etc) - nothing changes in fact. It DOES change in GUI but has no impact on how tablet behaves. What do I do? I have the latest driver installed + .hwdb and .tablet files .

liltingle, 2021-09-08:
> Need some help. Wacom drivers recognize the tablet, everything works fine as it looks like in GUI. However, when I change anything (pressure, express keys etc) - nothing changes in fact. It DOES [...]

try the command `xsetwacom list`if it does not recognize anything, you must have configured the file /usr/share/X11/xorg.conf.d/50-digimend.conf 

>...
Section "InputClass"
        Identifier "Ugee/XP-Pen tablets with Wacom driver"
        MatchUSBID "28bd:0905[1458]|28bd:0094|28bd:0042|28bd:0905|5543:004[57]|>
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
...

Maximumtix, 2021-09-19:
Here are my raw input samples:
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7193066/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7193067/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7193068/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7193069/pen_buttons.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7193072/frame_buttons.txt)

nic3-14159, 2021-09-19:
For reference, these dumps posted above by Maximuntix can be reproduced by running the following script, which sends an "initialization key" to the tablet to switch it into a different mode.
https://github.com/nic3-14159/uclogic-new-init/blob/base/uclogic_new_init.py

foriequal0, 2021-09-24:
@nic3-14159
The official driver sends these three packets to endpoint 3 when the tablet is connected:
 - `02b00400000000000000`
 - `02b40100010000000000`
 - `02b40100020000000000`

tkalfaoglu, 2022-02-14:
Hello - new XP PEN Deco 01 V2 user here.. only installed the latest drivers from its site for Fedora 35..
It works - but the changes I make with the GUI are ignored.. For example I set the F3 to Page Down button, but it still just prints an 'i'  instead.  What shall I do?
Many thanks, -t

tkalfaoglu, 2022-02-14:
It WORKS!  Now that I installed the digimend driver, it works fine.. Off to configure the buttons now :)