DIGImend/digimend-kernel-drivers #342: make error on kernel 5.4.6

Seraaron, 2020-01-13:
Hi there I'm trying to set up my WH1409 tablet on Manjaro 5.4.6-2 but I get the error
```
make -C /lib/modules/5.4.6-2-MANJARO/build M=/home/seraaron/Downloads/digimend-kernel-drivers-master modules
make[1]: Entering directory '/usr/lib/modules/5.4.6-2-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.4.6-2-MANJARO/build'
make: *** [Makefile:22: modules] Error 2
```
when I `make`.

I'm new to both Git and Linux, but searching around I saw that this was a recent [issue](https://github.com/DIGImend/digimend-kernel-drivers/issues/326) that was 'fixed' by #331 and then undone by #339 with the comment "Stop using deprecated SUBDIRS make var".

I tried editing the Makefile on line 22 with all three variants:
`$(MAKE) -C $(KDIR) M=$(PWD) SUBDIRS=$(PWD) $@`,
`$(MAKE) -C $(KDIR) SUBDIRS=$(PWD) $@`, and `$(MAKE) -C $(KDIR) M=$(PWD) $@`, 
but all three threw basically the same error as above.

Any ideas?

This seems to be a reoccurring problem...

FraYoshi, 2020-01-14:
Are you using the latest git version?
It has already been fixed in `master` so you don't longer need to patch it..

spbnick, 2020-01-14:
@Seraaron, the #331 and #339 are not opposite. The former adds the `M` variable, and the latter removes the `SUBDIRS`. The first one makes build work with recent kernels, the second one prevents it from breaking with future kernels.

Seraaron, 2020-01-14:
> Are you using the latest git version?
> It has already been fixed in `master` so you don't longer need to patch it..

Yes, I just re-downloaded and tried again to make sure. I get the same error.

I'm following a [guide](https://askubuntu.com/a/854842) to do this. So before attempting to `make` I also ran the commands
```
sudo rmmod hid-kye
sudo rmmod hid-uclogic
sudo rmmod hid-huion
```
then I go to the extracted folder and run `make` and get
```
make -C /lib/modules/5.4.6-2-MANJARO/build M=/home/seraaron/Applications/digimend-kernel-drivers-master modules
make[1]: Entering directory '/usr/lib/modules/5.4.6-2-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.4.6-2-MANJARO/build'
make: *** [Makefile:22: modules] Error 2
```
as before.

Maybe are there some other dependencies that I'm missing / not aware of?

Or does this guide only work for the H610, whereas I'm using a WH1409?

spbnick, 2020-01-14:
@Seraaron, the `sudo rmmod ...` commands shouldn't affect this. I would suspect you're missing some build files. Do you have the kernel-modules package for your kernel version?

Seraaron, 2020-01-14:
Like I said, I'm new to Linux... How do I find this information?

When I look for kernel in the package manager I see that I have something
called 'kernel-alive' v0.4-1 and 'kernel-modules-hook' v0.1.7-1 installed.

bunianox, 2020-01-30:
search for linux54-headers

Seraaron, 2020-02-05:
This can probably be closed now. My old pc died, I built a new one, and I ended up switching to Solus and then just made it from source. Although now I have a new problem #348

closed 2020-02-05 by Seraaron