DIGImend/digimend-kernel-drivers #346: Huion H950P Tilt

CheezBallzPi, 2020-02-01:
Tablet tilt works on Windows but is not detected on Artix (5.4.14-artix1-1).

I have tested the tablet in Windows to make sure that there is no hardware issue (tilt works there).
I have used `usbhid-dump -es -m 256c:006d` and see change when tilting.
I cannot see any tilt information in evdev, libinput or in any applications.

Currently using the latest master digimend-kernel-drivers.

spbnick, 2020-03-22:
Hi @CheezBallzPi, let's check if you're actually using the downloaded driver and not the stock kernel driver. Please post the output of `cat /sys/module/hid_uclogic/version` with your tablet plugged in.

CheezBallzPi, 2020-03-22:
I don't have access to the computer due to the pandemic. Could I get back to you later?

spbnick, 2020-03-23:
@CheezBallzPi, sure!

Nikorasu, 2021-02-20:
Has there been any update on this?
I've got an HS611, tilt works in Windows, but I can't seem to get it working in Linux..
I'm currently using v10 release of DIGImend.