DIGImend/digimend-kernel-drivers #347: Have several tablets if anyone wants data

BruceZ2015, 2020-02-02:
I have been testing with 4 Huion tablets, 3 of which are unlisted on the website.

H1060P
H610 Pro v2
WH1409
WH1409 v2

I've been testing under kernel 5.3.0-29-generic from Ubuntu 19.10, and also 4.15.0-76-generic from Ubuntu 16.04 LTS.

Both the WH1409 and the WH1409 v2 work with with the version 9 driver in both kernels.  Buttons can be programmed on both the pad and the stylus.  I did not observe tilt functioning with v2 which supports it (expected?).  I did observe it functioning in Windows (after learning way more about custom dynamic brushes in GIMP than I wanted to know).  These both come up as 256c:006e.

Both H1060P and H610Pro v2 have no cursor in either kernel with version 9.  These both come up as 256c:006d, and I added this to the Huion conf file.  I can draw, and I can see a small dot in the drawing area indicating where the pen is.  However, outside that area there is no cursor even though it is possible to select things corresponding to where the cursor should be.  So it would seem the information about the cursor location is getting through, and the cursor is simply not being displayed.  Both of these tablets have wireless pens with both hard and soft buttons on the pad.

Let me know if there is any data you'd like me to collect that would be helpful to you, or if you have a driver that might work with the H1060P.  I'll have that tablet for small window of time before it goes back, and also the WH1490's.  The H610 Pro v2 has already gone back.

BruceZ2015, 2020-02-02:
One other interesting note about the H1060P:  Attempts to draw a circle result in an ellipse elongated in the x-direction since the cursor moves about twice as far horizontally as vertically.  The area is coming up 32767 32767.  When I double the x-value, I can make circles, but then the circle can't move past the halfway point in the x-direction, so it's only using half the area.  32767 maps the tablet to the screen correctly in the y-direction.

Seraaron, 2020-02-06:
How did you configure your WH1409? Can you help me with issue #348

BruceZ2015, 2020-02-06:
I made a file called "tablet" in my home directory with these commands:

xsetwacom set "HID 256c:006e Pad pad" button 1 key Shift E
xsetwacom set "HID 256c:006e Pad pad" button 2 key Ctrl Z
xsetwacom set "HID 256c:006e Pad pad" button 3 key N
xsetwacom set "HID 256c:006e Pad pad" button 8 key Shift
xsetwacom set "HID 256c:006e Pad pad" button 9 key R
xsetwacom set "HID 256c:006e Pad pad" button 10 key E
xsetwacom set "HID 256c:006e Pad pad" button 11 key F
xsetwacom set "HID 256c:006e Pad pad" button 12 key Shift +Ctrl A
xsetwacom set "HID 256c:006e Pad pad" button 13 key Ctrl ,
xsetwacom set "HID 256c:006e Pad pad" button 14 key Ctrl .
xsetwacom set "HID 256c:006e Pad pad" button 15 key Ctrl B
xsetwacom set "HID 256c:006e Pad pad" button 16 key Delete
xsetwacom set "HID 256c:006e Pen stylus" button 2 button +3
xsetwacom set "HID 256c:006e Pen stylus" button 3 button +3

Those device names came from running

xsetwacom list devices

for which the response was

HID 256c:006e Pen stylus         id: 9 type: STYLUS
HID 256c:006e Pad pad           id: 10 type: PAD

I'm making both pen buttons the right mouse button to make sure I don't
miss it since these buttons are hard to feel.

Then I just made that file executable with

chmod +x tablet

and I run it from my home directory as

. ./tablet

each time I boot up (it can be done automatically).  The first dot runs it
in the current shell, and I wouldn't need that if I added a line to the
beginning of the script like

#!/bin/sh

The ./ is only needed because my home directory isn't in my path (it
usually isn't by default on unix systems).  Of course you can also provide
the full path when you call it.

The only thing I had to to the driver files for this tablet was comment out
the line:

NoMatchProduct "MousePen"

in /usr/share/X11/xorg.conf.d/50-digimend.conf

If I don't do that, I can't boot X windows in ubuntu 16.04 LTS with kernel
4.15.0.-76-generic, but it doesn't seem to bother 19.4 with kernel
5.3.0-29-generic for some reason.  Maybe in your OS it boots but ignores
the driver?

If you do

lsusb

You should see a line for your device like:

Bus 003 Device 009: ID 256c:006e

Also, if you do

dpkg --list

you should see a line for the digimend driver listed if it is installed.

jhnsf, 2020-04-16:
hey people!
im sorry if this is the wrong place to ask this question but, im switching from win10 to ubuntu and im quite a newbie in all of this. i have the WH1409 v2 mentioned above. Will i be able to use it with any of these drivers? Its really holding me back from switching Os'
thanks!!