DIGImend/digimend-kernel-drivers #348: How to configure WH1409 on Solus? (and 'No tablet detected' in settings)

Seraaron, 2020-02-05:
Hi there, this is a sort of follow-on from issue #342. I have a WH1409 Huion. I now have the driver installed and it seems to be working in most apps, except that it doesn't appear in Solus's own 'Devices: Stylus / Tablet' settings. See https://i.imgur.com/YHdewR0.png

It also only seems to work when using the wireless dongle, it doesn't work when plugged in directly (though the little green light indicating when the pen is making contact _does_ flash on).

Finally, I've seen other users making use of an `xsetwacom` file to configure their shortcuts / hotkeys and the like (such as in this short video https://www.youtube.com/watch?v=1jHEGahVIoc ) but I can't seem to find this file. Is this because I'm using Solus or is it because I'm using a WH1409? 

The download and install instructions talk about making your own configuration in `/etc/X11/xorg.conf.d/50-tablet.conf`, but is this the right thing for me to do? Could some one talk me though it if so?

Thank you for you patience

Seraaron, 2020-02-09:
> @Seraaron I made a file called "tablet" in my home directory with these commands: xsetwacom set "HID 256c:006e Pad pad" button 1 key Shift E xsetwacom set "HID 256c:006e Pad pad" button 2 key Ctrl Z xsetwacom set "HID 256c:006e Pad pad" button 3 key N xsetwacom set "HID 256c:006e Pad pad" button 8 key Shift xsetwacom set "HID 256c:006e Pad pad" button 9 key R xsetwacom set "HID 256c:006e Pad pad" button 10 key E xsetwacom set "HID 256c:006e Pad pad" button 11 key F xsetwacom set "HID 256c:006e Pad pad" button 12 key Shift +Ctrl A xsetwacom set "HID 256c:006e Pad pad" button 13 key Ctrl , xsetwacom set "HID 256c:006e Pad pad" button 14 key Ctrl . xsetwacom set "HID 256c:006e Pad pad" button 15 key Ctrl B xsetwacom set "HID 256c:006e Pad pad" button 16 key Delete xsetwacom set "HID 256c:006e Pen stylus" button 2 button +3 xsetwacom set "HID 256c:006e Pen stylus" button 3 button +3 Those device names came from running xsetwacom list devices for which the response was HID 256c:006e Pen stylus id: 9 type: STYLUS HID 256c:006e Pad pad id: 10 type: PAD I'm making both pen buttons the right mouse button to make sure I don't miss it since these buttons are hard to feel. Then I just made that file executable with chmod +x tablet and I run it from my home directory as . ./tablet each time I boot up (it can be done automatically). The first dot runs it in the current shell, and I wouldn't need that if I added a line to the beginning of the script like #!/bin/sh The ./ is only needed because my home directory isn't in my path (it usually isn't by default on unix systems). Of course you can also provide the full path when you call it. The only thing I had to to the driver files for this tablet was comment out the line: NoMatchProduct "MousePen" in /usr/share/X11/xorg.conf.d/50-digimend.conf If I don't do that, I can't boot X windows in ubuntu 16.04 LTS with kernel 4.15.0.-76-generic, but it doesn't seem to bother 19.4 with kernel 5.3.0-29-generic for some reason. Maybe in your OS it boots but ignores the driver? If you do lsusb You should see a line for your device like: Bus 003 Device 009: ID 256c:006e Also, if you do dpkg --list you should see a line for the digimend driver listed if it is installed.
> […](#)

Thanks for you help @BruceZ2015 I'm replying here because the following more directly relates to my issues: I have a few follow-up questions / comments:
- My WH1409 only has 12 buttons and two on the side of the pen. Are there 4 more buttons / button combinations that I don't know about, or are you using a different tablet?
- When I do `xsetwacom list devices && echo ok` I just get `ok`.
- When I make the file you suggested, and do `chmod +x tablet`, and run `. ./tablet` I get the errors `Cannot find device 'HID 256c:006e Pad pad'.` and `Cannot find device 'HID 256c:006e Pen stylus'.`
- When I do `lsusb` I see `Bus 001 Device 004: ID 256c:006e` but it has no name attached to it (unlike, say, `Bus 005 Device 005: ID 047d:2048 Kensington Orbit Trackball with Scroll Ring`).
- And when I do `eopkg list-installed` I don't see digimend. Probably because I installed from source?

I'm pretty sure that installed digimend correctly because the pressure feels right and I can configure the pressure settings in individual apps, like Krita, but I can't find anything like `xsetwacom` in my files...

BruceZ2015, 2020-02-09:
I only have 12 buttons.  Buttons 4-7 are reserved for extra mouse buttons, so the pad buttons are numbered 1,2,3,8,9,...16.  You can read about xsetwacom [here](https://github.com/linuxwacom/xf86-input-wacom/wiki/xsetwacom) also some [here](https://github.com/DIGImend/digimend-kernel-drivers), and there is also a man page.  [This page](https://wiki.archlinux.org/index.php/Wacom_tablet) under troubleshooting has information about what to do if xsetwacom doesn't recognize your tablet or doesn't display it.  xsetwacom is part of fx-input-wacom and it is one of 3 parts to the [linux wacom project](https://linuxwacom.github.io/) in case it wasn't included in your distro.

The descriptor field for my tablet is also blank when I do  lsusb.  I'm just installing the v9 deb package.

If the above doesn't fix it, people in the ArchLinux forum can probably help you.

Seraaron, 2020-02-13:
For the life of me, I couldn't get this working in Solus. I was able to install `input-wacom` from source, but I couldn't get all of the dependencies for the `xf86-input-wacom` working. Maybe it should be possible, maybe i'm just bad at linux, but it looks like xwacom just relies on too many other bits and peices and the Solus devs have made it harder than usual for users to install pre-requisites like that... I don't know.

Either way, because I _need_ this tablet working for my job, I've hopped ships again. Back to Manjaro. I shame because I was enjoying my time with Solus. But getting this set up in Manjaro was waaay easier in the end. I checked my kernels properly this time, and installed all the pre-requisites through the AUR (thank goodness for the AUR)!

Then I found [this article](https://www.davidrevoy.com/article331/setup-huion-giano-wh1409-tablet-on-linux-mint-18-1-ubuntu-16-04), by David Revoy that helped me setup and configure everything. I've run into the same issue that he did though, where the bottom 3 express keys don't work. (Also, when I tried changing my `xsetwacom` config file to resemble yours @BruceZ2015, it would stop working again. I'm using the original WH1409 with 2000 levels of pressure. Are you using the 8K variant, or the WH1409 V2 perhaps, maybe that's why we have differences?)

Weirdly too, when I look in Manjaro's system settings, the tablet is recgonised as a gamepad instead of a tablet and when testing buttons in those settings it _does_ register those bottom 3 express keys; but I can't assign them to anything in the UI.

And, again, the tablet _only_ works with the wifi dongle, it doesn't work when it's plugged in directly by USB (which I'm sure must be an issue with the driver because it works fine in Windows).

So my problem is partly solved now. At least the tablet is usable, and I don't use the express keys that much anyways; but there's definately still some peculiarities.

BruceZ2015, 2020-02-17:
@Seraaron  I'm using the WH1409 Inspiroy with 8192 levels of pressure v1 and also v2, but not the Giano with 2048 levels.  Inspiroy v1 is in the [list of supported tablets](https://digimend.github.io/tablets/).   When you switch between wi-fi and cable, the device name changes, so the script has to change.  To make it work both ways, I replaced these 2 lines from the script in your article:

**tabletstylus="TABLET Pen Tablet Pen stylus"
tabletpad="TABLET Pen Tablet Pad pad"**

with

**tabletstylus=$(xsetwacom list devices | grep STYLUS | awk -F '   ' '{print $1}')
tabletpad=$(xsetwacom list devices | grep PAD | awk -F '   ' '{print $1}')**

No guarantee it will work with everything.  The article used different syntax than mine for xsetwacom, and that worked for me too, except strangely when I changed the Control_L and Shift_L to Control_Z and Shift_E it could no longer parse it.  I don't see the article's syntax when I do

**xsetwacom list modifiers**

but it could be in a .h file.  I was able to add 3 lines to that script to program the last 3 keys.  I didn't need to add anything to 50-huion.conf.

One thing I noticed is that if I program one key to be a pencil and another to be an eraser in GIMP, and I go back and forth between them while holding the pen close to the tablet, there would sometimes be an inordinate delay in changing, and sometimes it wouldn't change at all.  I also noticed this in Krita if I chose one key to be a brush (B) and another to be the color changer (P).  However, it seems to change as soon as I move the pen, which I normally would.  I used the tablet for months without noticing this, so it may not be much of a problem.  It doesn't seem to affect undo or clear screen.  It doesn't happen when the pen isn't over the tablet, and it doesn't happen when issuing the hotkeys from the keyboard.  I also don't see any problem with these keys registering when I'm in something that gives a text output identifying the keys pressed.  This doesn't have to be a driver problem.

As for Solus, when you issued the xsetwacom command, it complained that it couldn't find the device, but it didn't complain about not finding xsetwacom.  It sounds like the device isn't recognized, perhaps because the driver wasn't installed successfully.  Some distros apparently have to use the DKMS install described [here](https://github.com/DIGImend/digimend-kernel-drivers/tree/v9) under code and in the readme.  It also describes errors to watch for and what to do about them.  Then you should see that it is installed by doing dkms status.  Did it show the device when you did xinput --list?

Seraaron, 2020-02-19:
Replacing the `TABLET Pen` with the `grep` script didn't seem to work for wired mode. In fact it disabled wireless mode too, so I switched back to the original. Though now the stylus buttons don't work...

I got mod key combinations like undo to work with `"key Control_L +z"`. So maybe you could try `"key Shift_L +e"`?

I can't quite remember, but I think it didn't show anything when I did `xinput --list`.

BruceZ2015, 2020-02-19:
From a state where it's working, enter these commands:

`xsetwacom list devices`
`tabletstylus=$(xsetwacom list devices | grep STYLUS | awk -F ' ' '{print $1}')`
`tabletpad=$(xsetwacom list devices | grep PAD | awk -F ' ' '{print $1}')`
`echo $tabletstylus`
`echo $tabletpad`

We want the results from the echos to match the device names for the pad and stylus.  Your system may not be listing them as PAD and STYLUS or some other formatting difference, but we can get that working, that's easy.  When they match, the script will work.  The  name for my device in wireless mode is `TABLET Pen Tablet Pad pad `same as in the article, so that should be yours too if that script is working for you.  The name in wired mode is different.

You may need to restart your application when switching modes.  Plugging the dongle into the motherboard may work more consistently than into an unpowered hub.

`"key Control_L +Z"` does the same thing as `"key Control Z"`, so it is probably just ignoring the _L.  You should be able to use any syntax listed in `xsetwacom list modifiers`.  There are several variations that are accepted.  The double quotes aren't necessary for me.

Seraaron, 2020-02-19:
When I list devices I get 
```
TABLET Pen Tablet Pen stylus            id: 11  type: STYLUS    
TABLET Pen Tablet Pad pad               id: 12  type: PAD
```
but when I do the two echoes I only get `TABLET` and `TABLET`. And when I run the script I get the error 'unable to find device TABLET Pen Tablet Mouse'. I believe this is referring to line 68 in Revoy's code: `xinput set-prop 'TABLET Pen Tablet Mouse' "Evdev Middle Button Emulation" 0`. Maybe that should be changed to `Pen Tablet Stylus`?

When I unplug the wireless dongle and plug the tablet directly to my motherboard and then do `xsetwacom list devices` I get nothing. I've tried both with the tablet turned on and off (when I used to use Windows the wired mode only worked when it was turned off, bizarrely).

BruceZ2015, 2020-02-19:
> When I list devices I get
> 
> ```
> TABLET Pen Tablet Pen stylus            id: 11  type: STYLUS    
> TABLET Pen Tablet Pad pad               id: 12  type: PAD
> ```
> 
> but when I do the two echoes I only get `TABLET` and `TABLET`. 

Did you copy and paste those lines?  There has to be 3 spaces between first 2 single quotes after the `awk -F`.   That says to use  3 spaces as field delimiter, so $1 should be everything before the first 3 spaces which is the whole device name.  It's using just a single space for some reason. 

>
>And when I run the script I get the error 'unable to find device TABLET Pen Tablet Mouse'. I believe this is referring to line 68 in Revoy's code: `xinput set-prop 'TABLET Pen Tablet Mouse' "Evdev Middle Button Emulation" 0`. Maybe that should be changed to `Pen Tablet Stylus`?
> 

There doesn't seem to be a device by that name, so perhaps he's using a different pen.  I'm not doing that.

> When I unplug the wireless dongle and plug the tablet directly to my motherboard and then do `xsetwacom list devices` I get nothing. I've tried both with the tablet turned on and off (when I used to use Windows the wired mode only worked when it was turned off, bizarrely).
> 

The digimend project exists because these  tablets don't follow the USB standard correctly, so digimend supplies the correct parameters and ignores what the tablet is reporting.  Perhaps turning it off does something similar idk.  At some point you may want the information in [this article](http://www.volnapc.com/all-posts/how-to-get-rid-of-those-annoying-circles-from-your-wacom-cursor-in-windows-7) if you are using Windows.  It seems some versions see it as a touch screen.

> > I was able to add 3 lines to that script to program the last 3 keys. I didn't need to add anything to a conf file.
> 
> So how did you do this exactly?

I just added 3 lines 

`xsetwacom --set "$tabletpad" Button 14 key Delete`
`xsetwacom --set "$tabletpad" Button 15 key Delete`
`xsetwacom --set "$tabletpad" Button 16 key Delete`

Actually the only lines from that script I need are the xsetwacom commands that program the keys, and the lines that set `tabletstylus` and `tabletpen`, and I commented everything else out as nothing else is necessary for me. The part that remaps the area wouldn't work for me because I'm using a 1920x1200 which is the default rather than a 1920x1080.  Running the whole thing seems to screw things up.  Even my keyboard stopped working right.

BruceZ2015, 2020-02-19:
Actually, just copying and pasting the lines doesn't work - it removes the extra spaces.  I got the same thing you did.  You need to put back those spaces, and then it should work.

BruceZ2015, 2020-02-20:
>When I unplug the wireless dongle and plug the tablet directly to my motherboard and then do xsetwacom list devices I get nothing.

Wait a short time and try it again.  I've seen there be a delay with that showing up.  If it stays like that after say 30 seconds, then maybe you have a bad tablet.

>(when I used to use Windows the wired mode only worked when it was turned off, bizarrely).

It looks like turning it off only turns off wireless mode.  I can turn it off in wired mode and keep drawing and hotkeys still work.  In wireless mode if I turn it off, everything stops.  I can also connect both modes at once, and then disconnect either one, and it keeps drawing.   With both connected, `xsetwacom list devices` shows both devices registered at once  (my awk script fails in that case because it tries to run the 2 device names together).  However, it seems to only use the wired connection when both are connected because when I connect wired to Linux and wireless to Windows, only the wired one works, but pulling the wired one causes the wireless one to start working.  Then plugging the wired one back in causes the wired to start working and the wireless to stop working. 

I put the script I'm using now [here](https://drive.google.com/open?id=1TH2b2aXt0hQw9P-F7jTRnGONHG0a3Ome).    It has the awk command to get the device name in either wired or wireless mode (but not both at once), and then it programs the hotkeys.  Save that anywhere you want, go to where it is, make it executable (`chmod +x`), and run it with `. ./filename`.  Change hotkeys as needed.  `echo $pad` and `echo $pen` should give the same device names as `xsetwacom list devices`.

Seraaron, 2020-02-20:
Thanks, I'll give this a go tomorrow and report back.

Seraaron, 2020-02-21:
So I ran your script and tried the other things you suggested, and it worked for wireless mode and all the extra express keys work now, but it still doesn't work wired. 
Something _is_ being detected: I look in USB devices in system settings and see `PenTablet  Pad (/dev/input/js0)` listed under Game Controllers. But I cannot configure it, and the pen doesn't work, and it says nothing is detected under graphics tablet. 
When I click for more details I get the following:

Class | 0 | ((Defined at Interface level))
-- | -- | --
Subclass | 0 |  
Protocol | 0 |  
USB Version | 1.01 |  
  |   |  
Vendor ID | 0x256c |  
Product ID | 0x6e |  
Revision | 0.00 |  
  |   |  
Speed | 12 Mbit/s |  
Channels | 0 |  
Max. Packet Size | 8 | 

versus when I plug the wireless in I get:

Class | 0 | ((Defined at Interface level))
-- | -- | --
Subclass | 0 |  
Protocol | 0 |  
USB Version | 2.00 |  
  |   |  
Vendor ID | 0x256c |  
Product ID | 0x6e |  
Revision | 0.00 |  
  |   |  
Speed | 12 Mbit/s |  
Channels | 0 |  
Max. Packet Size | 32 |  

Maybe digimend doesn't know what to do with the wired mode yet (as there's no official support for the WH1409 Giano listed on the website)? Or maybe it's because it's USB version 1.01 (which seems very strange)!

I think I said this before too, but since trying to `grep` the tablet and pen name the buttons on my stylus don't work anymore. I tried re-running David Revoy's code again, but that didn't revert it... Not a huge issue, since I suppose I can map those two buttons to two of the newly gained express keys, but still weird. Maybe I should try reinstalling the driver?

Seraaron, 2020-02-21:
Sorry, accidentally closed the issue.

BruceZ2015, 2020-02-21:
> Something _is_ being detected: I look in USB devices in system settings and see `PenTablet Pad (/dev/input/js0)` listed under Game Controllers. But I cannot configure it, and the pen doesn't work, and it says nothing is detected under graphics tablet.

/dev/input/js0  doesn't correspond to the MatchDevicePath listed in /usr/share/X11/xorg.conf.d/50-digimend.conf as /dev/input/event*.  I think  this means it would be looking at the wrong file for input (each key and button has a separate file).  You might try changing the MatchDevicePath to /dev/input/js0 to see what happens.    Does nothing come up when you do `xsetwacom list devices`?  

> I think I said this before too, but since trying to `grep` the tablet and pen name the buttons on my stylus don't work anymore. I tried re-running David Revoy's code again, but that didn't revert [...]

All the grep and awk commands do is parse the output of the `xsetwacom list devices` command.  They didn't cause the stylus buttons to stop working.  The `xsetwacom list devices` part of that is at least related to these devices, so assuming you hadn't used it before without incident,  it's remotely plausible it could have caused X to do something weird, but it seems unlikely, and even more unlikely that it would do something that would survive a reboot.   If they stopped working, it's more likely something else would have caused it coincidental with using that command.  Perhaps interference. Lots of devices use that 2.4 GHz frequency including wi-fi.

My theory is that your wireless transmitter is interfering with your wired connection.  You said that even on Windows, the wired connection didn't start working until you turned off the switch.  On my tablet, that switch only disables wireless, and it doesn't disconnect the usb device - that is, it isn't the same as pulling out the usb dongle.   When I plug in the cable while wireless is running, the cabled connection takes over for drawing, and the switch has no effect.  Perhaps on yours the transmitter is still running and not well isolated from your wired signal.  Do you connect both wired and wireless at the same time, or only one at a time?

I wrote this [bash script](https://drive.google.com/open?id=1ImATn9_1Fbebo_x6uVqp0dqf7lTkqJf2) which allows you to connect both wired and wireless at the same time, and you can even set different hotkeys independently for each.  So if you have both connected and then disconnect the cable to go wireless, it brings in an alternate set of hotkeys.  However, if you then plug the cable back in, the hotkeys for the cable will be gone until you run the script again, so this can't be used to double the number of hotkeys you have at one time which was my initial goal.  Turning off wireless with the switch preserves the wireless hotkeys, so if there were an analagous swtich for wireless, it would effectively be an alt button to place 2 functions on each button.  Perhaps that's a new product suggestion. It might be interesting to see what this does in your case.  If you only connect one at a time, then only the first block of keys will matter.  It doesn't attempt to determine which is cabled and which is wireless, but it should do something consistently if you plug them in consistently.  You can even plug in multiple tablets, and it should program hotkeys for all of them if the reporting format is sufficiently similar.

If you do a search for Giano on this issues page and read through the discussions, it sounds like it was supported at one point, and then a change made to certain models which caused digimend to stop working, and then a change was made to digimend in an attempt to intelligently support different models.  The project leader was involved with this, and the issue was closed.  Perhaps you could try some earlier versions of digimend.  There is also some information there about things to look at it to debug it if you want to get into that.  Personally I would just get an Inspiroy.  Both v1 and v2 are working great.  v2 has the bateryless pen with the raised buttons which supports tilt, but AFAICT, only under Windows.  I prefer the v1 with the smooth pen that you have to charge.   The flush pen buttons are all but useless, but you can make up for that by using the pad keys as mouse buttons.  v1 may be getting harder to find.  I think I got one of the last ones off Amazon.

BruceZ2015, 2020-02-22:
> Edit: Also I decided to try one other thing: I switched out the cable for a different one. Still no joy... but interestingly this time it the table says
> Class 	0 	((Defined at Interface level))
> Subclass 	0 	 
> Protocol 	0 	 
> USB Version 	2.00 	 
>   	  	 
> Vendor ID 	0x5ac 	(Apple, Inc.)
> Product ID 	0x24f 	 
> Revision 	0.00 	 
>   	  	 
> Speed 	12 Mbit/s 	 
> Channels 	0 	 
> Max. Packet Size 	64

That appears to be an [Apple keyboard](https://usb-ids.gowdy.us/read/UD/05ac/024f).

Both my wired and wireless devices show USB 1.1 when I do `lsusb -v`.  The wired devices has a much higher max current at 480 mA vs. 50 mA for wireless.  The most a USB 2.0 port can provide is 500 mA.  You might try disconnecting as many USB devices as you can, or switch to a USB 3.0 port or powered hub if possible.  Laptops especially can have power issues.

Seraaron, 2020-02-29:
>Do you connect both wired and wireless at the same time, or only one at a time?
I've tried running your script, with every combination of cabled (yes / no) / wireless switch (on / off) / wireless dongle (in / out); and it _might_ have worked, but I can't quite tell. It got recognized as a tablet while plugged in, with the wifi turned off, but _only_ when the wifi dongle was also plugged in. Very strange. However I can't test it, because I think the pen still isn't being recognized. The green indicator light flashes on the tablet when I bring the pen near, but nothing is happening to the cursor on screen.

>That appears to be an Apple keyboard.
This is also strange, because that cable is actually from a Kodak camera. (Maybe they used the same manufacturer).

I also had go at plugging into usb 3.0 to no avail. I doubt it's a power issue, because I've tried several different cables now and I'm using a desktop with a 650W psu.

>Perhaps you could try some earlier versions of digimend.
To be honest, at this point I'm probably just going to give up. Thanks to your help, I've got it working enough that it's usable, and I'm kind of scared of messing it up again. As long as I remember to charge it daily I won't run into an issue with the battery life too often (and if I do use it long enough one day for it to run out of battery, that's probably a sign that I should take a break anyway). If the battery eventually dies completely then I might come back to the issue and try old digimend drivers; although at that point I might just get a new tablet anyway.

Thanks again for your help! I think this can be closed now, unless you have anything else to ask / try?

BruceZ2015, 2020-02-29:
> I've tried running your script, with every combination of cabled (yes / no) / wireless switch (on / off) / wireless dongle (in / out); and it _might_ have worked, but I can't quite tell. It got [...]
Very strange.

Did it recognize it as `TABLET Pen Tablet Pad pad` or `HID 256c:006e Pad pad`?    The first is wireless, and the second is wired.  It sounds it isn't switching from wireless to wired properly. That's in response to `xsetwacom list devices`.  The pen should have a separate name.

> 
> > That appears to be an Apple keyboard.
> 
> This is also strange, because that cable is actually from a Kodak camera. (Maybe they used the same manufacturer).

It looked like it was the report for a different device.  Do you have an Apple keyboard?

> 
> I also had go at plugging into usb 3.0 to no avail. I doubt it's a power issue, because I've tried several different cables now and I'm using a desktop with a 650W psu.

It wouldn't be a PSU issue, but a problem with the USB circuitry not providing enough current; however, if you are seeing it on more than one computer, then it's more likely a tablet problem.  The current it's requesting is very close to what the port is providing (and the most USB 2.0 can provide). If it needs more than that for some reason it can fail.  While USB 3.0 can provide more current,  it won't provide any more than USB 2.0 if that was all the device is requesting.  I only suggested trying other ports including 3.0, removing other devices, and trying powered ports in case the ports were an issue.  I thought power problem fit the best because it's the cleanest explanation for why it would with Windows wired only when you turn off the transmitter - it draws less current than in wireless mode.  There is no current being pulled by the tablet from the USB, only from the battery.   Power issues can cause a lot of intermittent unpredictable weirdness.

> To be honest, at this point I'm probably just going to give up. Thanks to your help, I've got it working enough that it's usable, and I'm kind of scared of messing it up again. As long as I remember [...]

Hopefully the battery charger works.