DIGImend/digimend-kernel-drivers #350: Problem with install .deb backage

tomwiet, 2020-02-11:
Hi
I try install drivers) to Huion 1060Plus on my Debian 11(bullseye) with kernel 5.4.0-3 and installation (both v8 and v9) ends with this error:
Loading new digimend-9 DKMS files...
Building for 5.4.0-3-amd64
Building initial module for 5.4.0-3-amd64
Error! Bad return status for module build on kernel: 5.4.0-3-amd64 (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.

make.log:
DKMS make.log for digimend-9 for kernel 5.4.0-3-amd64 (x86_64)
wto, 11 lut 2020, 16:18:53 CET
make -C /lib/modules/5.4.0-3-amd64/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[1]: Enter to directory '/usr/src/linux-headers-5.4.0-3-amd64'
sh: 0: Can't open /usr/src/linux-headers-5.4.0-3-common/scripts/mkmakefile
make[3]: *** [/usr/src/linux-headers-5.4.0-3-common/Makefile:513: outputmakefile] Błąd 127
make[3]: *** Waiting for unfinished tasks....
/usr/src/linux-headers-5.4.0-3-common/scripts/Makefile.build:42: /usr/src/linux-headers-5.4.0-3-common/scripts/basic/Makefile: There is no such file or directory
make[4]: *** There are no rules to execute the object '/usr/src/linux-headers-5.4.0-3-common/scripts/basic/Makefile'. Stop.
make[3]: *** [/usr/src/linux-headers-5.4.0-3-common/Makefile:499: scripts_basic] Error 2
make[2]: *** [/usr/src/linux-headers-5.4.0-3-common/Makefile:677: include/config/auto.conf.cmd] Error 2
make[1]: *** [/usr/src/linux-headers-5.4.0-3-common/Makefile:179: sub-make] Error 2
make[1]: Leaving the directory '/usr/src/linux-headers-5.4.0-3-amd64'
make: *** [Makefile:22: modules] Error 2

BruceZ2015, 2020-02-17:
Did you install the headers for your kernel?

**sudo apt-get install -y "linux-headers-$(uname -r)"**

sgrc31, 2020-03-27:
Same issue here. System info:
``Linux t420 5.4.0-4-amd64 #1 SMP Debian 5.4.19-1 (2020-02-13) x86_64 GNU/Linux``

Headers are already installed:
```
sgrc@t420:~/.debs_scaricati$ sudo aptitude install "linux-headers-$(uname -r)"
linux-headers-5.4.0-4-amd64 is already installed at the requested version (5.4.19-1)
linux-headers-5.4.0-4-amd64 is already installed at the requested version (5.4.19-1)
The following partially installed packages will be configured:
  digimend-dkms 
No packages will be installed, upgraded, or removed.
0 packages upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B of archives. After unpacking 0 B will be used.
Setting up digimend-dkms (9) ...         
Removing old digimend-9 DKMS files...

------------------------------
Deleting module version: 9
completely from the DKMS tree.
------------------------------
Done.
Loading new digimend-9 DKMS files...
Building for 5.4.0-4-amd64
Building initial module for 5.4.0-4-amd64
Error! Bad return status for module build on kernel: 5.4.0-4-amd64 (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: error processing package digimend-dkms (--configure):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
Setting up digimend-dkms (9) ...
Removing old digimend-9 DKMS files...

------------------------------
Deleting module version: 9
completely from the DKMS tree.
------------------------------
Done.
Loading new digimend-9 DKMS files...
Building for 5.4.0-4-amd64
Building initial module for 5.4.0-4-amd64
Error! Bad return status for module build on kernel: 5.4.0-4-amd64 (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: error processing package digimend-dkms (--configure):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms
```

And this is the failed make logfile:
```
DKMS make.log for digimend-9 for kernel 5.4.0-4-amd64 (x86_64)
Fri 27 Mar 2020 09:41:42 AM CET
make -C /lib/modules/5.4.0-4-amd64/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-4-amd64'
sh: 0: Can't open /usr/src/linux-headers-5.4.0-4-common/scripts/mkmakefile
make[3]: *** [/usr/src/linux-headers-5.4.0-4-common/Makefile:513: outputmakefile] Error 127
make[3]: *** Waiting for unfinished jobs....
/usr/src/linux-headers-5.4.0-4-common/scripts/Makefile.build:42: /usr/src/linux-headers-5.4.0-4-common/scripts/basic/Makefile: No such file or directory
make[4]: *** No rule to make target '/usr/src/linux-headers-5.4.0-4-common/scripts/basic/Makefile'.  Stop.
make[3]: *** [/usr/src/linux-headers-5.4.0-4-common/Makefile:499: scripts_basic] Error 2
make[2]: *** [/usr/src/linux-headers-5.4.0-4-common/Makefile:677: include/config/auto.conf.cmd] Error 2
make[1]: *** [/usr/src/linux-headers-5.4.0-4-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-4-amd64'
make: *** [Makefile:22: modules] Error 2
```

FutonRocks, 2020-04-29:
I had a similar problem. System Info:

`Linux damien-VirtualBox 5.4.0-28-generic #32-Ubuntu SMP Wed Apr 22 17:40:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

Headers are already installed:

`damien@damien-VirtualBox:~/Downloads$ sudo apt-get install -y "linux-headers-$(uname -r)"
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-5.4.0-28-generic is already the newest version (5.4.0-28.32).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up digimend-dkms (9) ...
Removing old digimend-9 DKMS files...

------------------------------
Deleting module version: 9
completely from the DKMS tree.
------------------------------
Done.
Loading new digimend-9 DKMS files...
Building for 5.4.0-28-generic
Building initial module for 5.4.0-28-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/digimend-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.4.0-28-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: error processing package digimend-dkms (--configure):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)`

(In the preview of this post "Deleting module version: 9 completely from the DKMS tree." is formated weird and I don't know how to change it.)

This is make.log:

`DKMS make.log for digimend-9 for kernel 5.4.0-28-generic (x86_64)
Wed 29 Apr 2020 03:56:39 PM EDT
make -C /lib/modules/5.4.0-28-generic/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
make[2]: *** No rule to make target 'arch/x86/tools/relocs_32.c', needed by 'arch/x86/tools/relocs_32.o'.  Stop.
make[1]: *** [arch/x86/Makefile:232: archscripts] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
make: *** [Makefile:22: modules] Error 2`

I am still new to using Linux so it's very possible that I'm doing something wrong.

thequailman, 2020-07-13:
Seeing this Debian buster with the backports linux-image kernel:

```
make -C /lib/modules/5.6.0-0.bpo.2-amd64/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.0-0.bpo.2-amd64'
sh: 0: Can't open /usr/src/linux-headers-5.6.0-0.bpo.2-common/scripts/mkmakefile
make[3]: *** [/usr/src/linux-headers-5.6.0-0.bpo.2-common/Makefile:514: outputmakefile] Error 127
make[3]: *** Waiting for unfinished jobs....
/usr/src/linux-headers-5.6.0-0.bpo.2-common/scripts/Makefile.build:44: /usr/src/linux-headers-5.6.0-0.bpo.2-common/scripts/basic/Makefile: No such file or directory
make[4]: *** No rule to make target '/usr/src/linux-headers-5.6.0-0.bpo.2-common/scripts/basic/Makefile'.  Stop.
make[3]: *** [/usr/src/linux-headers-5.6.0-0.bpo.2-common/Makefile:500: scripts_basic] Error 2
make[2]: *** [/usr/src/linux-headers-5.6.0-0.bpo.2-common/Makefile:677: include/config/auto.conf.cmd] Error 2
make[1]: *** [/usr/src/linux-headers-5.6.0-0.bpo.2-common/Makefile:180: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.6.0-0.bpo.2-amd64'
```

Lusitaniae, 2021-01-23:
me too, Linux Mint 20 (Ubuntu 20.04)

macthecadillac, 2021-02-10:
Same here. Debian Bullseye 5.10.0-3. Same error as OP

nic3-14159, 2021-02-11:
@macthecadillac @Lusitaniae Are you trying to install v9 or v10? v9 no longer builds on newer kernels due to changes, requiring a minor change in the Makefile. This has been fixed in v10.

macthecadillac, 2021-02-12:
I tried both. Same error whether I'm building the .deb or installing manually.

nic3-14159, 2021-02-12:
@macthecadillac What does your make.log contain?