DIGImend/digimend-kernel-drivers #351: Support for XP-Pen Artist Pro 12

Alex-PK, 2020-02-12:
First of all: thank you all for your work!

I just got an XP-Pen Artist Pro 12, and it's working with the standard drivers, but the buttons are not configurable.

```
$ lsusb 
Bus 001 Device 012: ID 28bd:1227 UGTABLET TABLET

$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech K270                           	id=9	[slave  pointer  (2)]
⎜   ↳ Logitech M185                           	id=10	[slave  pointer  (2)]
⎜   ↳ Logitech M525                           	id=11	[slave  pointer  (2)]
⎜   ↳ HOLTEK USB-HID Keyboard Mouse           	id=17	[slave  pointer  (2)]
⎜   ↳ HOLTEK USB-HID Keyboard Consumer Control	id=19	[slave  pointer  (2)]
⎜   ↳ UGTABLET TABLET Mouse                   	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ Dell Dell AC511 USB SoundBar            	id=12	[slave  keyboard (3)]
    ↳ HOLTEK USB-HID Keyboard                 	id=16	[slave  keyboard (3)]
    ↳ HOLTEK USB-HID Keyboard System Control  	id=18	[slave  keyboard (3)]
    ↳ HOLTEK USB-HID Keyboard                 	id=20	[slave  keyboard (3)]
    ↳ Logitech K270                           	id=21	[slave  keyboard (3)]
    ↳ HOLTEK USB-HID Keyboard Consumer Control	id=22	[slave  keyboard (3)]
    ↳ UGTABLET TABLET Keyboard                	id=13	[slave  keyboard (3)]
    ↳ UGTABLET TABLET                         	id=15	[slave  keyboard (3)]

$ ./uclogic-probe 1 2
M 55 00 47 00 54 00 41 00 42 00 4C 00 45 00 54 00
P 31 00 31 00 2E 00 36 00 20 00 69 00 6E 00 63 00 68 00 20 00 50 00 65 00 6E 00 44 00 69 00 73 00 70 00 6C 00 61 00 79 00
S 64 0E 03 22 64 4F 38 08 01 FF 1F EC 09 00 00
S 65 04 03 00 67
S 6E 0B 03 00 30 00 03 08 01 00 00 00
S 79 10 03 20 20 20 20 20 20 20 20 20 20 20 20 20 20
S 7A 08 03 00 00 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
Failed to get string descriptor 0xC8: Input/Output Error
```

Please let me know if I can help in any way (collecting more data, testing patches, ...)

alfazaz, 2020-05-14:
Same state as @Alex-PK ... How to make this graphical tablet all working ?

alfazaz, 2020-05-16:
Tablet diagnostic :

lsusb : 
Bus 001 Device 006: ID 28bd:091f UGTABLET 11.6 inch PenDisplay

T=28bd:091f

USB descriptors : 
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4638388/descriptors.txt)

[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4638393/hid_report_descriptors.txt)

uclogic-probe : (which ends with a "Failed to get string descriptor 0xC8: Input/Output Error" ???)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4638398/probe.txt)

I tried to adapt things with https://github.com/DIGImend/digimend-kernel-drivers/pull/291 but without any success...

I lack guidelines to make all this tablet all work... :-(

Korvox, 2020-05-16:
If you think this tablet behaves _identically_ to the 15.6 Pro you can just add the USB ids to hid-ids (28bd:1227) and a case fallthrough for it in `uclogic_params_init` in  hid-uclogic-params.c. Then the driver would treat it as an identical device. The patches I wrote query for the physical device info rather than have it hardcoded because I assumed other UGEE tablets would behave similarly.

alfazaz, 2020-05-17:
Thanks for your answer.

I will give it a try as soon as possible with details (to see at what step things go wrong if Murphys Laws are true ;-) ). 

Must I start with https://github.com/Korvox/digimend-kernel-drivers or https://github.com/underdoeg/digimend-kernel-drivers or another ?

alfazaz, 2020-05-17:
Hello @Korvox 

Here are my steps. If you have time to look if things are normal or stranges... Thanks !

1. Install clean kernel + kernel-headers (last archlinux version = 5.6.13-arch1-1)

2. Downloaded and extracted `digimend-kernel-drivers-master.zip` from https://github.com/underdoeg/digimend-kernel-drivers

3. Edited `hid-ids.h` : 

After
```
#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_A156P	0x090d
```

Added
```
#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12P	0x091f
```

_(it's not 0x1227 because lsusb shows 
Bus 001 Device 007: ID 28bd:091f UGTABLET 11.6 inch PenDisplay
and I think the 1227 comes after a failed uclogic-probe execution...)_

4. Edited `hid-uclogic-params.c` :

Changed
```
	case VID_PID(USB_VENDOR_ID_UGEE,
			USB_DEVICE_ID_UGEE_XPPEN_TABLET_A156P):
```
Into
```
	/* FALL THROUGH */
	case VID_PID(USB_VENDOR_ID_UGEE,
			USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12P):
	case VID_PID(USB_VENDOR_ID_UGEE,
			USB_DEVICE_ID_UGEE_XPPEN_TABLET_A156P):
```

5. Edited `hid-uclogic-core.c` (you didn't say this but it's useful no ?) :

After
```
	{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
				USB_DEVICE_ID_UGEE_XPPEN_TABLET_A156P) },
```
Added
```
	{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
				USB_DEVICE_ID_UGEE_XPPEN_TABLET_A12P) },
```
			
6. DKMS install with `sudo make dkms_install` which gives these messages :

```
dkms add .

Creating symlink /var/lib/dkms/digimend/10/source ->
                 /usr/src/digimend-10

DKMS: add completed.
dkms build digimend/10

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.6.13-arch1-1 KVERSION=5.6.13-arch1-1...
cleaning build area...
[... 238 lines cut ...]
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ DLL07BE:01 06CB:7A13 Mouse                id=13   [slave  pointer  (2)]
⎜   ↳ DLL07BE:01 06CB:7A13 Touchpad             id=14   [slave  pointer  (2)]
⎜   ↳ Logitech M325                             id=11   [slave  pointer  (2)]
⎜   ↳ PS/2 Synaptics TouchPad                   id=19   [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay Mouse       id=21   [slave  pointer  (2)]
⎜   ↳ UGTABLET 11.6 inch PenDisplay Pen (0)     id=23   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Power Button                              id=9    [slave  keyboard (3)]
    ↳ Sleep Button                              id=10   [slave  keyboard (3)]
    ↳ Intel HID events                          id=15   [slave  keyboard (3)]
    ↳ Intel HID 5 button array                  id=16   [slave  keyboard (3)]
    ↳ Dell WMI hotkeys                          id=17   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=18   [slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate           id=12   [slave  keyboard (3)]
    ↳ UGTABLET 11.6 inch PenDisplay Keyboard    id=20   [slave  keyboard (3)]
    ↳ UGTABLET 11.6 inch PenDisplay             id=22   [slave  keyboard (3)]
```

@Korvox : do you think that all is normal until there ? Steps 6-7. seem to be really strange... I will try to load wacom driver in Xorg after if nothing it's wrong... but I think there is a problem with steps 6-7.

Korvox, 2020-05-17:
Doing dkms install on Arch with the upstream sources names the modules differently (drivers/hid vs drivers/extra). Use the AUR package digimend-kernel-drivers-dkms-git and change its source to your local fork. It sounds like the make variant might not be installing depmod properly.

alfazaz, 2020-05-17:
Thanks !

There is the same problem with the AUR package and I think it comes form the content of dkms.conf.

I think that the following lines in dkms.conf are bad : 
```
DEST_MODULE_LOCATION[0]="/extra"
DEST_MODULE_LOCATION[1]="/extra"
DEST_MODULE_LOCATION[2]="/extra"
DEST_MODULE_LOCATION[3]="/extra"
```

I will make more tests when I will have time again... I cross my fingers...

nic3-14159, 2020-05-18:
I had a similar problem, and it seems that make dkms_install is not installing depmod.conf, which tells Linux to override the kernel modules in kernel/drivers/hid with the modules in extra/. I am also on Arch. I fixed this by editing the Makefile to install/uninstall depmod.conf.

From:
`dkms_install: dkms_modules_install udev_rules_install xorg_conf_install`
to
`dkms_install: depmod_conf_install dkms_modules_install udev_rules_install xorg_conf_install`
and
`dkms_uninstall: dkms_modules_uninstall udev_rules_uninstall xorg_conf_uninstall`
to
`dkms_uninstall: depmod_conf_uninstall dkms_modules_uninstall udev_rules_uninstall xorg_conf_uninstall`

alfazaz, 2020-05-18:
@nic3-14159 Ok. I tried changing dkms.conf and the modinfo is ok now but your way is cleaner may be. I will go on your way to see...

alfazaz, 2020-05-18:
Hello @Korvox 

Here are my new steps...

1. to 5. Like before.

Added a new thing before the 6. step (no need to do the two ways at the same time and I don't know exactly which is the greatest way but each ways is ok) :

* OLD way :
Edited `dkms.conf` so that dkms install works in a right way...

Changed
```
DEST_MODULE_LOCATION[0]="/extra"
DEST_MODULE_LOCATION[1]="/extra"
DEST_MODULE_LOCATION[2]="/extra"
DEST_MODULE_LOCATION[3]="/extra"
```

To
```
DEST_MODULE_LOCATION[0]="/kernel/drivers/hid"
DEST_MODULE_LOCATION[1]="/kernel/drivers/hid"
DEST_MODULE_LOCATION[2]="/kernel/drivers/hid"
DEST_MODULE_LOCATION[3]="/kernel/drivers/hid"
```

* NEW way :
Edited `Makefile` so that modules in extra/ override the same already in kernel/drivers/hid. 

Changed
```
dkms_install: dkms_modules_install udev_rules_install xorg_conf_install
	udevadm control --reload

dkms_uninstall: dkms_modules_uninstall udev_rules_uninstall xorg_conf_uninstall
	udevadm control --reload
```

To
```
dkms_install: depmod_conf_install dkms_modules_install udev_rules_install xorg_conf_install
	udevadm control --reload

dkms_uninstall: depmod_conf_uninstall dkms_modules_uninstall udev_rules_uninstall xorg_conf_uninstall
	udevadm control --reload
```

6. DKMS install with `sudo make dkms_install` (messages are normal)

7. and 8.  Strange things of my last try disappeared... Great !

`modinfo hid-uclogic` gives a good answer :

```
filename:       /lib/modules/5.6.13-arch1-1/kernel/drivers/hid/hid-uclogic.ko.xz
version:        10
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     0DEE6ED60839C655406E636
alias:          hid:b0003g*v000028BDp0000091F
alias:          hid:b0003g*v000028BDp0000090D
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000094
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000055
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
[... 248 lines cut ...]
mai 18 11:30:16 xps mtp-probe[2336]: checking bus 1, device 5: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
mai 18 11:30:16 xps mtp-probe[2336]: bus: 1, device: 5 was not an MTP device
mai 18 11:30:16 xps acpid[661]: input device has been disconnected, fd 15
mai 18 11:30:17 xps kernel: uclogic 0003:28BD:091F.0006: interface is invalid, ignoring
mai 18 11:30:17 xps kernel: uclogic 0003:28BD:091F.0007: interface is invalid, ignoring
mai 18 11:30:17 xps kernel: input: UGTABLET 11.6 inch PenDisplay Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:28BD:091F.0008/input/input39
mai 18 11:30:17 xps kernel: input: UGTABLET 11.6 inch PenDisplay Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:28BD:091F.0008/input/input40
mai 18 11:30:17 xps kernel: uclogic 0003:28BD:091F.0008: input,hidraw3: USB HID v1.00 Keypad [UGTABLET 11.6 inch PenDisplay] on usb-0000:00:14.0-2/input2
mai 18 11:30:17 xps hid-rebind[2421]: rebinding /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:28BD:091F.0006 to "uclogic" driver
mai 18 11:30:17 xps hid-rebind[2422]: rebinding /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:28BD:091F.0007 to "uclogic" driver
mai 18 11:30:17 xps kernel: uclogic 0003:28BD:091F.0006: interface is invalid, ignoring
mai 18 11:30:17 xps kernel: uclogic 0003:28BD:091F.0007: interface is invalid, ignoring
mai 18 11:30:17 xps mtp-probe[2436]: checking bus 1, device 5: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
mai 18 11:30:17 xps mtp-probe[2436]: bus: 1, device: 5 was not an MTP device
mai 18 11:32:21 xps kernel: usb 1-2: input irq status -75 received
mai 18 11:32:21 xps kernel: usb 1-2: input irq status -75 received
mai 18 11:32:21 xps kernel: usb 1-2: input irq status -75 received
mai 18 11:32:21 xps kernel: usb 1-2: input irq status -75 received
mai 18 11:32:21 xps kernel: usb 1-2: input irq status -75 received
mai 18 11:32:21 xps kernel: usb 1-2: input irq status -75 received
```

The last line with input irq status -75 repeated many many many times... It's made each I time I play with stylus or pad buttons... :-(

So, something is wrong... "Zut !" (as we say in french). If you have ideas to progress on this... Thanks !

alfazaz, 2020-05-18:
I don't know if it's useful but I read these rdesc in `/sys/kernel/debug/hid/` :

* For `0003:28BD:091F.001B` : 
```
05 01 09 02 a1 01 85 09 09 01 a1 00 05 09 19 01 29 03 15 00 25 01 95 03 75 01 81 02 95 05 81 01 05 01 09 30 09 31 26 ff 7f 95 02 75 10 81 02 05 0d 09 30 26 ff 07 95 01 75 10 81 02 c0 c0 05 01 09 02 a1 01 09 01 a1 00 85 01 05 09 19 01 29 05 95 05 75 01 15 00 25 01 81 02 95 03 81 01 05 01 09 30 09 31 95 02 75 10 16 00 80 26 ff 7f 81 06 09 38 15 81 25 7f 95 01 75 08 81 06 05 0c 0a 38 02 95 01 75 08 81 06 c0 c0 05 01 09 06 a1 01 85 06 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 05 07 19 00 29 ff 26 ff 00 75 08 95 06 81 00 c0
```

* For `0003:28BD:091F.001C` : 
```
05 0d 09 02 a1 01 85 07 09 20 a1 00 09 42 09 44 09 45 15 00 25 01 75 01 95 03 81 02 95 02 81 03 09 32 95 01 81 02 95 02 81 03 75 10 95 01 35 00 a4 05 01 09 30 65 13 55 0d 46 6c 27 26 ff 7f 81 02 09 31 46 2b 16 26 ff 7f 81 02 b4 09 30 45 00 26 ff 1f 81 42 09 3d 15 81 25 7f 75 08 95 01 81 02 09 3e 15 81 25 7f 81 02 c0 c0
```

* For `0003:28BD:091F.001D` : 
```
05 0d 09 02 a1 01 85 02 09 20 a0 14 25 01 09 42 09 44 09 46 75 01 95 03 81 02 95 02 81 01 09 32 95 01 81 02 95 02 81 01 75 10 95 01 a4 05 01 55 fd 65 13 34 09 30 27 22 64 00 00 47 6c 27 00 00 81 02 09 31 27 4f 38 00 00 47 2b 16 00 00 81 02 b4 09 30 27 ff 1f 00 00 81 02 a4 54 65 14 35 c3 45 3c 15 c3 25 3c 75 08 95 02 09 3d 09 3e 81 02 b4 c0 c0 05 01 09 07 a1 01 85 f7 05 0d 09 39 a0 14 25 01 75 01 95 08 81 01 05 09 19 01 29 08 95 08 81 02 95 20 81 01 19 09 29 0a 95 02 81 02 95 16 81 01 c0 c0 

  INPUT(2)[INPUT]
    Field(0)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(3)
        Digitizers.TipSwitch
        Digitizers.BarrelSwitch
        Digitizers.TabletPick
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(3)
      Report Offset(0)
      Flags( Variable Absolute )
    Field(1)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        Digitizers.InRange
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(1)
      Report Offset(5)
      Flags( Variable Absolute )
    Field(2)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        GenericDesktop.X
      Logical Minimum(0)
      Logical Maximum(25634)
      Physical Minimum(0)
      Physical Maximum(10092)
      Unit Exponent(-3)
      Unit(English Linear : Inch)
      Report Size(16)
      Report Count(1)
      Report Offset(8)
      Flags( Variable Absolute )
    Field(3)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        GenericDesktop.Y
      Logical Minimum(0)
      Logical Maximum(14415)
      Physical Minimum(0)
      Physical Maximum(5675)
      Unit Exponent(-3)
      Unit(English Linear : Inch)
      Report Size(16)
      Report Count(1)
      Report Offset(24)
      Flags( Variable Absolute )
    Field(4)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        Digitizers.TipPressure
      Logical Minimum(0)
      Logical Maximum(8191)
      Report Size(16)
      Report Count(1)
      Report Offset(40)
      Flags( Variable Absolute )
    Field(5)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(2)
        Digitizers.003d
        Digitizers.003e
      Logical Minimum(-61)
      Logical Maximum(60)
      Physical Minimum(-61)
      Physical Maximum(60)
      Unit(English Rotation : Degrees)
      Report Size(8)
      Report Count(2)
      Report Offset(56)
      Flags( Variable Absolute )
  INPUT(247)[INPUT]
    Field(0)
      Physical(Digitizers.TabletFunctionKey)
      Application(GenericDesktop.Keypad)
      Usage(8)
        Button.0001
        Button.0002
        Button.0003
        Button.0004
        Button.0005
        Button.0006
        Button.0007
        Button.0008
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(8)
      Report Offset(8)
      Flags( Variable Absolute )
    Field(1)
      Physical(Digitizers.TabletFunctionKey)
      Application(GenericDesktop.Keypad)
      Usage(2)
        Button.0009
        Button.000a
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(2)
      Report Offset(48)
      Flags( Variable Absolute )

Digitizers.TipSwitch ---> Key.Touch
Digitizers.BarrelSwitch ---> Key.Stylus
Digitizers.TabletPick ---> Key.Stylus2
Digitizers.InRange ---> Key.ToolPen
GenericDesktop.X ---> Absolute.X
GenericDesktop.Y ---> Absolute.Y
Digitizers.TipPressure ---> Absolute.Pressure
Digitizers.003d ---> Absolute.XTilt
Digitizers.003e ---> Absolute.YTilt
Button.0001 ---> Key.Btn0
Button.0002 ---> Key.Btn1
Button.0003 ---> Key.Btn2
Button.0004 ---> Key.Btn3
Button.0005 ---> Key.Btn4
Button.0006 ---> Key.Btn5
Button.0007 ---> Key.Btn6
Button.0008 ---> Key.Btn7
Button.0009 ---> Key.Btn8
Button.000a ---> Key.Btn9
```

alfazaz, 2020-05-19:
Tried to adapt uclogic_rdesc_xppen_a156p_pen_arr[] and uclogic_rdesc_xppen_a156p_frame_arr[] for the Artist 12 Pro using my rdesc but still getting `input irq status -75 received`...

I don't know how to proceed to debug this. :disappointed:

alfazaz, 2020-05-20:
Logs in journalctl in debug mode for hid_uclogic module without changing rdesc content (so using Artist 15.6 Pro rdesc for my Artist 12 Pro rdesc) and on connection : 

[logs.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4658565/logs.txt)

After this, only this concerning the tablet with xinput : 

```
 ↳ UGTABLET 11.6 inch PenDisplay Pen         id=20   [slave  keyboard (3)]
```

For evtest, only these : 
```
/dev/input/event22:     UGTABLET 11.6 inch PenDisplay Pen
/dev/input/event23:     UGTABLET 11.6 inch PenDisplay Pad
```

and nothing detected in events on 22 and 23 device... :-(

Note : I saw with wireshark the same thing written  in https://github.com/DIGImend/digimend-kernel-drivers/issues/286#issue-452139401 about init packet for example (same values...).

@Korvox If one day you have time to help (or to give guidelines to debug like you did for your tablet)... I really think that these tablets are close when I read your messages about 15.6 Pro...

alfazaz, 2020-05-22:
Another day on this tablet and still the hell... :-(

I don't understand where it goes wrong.

Using the Artist 15.6 Pro rdesc of Korvox, I get this with `sudo usbhid-dump -es -m 28bd:091f` at start and when approaching/touching/releasing the pen : 

```
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:012:000:STREAM             1590178740.879005
 02 B1 04 00 00 00 00 00

001:012:002:STREAM             1590178740.880577
 02 F0 00 00 00 00 00 00 00 00 00 00

001:012:002:STREAM             1590178745.704729
 02 A0 B0 20 1D 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.708610
 02 A0 B0 20 1D 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.714739
 02 A0 BA 20 2D 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.718741
 02 A0 C0 20 36 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.722708
 02 A0 D1 20 50 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.728607
 02 A0 DB 20 65 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.732704
 02 A0 EC 20 7E 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.736696
 02 A0 FD 20 97 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.742759
 02 A0 0A 21 AB 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.746601
 02 A0 1C 21 C5 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.750692
 02 A0 2D 21 E0 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.760654
 02 A1 3D 21 FC 14 8B 05 00 00 00 00

001:012:002:STREAM             1590178745.764587
 02 A1 3E 21 07 15 5B 09 00 00 00 00

001:012:002:STREAM             1590178745.770609
 02 A1 3C 21 0F 15 27 0B 00 00 00 00

001:012:002:STREAM             1590178745.774725
 02 A1 39 21 16 15 22 0C 00 00 00 00

001:012:002:STREAM             1590178745.778663
 02 A1 35 21 1C 15 9A 0C 00 00 00 00

001:012:002:STREAM             1590178745.784730
 02 A1 32 21 20 15 D3 0C 00 00 00 00

001:012:002:STREAM             1590178745.788616
 02 A1 30 21 24 15 ED 0C 00 00 00 00

001:012:002:STREAM             1590178745.792598
 02 A1 2E 21 28 15 F6 0C 00 00 00 00

001:012:002:STREAM             1590178745.798629
 02 A1 2C 21 2B 15 F6 0C 00 00 00 00

001:012:002:STREAM             1590178745.802550
 02 A1 2A 21 2E 15 EA 0C 00 00 00 00

001:012:002:STREAM             1590178745.806598
 02 A1 28 21 30 15 D7 0C 00 00 00 00

001:012:002:STREAM             1590178745.812717
 02 A1 28 21 30 15 B5 0C 00 00 00 00

001:012:002:STREAM             1590178745.816717
 02 A1 25 21 33 15 7F 0C 00 00 00 00

001:012:002:STREAM             1590178745.820723
 02 A1 25 21 33 15 32 0C 00 00 00 00

001:012:002:STREAM             1590178745.826734
 02 A1 23 21 37 15 C5 0B 00 00 00 00

001:012:002:STREAM             1590178745.830735
 02 A1 23 21 37 15 3B 0B 00 00 00 00

001:012:002:STREAM             1590178745.834609
 02 A1 23 21 37 15 92 0A 00 00 00 00

001:012:002:STREAM             1590178745.840735
 02 A1 23 21 37 15 B0 09 00 00 00 00

001:012:002:STREAM             1590178745.844632
 02 A1 29 21 3B 15 12 08 00 00 00 00

001:012:002:STREAM             1590178745.848668
 02 A1 30 21 3A 15 B9 06 00 00 00 00

001:012:002:STREAM             1590178745.854734
 02 A1 3C 21 37 15 99 05 00 00 00 00

001:012:002:STREAM             1590178745.858717
 02 A0 48 21 32 15 00 00 00 00 00 00

001:012:002:STREAM             1590178745.862606
 02 A0 63 21 23 15 00 00 00 00 00 00

001:012:002:STREAM             1590178745.868612
 02 A0 78 21 0D 15 00 00 00 00 00 00

001:012:002:STREAM             1590178745.872703
 02 A0 83 21 FA 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.876716
 02 A0 88 21 E7 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.882703
 02 A0 87 21 D3 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.886718
 02 A0 7C 21 AE 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.890600
 02 A0 6C 21 8A 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.894720
 02 A0 6C 21 8A 14 00 00 00 00 00 00

001:012:002:STREAM             1590178745.902711
 02 C0 6C 21 8A 14 00 00 00 00 00 00
```

but evtest doesn't react... It gives with Pen input no more things than the following : 

```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x28bd product 0x91f version 0x100
Input device name: "UGTABLET 11.6 inch PenDisplay Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    25634
      Resolution     100
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    14415
      Resolution     100
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8191
    Event code 26 (ABS_TILT_X)
      Value      0
      Min      -61
      Max       60
      Resolution      57
    Event code 27 (ABS_TILT_Y)
      Value      0
      Min      -61
      Max       60
      Resolution      57
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
```

How can I debug this ???

Korvox, 2020-05-25:
Might point you in the right direction to know that I figured out the a156p had an "activation" protocol via recording how the Windows driver handled init and by `uclogic-probe` triggering several special functions while iterating over the interface strings _including_ that parameters activation string. [Its all in this function](https://github.com/Korvox/digimend-kernel-drivers/blob/45e1d4dffe11acb46469eed04965d04998531141/hid-uclogic-params.c#L833).

From your report descriptors it looks like the 12 is nothing like the 15.6 - this interface rdesc from the a156p refers to the acitvation interface that advertises basically a garbage rdesc because it requires a special packet to switch the device mode:

`
002:003:002:DESCRIPTOR         1559668842.520668
 06 0A FF 09 01 A1 01 85 02 09 02 75 08 95 09 15
 00 26 FF 00 81 02 09 03 75 08 95 07 15 00 26 FF
 00 91 02 C0
`
Your rdescs are all pretty standard - an emulated mouse, a pen without pad buttons, and a pen with the pad buttons. That would imply you don't need any of what I was doing - which is good. It makes your job easier in theory. I had to write wholly custom rdecs but for an undescribed interface because the generic ones were broken. Your rdescs look like they describe a working pen and properly mapped buttons on `0003:28BD:691F.001D`. From your descriptions it doesn't work out of the box but it doesn't look like its broken in the same way the a156p was.

alfazaz, 2020-05-25:
Hello @Korvox

I don't know if I understand all the things correctly and if my latest posts were clear.

1. When **I don't use hid-uclogic at all**, my Artist 12 Pro tablet is driven by hid-generic and HID descriptors are these :

* On `0003:28BD:091F.0021` (first interface ; it seems to be mouse + keyboard when I read full infos given by the kernel with `sudo more /sys/kernel/debug/hid/0003:28BD:091F.0021/rdesc`) :
```
05 01 09 02 a1 01 85 09 09 01 a1 00 05 09 19 01 29 03 15 00 25 01 95 03 75 01 81 02 95 05 81 01 05 01 09 30 09 31 26 
ff 7f 95 02 75 10 81 02 05 0d 09 30 26 ff 07 95 01 75 10 81 02 c0 c0 05 01 09 02 a1 01 09 01 a1 00 85 01 05 09 19 01 
29 05 95 05 75 01 15 00 25 01 81 02 95 03 81 01 05 01 09 30 09 31 95 02 75 10 16 00 80 26 ff 7f 81 06 09 38 15 81 25 
7f 95 01 75 08 81 06 05 0c 0a 38 02 95 01 75 08 81 06 c0 c0 05 01 09 06 a1 01 85 06 05 07 19 e0 29 e7 15 00 25 01 75 
01 95 08 81 02 05 07 19 00 29 ff 26 ff 00 75 08 95 06 81 00 c0 
```

* On `0003:28BD:091F.0022` (second interface ; it's stylus and its buttons when I read full infos given by the kernel with `sudo more /sys/kernel/debug/hid/0003:28BD:091F.0022/rdesc`) :
```
05 0d 09 02 a1 01 85 07 09 20 a1 00 09 42 09 44 09 45 15 00 25 01 75 01 95 03 81 02 95 02 81 03 09 32 95 01 81 02 95 
02 81 03 75 10 95 01 35 00 a4 05 01 09 30 65 13 55 0d 46 6c 27 26 ff 7f 81 02 09 31 46 2b 16 26 ff 7f 81 02 b4 09 30 
45 00 26 ff 1f 81 42 09 3d 15 81 25 7f 75 08 95 01 81 02 09 3e 15 81 25 7f 81 02 c0 c0 
```

* On `0003:28BD:091F.0023` (third interface ; it's like the 15.6 Pro activation interface ?) :
```
06 0a ff 09 01 a1 01 85 02 09 02 75 08 95 0b 15 00 26 ff 00 81 02 09 03 75 08 95 0b 15 00 26 ff 00 91 02 c0 
```

With this situation, with hid-generic and without uc-logic, stylus position is detected but the upper button of the stylus doesn't work (seems to be interpreted as eraser button instead of middle mouse button) and the lower button is interpreted as middle mouse button (I preferred it to be the right mouse button). Pad/wheel buttons are not working and there is also the alignment problem mentionned in https://github.com/DIGImend/digimend-kernel-drivers/issues/286#issue-452139401 .

`sudo usbhid-dump -es -m 28bd:091f` gives messages with 10 bytes beginning with 07 on all stylus actions and with 06 on buttons/wheel pad actions :  [penactions.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4675898/penactions.txt) and 
[padbuttonswheelactions.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4675906/padbuttonswheelactions.txt)

2. If if play with windows driver in a VM + wireshark, I see like in https://github.com/DIGImend/digimend-kernel-drivers/issues/286#issue-452139401 activation on interface the latest interface.

3. Now, when **I use hid-uclogic modified** (adding my Artist 12 Pro tablet as indicated in https://github.com/DIGImend/digimend-kernel-drivers/issues/351#issuecomment-629719361), the tablet is driven by uclogic and :

* The third interface receive the activation signal well. I see in logs for this interface (the others are considered as invalid) : 
```
uclogic_params_init_ugee_xppen_a12p:1089: uclogic 0003:28BD:091F.0038: Received parameters: X: 25634 Y: 14415 Pressure: 8191 Resolution: 2540 
```
Wireshark shows also the activation.

* On `0003:28BD:091F.0036` (first interface) : `sudo more /sys/kernel/debug/hid/0003:28BD:091F.0036/rdesc` gives exactly the same first interface rdesc as in point 1. with hid-generic, but without the full infos part following it (???????).

* On `0003:28BD:091F.0037` (second interface) : `sudo more /sys/kernel/debug/hid/0003:28BD:091F.0037/rdesc` gives exactly the same second interface rdesc as in point 1. with hid-generic, but without the full infos part following it (???????).

* On `0003:28BD:091F.0038` (third interface) : `sudo more /sys/kernel/debug/hid/0003:28BD:091F.0038/rdesc` gives the new rdesc which comes from modified uc-logic code :
```
05 0d 09 02 a1 01 85 02 09 20 a0 14 25 01 09 42 09 44 09 46 75 01 95 03 81 02 95 02 81 01 09 32 95 01 81 02 95 02 81 
01 75 10 95 01 a4 05 01 55 fd 65 13 34 09 30 27 22 64 00 00 47 6c 27 00 00 81 02 09 31 27 4f 38 00 00 47 2b 16 00 00 
81 02 b4 09 30 27 ff 1f 00 00 81 02 a4 54 65 14 35 c3 45 3c 15 c3 25 3c 75 08 95 02 09 3d 09 3e 81 02 b4 c0 c0 05 01 
09 07 a1 01 85 f7 05 0d 09 39 a0 14 25 01 75 01 95 08 81 01 05 09 19 01 29 08 95 08 81 02 95 20 81 01 19 09 29 0a 95 
02 81 02 95 16 81 01 c0 c0 

  INPUT(2)[INPUT]
    Field(0)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(3)
        Digitizers.TipSwitch
        Digitizers.BarrelSwitch
        Digitizers.TabletPick
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(3)
      Report Offset(0)
      Flags( Variable Absolute )
    Field(1)
      Physical(Digitizers.Stylus)
[... 120 lines cut ...]

* If I do `sudo usbhid-dump -es -m 28bd:091f`, I get first :  
```
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:013:000:STREAM             1590394296.036374
 02 B1 04 00 00 00 00 00

001:013:002:STREAM             1590394296.038426
 02 F0 00 00 00 00 00 00 00 00 00 00
```
and then, following messages are all with 12 bytes beginning with 02 on all stylus/pad actions :  
[penactions2.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4676056/penactions2.txt)
[... cut ...]
* Why activation step was necessary on your tablet and why it wouldn't be necessary on mine ? I see this same activation with my windows VM+XP PEN drivers+wireshark.
* If I use activation, how to debug rdesc thing after and how to avoid the EOVERFLOW ?
* If activation is not necessary, what must I do to have good actions for stylus buttons, button/wheel pad working and also good alignment (taking tilt into account ; see https://github.com/DIGImend/digimend-kernel-drivers/issues/286#issue-452139401 ) ?

Thanks for help.

Korvox, 2020-05-30:
I didn't get that you were just showing interfaces post-activation, but yes, looks like it needs it - you are getting back the parameters packet which means it works.

The 02 leading byte on a usb packet is the interface identifier. Its meant for interface 2. If i0 and i1 are the mouse / keyboard / fallback interfaces then 02 is the one you activated. Those packets look comparable to the ones the a156p generates.

Those irq -75s indicate some value isn't mapping from your inbound packets to your rdescs. hiddump reattaches the device so it would imply something in either init or the handling is off. Or maybe the hardware is just bad and can't handle renegotiation? It has happened elsewhere, albeit it should be very unlikely on a display tablet.

It will probably just take double checking your work making sure incoming packets map to the rdescs properly. You seem to be able to get the packet data after activation which is 99% of the battle.

If this does end up being generic across both tablets I'd be very interested in if we could find someone with the 22R Pro, 24 Pro, and / or 13.3 Pro to see if they all behave similarly. There doesn't look to be any diagnostics in the tablets db for any of them yet though to check against. I wouldn't put it past Xp-pen to be using the same digitizer across the whole product range though. From their website it looks like at least the 13.3 and 22R have all the same parameters and the use the same pen as the 15.6 and 12. If that is all the case, let me know when you are done and I'll rebase my branch off upstream and you can merge into my tree and we can try to get support for both tablets merged at once - having two devices with as many users as this patch series has seen is probably enough test coverage to start moving towards getting it all into the kernel proper.

alfazaz, 2020-06-02:
Hello @Korvox !

About activation, it seems that things are common for many xp-pen tablets (I saw it also for XP-PEN Star G430S_B - see links above).

Well, I don't know how to solve the `irq status -75` messages... I tried many things but I'm like a blind man in the fog...

I forked your github and made changes you can see there (for the moment, things for artist 12 pro and 15.6 pro are common - using a case fall through - but I can separate them if necessary) :

https://github.com/alfazaz/digimend-kernel-drivers

I didn't changed rdescs.

If you have ideas to help me solve my problems... What should I try ? Which tools to use and how ?
I'm not an expert on USB things and kernel modules (but I know how to play with archlinux since many years...).

Add-on : I didn't know if I should fork your github or another one... It seems that it is many commits behind digimend master.

Korvox, 2020-06-06:
That irq means the incoming packets aren't mapping properly to the descriptors. You have the packets and you can get the descriptors XP-Pen is using on Windows via dumping tools there. In my case I just wrote it out by hand since I knew how the packets were formatted and often vendor descriptors are poorly written. A common gotcha is that the packets are little endian so multibyte numbers are least significant digit first. The USB site has the [official spec](https://www.usb.org/sites/default/files/documents/hid1_11.pdf) for descriptors and the [usage tables](https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf) as PDFs if you want to actually know how they work and end up writing your own.

alfazaz, 2020-08-18:
Hello @Korvox 

I have time to work again on support for XP-Pen Artist 12 Pro. I restarted from your latest version for Artist 15.6 Pro (https://github.com/DIGImend/digimend-kernel-drivers/pull/291#issuecomment-658417116) adding the Artist 12 Pro but... I still have same problems. It's not working (pen and pad not detected with hid_uclogic).

I verified your 15.6 Pro rdesc and applied it to 12 Pro because I have exactly the same results as you in https://github.com/DIGImend/digimend-kernel-drivers/issues/286#issuecomment-503792872 :  after activation (which works), packets are like this :

* Byte 1 : 02 for report ID
* Byte 2 : 1 bit for tip switch bit + 1 bit for barrel switch + 1 bit for tablet pick + 2 bits unused + 1 bit for in range + 2 bits unused
* Bytes 3 and 4 : 16 bits for Desktop X
* Bytes 5 and 6 : 16 bits for Desktop Y
* Bytes 7 and 8 : 16 bits for Tip Pressure
* Byte 9 : 8 bits for X Tilt
* Byte 10 : 8 bits for YTilt
* Bytes 11 and 12 : unused

Any idea to help finding solution ?

State of my work is here : 
https://github.com/alfazaz/digimend-kernel-drivers

Add-on : @Korvox , you said that you changed the name of `uclogic_params_init_ugee_xppen_a156p` to `uclogic_params_init_ugee_xppen_pro` since it seems common across several models in the same family but there are references to `uclogic_rdesc_xppen_a156p_pen_arr` and `uclogic_rdesc_xppen_a156p_pen_size` which should have more common names if these rdesc are really commons. 

It seems to be the case between Artist 12 Pro and Artist 15.6 Pro and something else is wrong but what ?

alfazaz, 2020-08-19:
I wrote to XP PEN to have help for linux kernel driver for their tablets... I'm waiting for an useful answer...

alfazaz, 2020-09-14:
No answer... I gave up until someone find a solution (I think it's only a detail I didn't find...).

Kaspi314, 2020-11-16:
I too would like to see this as I have the same tablet but no.

It would be nice to have left handed support and tilt.

Korvox, 2020-11-17:
It won't help with the tilt but if you want GUI based left handed mode emulation on KDE I made a [Tablet KCM](https://gitlab.com/zanny/kcm-tablet) that supports it.

Quentin59000, 2020-11-27:
I there, 

Thank's for this helpful thread. I have the XP-Pen Artist Pro 13.3 and I'm facing the same issues with the keyboard. I tryed reusing the @Korvox fork without success. 

By the way I also found [this fork](https://github.com/megamuteki/xppenexp) and it seems that this can handle the pad on multiple XP Pen devices but unfortunately the 13.3 Pro model (and neither the 12 Pro) is defined here and I was not able to have a proper usage of my tablet with it.

So like @alfazaz I gave up and I was a bit disappointed with the answer of XP Pen company because publishing a driver on their website made me understand that I could configure the buttons through their interfaces (they answers me "Après demande de confirmation auprès de mes collègues : le pilote pour Linux ne permet pas de modifier ces paramètres." which basicaly means that their interfaces are unable to configure anything ...)

So I have quickly wrote a workaround to allow me to [configure the xp pen keyboard buttons](https://github.com/Quentin59000/xp-pen-pad-keyboard-python-driver/). 

If this can help while waiting for a better solution.

kurikaesu, 2021-08-22:
@alfazaz @Quentin59000 Please try the code in my PR https://github.com/DIGImend/digimend-kernel-drivers/pull/557.
I've gone through all of the correspondence in this issue and the 12 Pro and 13.3 Pro are implemented identically.

https://github.com/DIGImend/digimend-kernel-drivers/pull/557#issuecomment-903073707 is the results of my tests on my 13.3 Pro. Let me know if my PR works for either of you.

alfazaz, 2021-08-22:
@kurikaesu I was disappointed and gave up on this project... and I was happy to test your #557 work and... Wouahou !!!!! Yes, it works on my 12 Pro !!!!!!

Really, really many thanks ! 

Hope this will be committed "soon" in the kernel...

terragonjohn, 2021-12-26:
Hi.

I've read this from the OP:

> I just got an XP-Pen Artist Pro 12, and it's working with the standard drivers, but the buttons are not configurable

and since I have to decide whether or not to buy an artist pro 12, I wanted to ask what the OP meant exactly.
What are the standard drivers (libinput, evdev, digimend...)? And what exactly works out of the box?
Because I'd be already sold even if all that worked out of the box was the display and basic drawing (no tilt, no pressure levels etc). I just need it for online lectures so if the display works and if I'm able to just draw  lines with minimal configuration I'd be set.

Thanks
John

kurikaesu, 2021-12-26:
Currently nothing works out of the box with the digimend drivers as you need to pull my branch in order for the buttons to work but without any default bindings. Bindings need to be set with xsetwacom.

You could alternatively try and use my userspace drivers here where I support the Artist Pro 12: https://github.com/kurikaesu/userspace-tablet-driver-daemon

terragonjohn, 2021-12-26:
> Currently nothing works out of the box with the digimend drivers as you need to pull my branch in order for the buttons to work but without any default bindings. Bindings need to be set with [...]

OK thanks.
I guess I misunderstood the OP.
I was hoping to do as the OP in [498](https://github.com/DIGImend/digimend-kernel-drivers/issues/498) did. The Pro 13.3 seems to be working (minimally, which is all I need) with the wacom and evdev drivers.