DIGImend/digimend-kernel-drivers #352: Install script not working in Fedora 31 64 bit

binarydepth, 2020-02-19:
**$ sudo make dkms_install**
dkms add .

Creating symlink /var/lib/dkms/digimend/9/source ->
                 /usr/src/digimend-9

DKMS: add completed.
dkms build digimend/9

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.17-200.fc31.x86_64 KVERSION=5.4.17-200.fc31.x86_64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.17-200.fc31.x86_64 (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
make: *** [Makefile:74: dkms_modules_install] Error 10

-----------------------------------------------------------------------------------------------------------------------------
**$ cat /var/lib/dkms/digimend/9/build/make.log**
DKMS make.log for digimend-9 for kernel 5.4.17-200.fc31.x86_64 (x86_64)
Wed 19 Feb 2020 07:30:31 AM AST
make[1]: Entering directory '/var/lib/dkms/digimend/9/build'
make -C /lib/modules/5.4.17-200.fc31.x86_64/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[2]: Entering directory '/usr/src/kernels/5.4.17-200.fc31.x86_64'
  LEX     scripts/kconfig/lexer.lex.c
/bin/sh: flex: command not found
make[4]: *** [scripts/Makefile.host:9: scripts/kconfig/lexer.lex.c] Error 127
make[4]: *** Waiting for unfinished jobs....
  YACC    scripts/kconfig/parser.tab.[ch]
/bin/sh: bison: command not found
make[4]: *** [scripts/Makefile.host:17: scripts/kconfig/parser.tab.h] Error 127
make[3]: *** [Makefile:567: syncconfig] Error 2
make[2]: *** [Makefile:678: include/config/auto.conf.cmd] Error 2
make[2]: Leaving directory '/usr/src/kernels/5.4.17-200.fc31.x86_64'
make[1]: *** [Makefile:22: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/digimend/9/build'
-------------------------------------------------------------------------------------------------------------
**Added Flex and Bison**
-------------------------------------------------------------------------------------------------------------
**$ sudo make dkms_install**
dkms add .

Creating symlink /var/lib/dkms/digimend/9/source ->
                 /usr/src/digimend-9

DKMS: add completed.
dkms build digimend/9

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.17-200.fc31.x86_64 KVERSION=5.4.17-200.fc31.x86_64....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.17-200.fc31.x86_64 (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
make: *** [Makefile:74: dkms_modules_install] Error 10
 
**$ cat /var/lib/dkms/digimend/9/build/make.log** 
DKMS make.log for digimend-9 for kernel 5.4.17-200.fc31.x86_64 (x86_64)
Wed 19 Feb 2020 07:42:15 AM AST
make[1]: Entering directory '/var/lib/dkms/digimend/9/build'
make -C /lib/modules/5.4.17-200.fc31.x86_64/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[2]: Entering directory '/usr/src/kernels/5.4.17-200.fc31.x86_64'
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
  HOSTCC  arch/x86/tools/relocs_32.o
  DESCEND  objtool
  HOSTCC  arch/x86/tools/relocs_64.o
  HOSTCC  arch/x86/tools/relocs_common.o
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile.build:37: /usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Build.include: No such file or directory
make[6]: *** No rule to make target '/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Build.include'.  Stop.
make[5]: *** [Makefile:43: /usr/src/kernels/5.4.17-200.fc31.x86_64/tools/objtool/fixdep-in.o] Error 2
make[4]: *** [/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile.include:5: fixdep] Error 2
make[3]: *** [Makefile:67: objtool] Error 2
make[2]: *** [Makefile:1754: tools/objtool] Error 2
make[2]: *** Waiting for unfinished jobs....
  HOSTLD  arch/x86/tools/relocs
make[2]: Leaving directory '/usr/src/kernels/5.4.17-200.fc31.x86_64'
make[1]: *** [Makefile:22: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/digimend/9/build'

spbnick, 2020-02-19:
Did you run this command from the [README.md](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/README.md):

    sudo dnf install -y "kernel-devel-uname-r == $(uname -r)"

binarydepth, 2020-02-19:
Ran it again
`Package kernel-devel-5.4.17-200.fc31.x86_64 is already installed.`

binarydepth, 2020-02-20:
It worked with the original kernel on a fresh install. But the tablet doesn't work, HUION 1409.

closed 2020-02-20 by binarydepth