DIGImend/digimend-kernel-drivers #358: XP-Pen Pro

gs-foto, 2020-03-12:
Is there work being done on the XP-Pen Pro Small? The tablet, touchpad and stylus are recognized, and I can draw with pressure in gimp and krita, but I cannot assign any buttons, the wheel or the touchpad. Here is the output from xsetwacom and uclogic-probe:

$ xsetwacom --list
9 inch PenTablet touch          	id: 11	type: TOUCH     
9 inch PenTablet pad            	id: 12	type: PAD       
9 inch PenTablet stylus         	id: 15	type: STYLUS    

$ uclogic-probe 1 5
M
P 39 00 20 00 69 00 6E 00 63 00 68 00 20 00 50 00 65 00 6E 00 54 00 61 00 62 00 6C 00 65 00 74 00
S 64 0E 03 C8 B3 34 65 08 02 FF 1F D8 13 00 00
S 65 04 03 01 67
S 6E 0B 03 00 10 00 0D 08 02 00 00 03
S 79 10 03 20 20 20 20 20 20 20 20 20 20 20 20 20 20
S 7A 08 03 00 00 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
Failed to get string descriptor 0xC8: Input/Output Error

The ring light changes from blue to red at the end uclogic-probe.

If there's anything else I can do to help, let me know.

DjogaRo, 2020-03-16:
@gs-foto Just to be safe, I assume you mean the XP-Pen Deco Pro Small.? (Maybe change the title to not get overlooked.)

I've got that model, too. I don't know, what to do on the developer's side of things, but I gather I can add intel.
Attached are the rdesc files from the kernel's debugfs, the output of
`sudo lsusb -vd 0x28bd:0909`
a description of what outgoing payloads I gathered and when - using the driver software in Windows to change settings - (sniffed with Wireshark) and a Wireshark session containing plugging in the device and changing every setting in the driver software that causes outgoing packets (orientation, touch pad mode, LED states (2x [0%|50%|100%])).

Can the intel from the latter two be used to change the touch pad mode from Linux? If so, how?

When I get to it, I'll gather input events from actual usage as described here
https://digimend.github.io/support/howto/trbl/diagnostics/

As I don't know much about drivers, how come I can use the device already in Linux? Does that mean the PID gets recognized or does my libinput driver genericaly take over due to some info in the report descriptors?

[0003:28BD:0909.0006_kb_touch_wheel.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4337923/0003.28BD.0909.0006_kb_touch_wheel.txt)
[0003:28BD:0909.0007_pen.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4337924/0003.28BD.0909.0007_pen.txt)
[0003:28BD:0909.0008_out_control.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4337925/0003.28BD.0909.0008_out_control.txt)
[sudo_lsusb_-vd_0x28bd:0909.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4337926/sudo_lsusb_-vd_0x28bd.0909.txt)
[tablet_control_packets.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4337927/tablet_control_packets.txt)
[tablet_initialization_and_control_packets__filtered.pcapng.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/4337992/tablet_initialization_and_control_packets__filtered.pcapng.zip)

DjogaRo, 2020-03-18:
Attached are the outputs of
`sudo usbhid-dump -ed -m 28bd:0909`
and
`sudo uclogic-probe _<Bus#> <Device#>_`
I'm not sure, how reliable the latter is. Same as for gs-foto, the ring light changed from blue to red. The active-field-corner-marker LEDs switched off, lsusb and usbhid-dump not able anymore to find a device 28bd:0909. After physically reattaching the device, its LEDs are back at full throttle. (Before, I had the ring light off and the markers at 50%, which was kept beyond system reboots and disconnecting the device - those settings didn't survive uclogic-probe, though.)

Later I'll gather input events - but without running uclogic-probe first, obviously.

[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4348919/hid_report_descriptors.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4348921/probe.txt)

DjogaRo, 2020-03-19:
In the archive are the input events. I made seperate files for each action and direction to keep it manageable for anyone, who wants to take a look at it.

The touchpad in mousemode, when used with two fingers, only works vertically. There's a deadzone, where you have put your fingers down. If you swipe up or down beyond the deadzone, it starts firing events, until you swipe back into the deadzone or lift your fingers.

I couldn't gather events for the touchpad in the other two modes, since I don't know how to change the mode, when I'm in Linux. (In Windows the driver tells the device to change modes. Even when a button is set to change modes, it's the driver, which interprets the button press and decides to change modes. (Deducted from packet sniffing - see my post up there.))

[raw_input_samples.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/4352205/raw_input_samples.zip)

gs-foto, 2020-03-25:
Stupid question, but I'm missing something here: How do I attach files to my post?

gs-foto, 2020-03-25:
Here are my diagnostic results.
Notes: touchpad scroll:
One revolution clockwise, then one revolution counter-clockwise.
touchpad drag:
left to right
right to left
top to bottom
bottom to top

[touchpad_scroll.txt](https://github.com/DIGImend/digimend-kern
el-drivers/files/4380900/touchpad_scroll.txt)
[touchpad_drag.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380915/touchpad_drag.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380922/probe.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380923/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380924/pen_pressure.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380925/pen_coords.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380926/pen_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380927/hid_report_descriptors.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380928/frame_dials.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380929/frame_buttons.txt)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4380930/descriptors.txt)

JoseExposito, 2022-07-10:
Just sent support for your tablet to the upstream kernel:
https://lore.kernel.org/linux-input/20220710175043.192901-1-jose.exposito89@gmail.com/T/

Meaning that, hopefully, Linux 5.20 or 5.21 will include support for your tablet out of the box.

Meanwhile, I built a DIGImend modified version so the user who reported the issue could test it:
https://gitlab.freedesktop.org/libinput/libinput/-/issues/738#note_1453926

I hope you can start using your tablet with that driver.