DIGImend/digimend-kernel-drivers #360: debian 10 (buster), huion 1409v2 in wireless mode, wayland -> generally working, but how to configure?

stockh0lm, 2020-03-21:
I run debian buster, use the most recent digimed kernel drivers (v9) as downloaded from here, have a XWayland v1.20 (which is said to handle tablets), and gnome (v3.30) lists wacom stylus tablet in its device menu, but shows no tablets found. 

As discussed elsewhere, xsetwacom does not list devices because the disjointed input handling between wayland, xwayland, xsetwacom and libinput. 

So how can i configure and use the buttons on the tablet? I hear they make life with the tablet so much better. I dont have any personal experience - this is my first tablet input device.

0wn3r, 2020-05-10:
Not sure if this will be the same for your wireless device, but I was able to configure pressure and map buttons for H640P under Ubuntu 20.04 Wayland.
gnome-control-settings uses libwacom as a database of tablet information. Many of Huion devices are missing there, possibly because of reusing same ProductID. I found this https://github.com/linuxwacom/libwacom/pull/235 as an example to create configuration for H640P.
[huion-h640p.tablet.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4606378/huion-h640p.tablet.txt)
[huion-h640p.svg.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4606379/huion-h640p.svg.txt)
Edit them for your device and put under /usr/share/libwacom/ and /usr/share/libwacom/layouts/ with correct names.
This allowed me to change stylus pressure curve and map stylus to monitor with correct aspect ration, but I couldn't map buttons.
Reason for that is that tabled pad is not reported as Pad, but a Tablet to libinput. You need to create an udev rule to report it as a Pad.
/lib/udev/rules.d/65-libwacom.rules
```
# Huion H640P
ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="256c", ENV{ID_MODEL_ID}=="006d",  ENV{ID_INPUT}="1", ENV{ID_INPUT_JOYSTICK}="", ENV{ID_INPUT_TABLET}="1"
ATTRS{name}=="* Pad", ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="256c", ENV{ID_MODEL_ID}=="006d",  ENV{ID_INPUT_TABLET_PAD}="1"
```
You can view how it's reported with 
sudo libinput debug-events
```
-event7   DEVICE_ADDED     HID 256c:006d Pen                 seat0 default group3  cap:T  size 160x100mm left calib
-event8   DEVICE_ADDED     HID 256c:006d Pad                 seat0 default group3  cap:P left buttons:13 strips:0 rings:0 mode groups:1
```

Luke-101, 2021-02-18:
sorry to disturb you @0wn3r , but I'm pretty new to linux so can you please expand on "with correct names", or what the code on "/lib/udev/rules.d/65-libwacom.rules" does.
Because I tried your solution and when i press the buttons on my tablet they don't appear on the log from "sudo libinput debug-events".
Thank you in advance.

0wn3r, 2021-02-18:
Hi @Luke-101, "with correct names" means that file name should match your tablet name and also it should not have the .txt file extension. I added .txt because github doesn't accept the original name. You should modify both tablet and svg files to match your tablet.
Those rules in /lib/udev/rules.d/65-libwacom.rules change how device is seen by system.
In this case first line says that device with VendorID 256c ModelID 006d is an input device ID_INPUT and a tablet ID_INPUT_TABLET
Second line says that device with VendorID 256c, ModelID 006d and name matching "* Pad" which will translate into "256c:006d Pad" is a tablet pad ID_INPUT_TABLET_PAD.
You don't see any buttons pressed because Pad device is not seen as Pad in your system. This is what it looks when there are no udev rules for my pad:
`event257 - HID 256c:006d Pad: libinput bug: missing tablet capabilities: pen resolution. Ignoring this device.`
Some additional info can be found on libwacom wiki page:
https://github.com/linuxwacom/libwacom/wiki/Tablet-Definition-Files
https://github.com/linuxwacom/libwacom/wiki/Troubleshooting
https://github.com/linuxwacom/libwacom/wiki/Udev-Rules

Doffur, 2021-07-31:
> Not sure if this will be the same for your wireless device, but I was able to configure pressure and map buttons for H640P under Ubuntu 20.04 Wayland.
> gnome-control-settings uses libwacom as a database of tablet information. Many of Huion devices are missing there, possibly because of reusing same ProductID. I found this [linuxwacom/libwacom#235](https://github.com/linuxwacom/libwacom/pull/235) as an example to create configuration for H640P.
> [huion-h640p.tablet.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4606378/huion-h640p.tablet.txt)
> [huion-h640p.svg.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4606379/huion-h640p.svg.txt)
> Edit them for your device and put under /usr/share/libwacom/ and /usr/share/libwacom/layouts/ with correct names.
> This allowed me to change stylus pressure curve and map stylus to monitor with correct aspect ration, but I couldn't map buttons.
> Reason for that is that tabled pad is not reported as Pad, but a Tablet to libinput. You need to create an udev rule to report it as a Pad.
> /lib/udev/rules.d/65-libwacom.rules
> 
> ```
> # Huion H640P
> ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="256c", ENV{ID_MODEL_ID}=="006d",  ENV{ID_INPUT}="1", ENV{ID_INPUT_JOYSTICK}="", ENV{ID_INPUT_TABLET}="1"
> ATTRS{name}=="* Pad", ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="256c", ENV{ID_MODEL_ID}=="006d",  ENV{ID_INPUT_TABLET_PAD}="1"
> ```
> 
> You can view how it's reported with
> sudo libinput debug-events
> 
> ```
> -event7   DEVICE_ADDED     HID 256c:006d Pen                 seat0 default group3  cap:T  size 160x100mm left calib
> -event8   DEVICE_ADDED     HID 256c:006d Pad                 seat0 default group3  cap:P left buttons:13 strips:0 rings:0 mode groups:1
> ```
 
I am new to Linux.
How to add these, I am really confused right I now.

mk30, 2021-10-07:
@0wn3r thank you so much for sharing these config files & providing instructions. i run lubuntu 20.04 and just got a huion h640p. i copied huion-h640p.tablet into /usr/share/libwacom, copied huion-h640p.svg into /usr/share/libwacom/layouts, and modified /lib/udev/rules.d/65-libwacom.rules by adding the section you included above.

i'm trying to get pressure from the pad, but it seems like i'm still not getting it. is there something i've missed? also, you mention changing the stylus pressure curve, but i can't figure out how i would do that.

also, at the beginning of my /lib/udev/rules.d/65-libwacom.rules, i see this rule:

```
# HUION consumer control devices are not tablets.
ATTRS{name}=="HUION * Consumer Control", GOTO="libwacom_end"
```

should i do anything with that? thank you.