DIGImend/digimend-kernel-drivers #361: Huion 1060 Plus(8192) in ubuntu

EloyAnguianoRey, 2020-03-22:
Hello,

I just bought a wacom but when I try to operate on it in linux it doesn't do anything. I've followed all the steps they indicate using the debian package. There were no errors during installation. Everything looks good but it doesn't do anything. Can you help me?

Some info:

uname -a
Linux amaltea 5.3.0-42-generic #34-Ubuntu SMP Fri Feb 28 05:49:40 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
 xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ AUI165B:01 044E:121E Mouse              	id=14	[slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G RF Keyboard & Mouse Consumer Control	id=11	[slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G RF Keyboard & Mouse   	id=19	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Integrated Camera: Integrated C         	id=13	[slave  keyboard (3)]
    ↳ AUI165B:01 044E:121E Keyboard           	id=15	[slave  keyboard (3)]
    ↳ AUI165B:01 044E:121E                    	id=16	[slave  keyboard (3)]
    ↳ Ideapad extra buttons                   	id=17	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=18	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G RF Keyboard & Mouse   	id=9	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G RF Keyboard & Mouse System Control	id=10	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G RF Keyboard & Mouse Consumer Control	id=12	[slave  keyboard (3)]

dmesg
[  177.398700] usb 1-1: new full-speed USB device number 11 using xhci_hcd
[  177.549093] usb 1-1: New USB device found, idVendor=3938, idProduct=1032, bcdDevice= 1.37
[  177.549098] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  177.549102] usb 1-1: Product: 2.4G RF Keyboard & Mouse
[  177.549105] usb 1-1: Manufacturer: MOSART Semi.
[  177.554102] input: MOSART Semi. 2.4G RF Keyboard & Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:3938:1032.0004/input/input23
[  177.611160] hid-generic 0003:3938:1032.0004: input,hidraw1: USB HID v1.10 Keyboard [MOSART Semi. 2.4G RF Keyboard & Mouse] on usb-0000:00:14.0-1/input0
[  177.614452] input: MOSART Semi. 2.4G RF Keyboard & Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:3938:1032.0005/input/input24
[  177.615695] input: MOSART Semi. 2.4G RF Keyboard & Mouse Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:3938:1032.0005/input/input25
[  177.671013] input: MOSART Semi. 2.4G RF Keyboard & Mouse System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:3938:1032.0005/input/input26
[  177.671190] input: MOSART Semi. 2.4G RF Keyboard & Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:3938:1032.0005/input/input28
[  177.671490] hid-generic 0003:3938:1032.0005: input,hiddev0,hidraw2: USB HID v1.10 Mouse [MOSART Semi. 2.4G RF Keyboard & Mouse] on usb-0000:00:14.0-1/input1

belamenso, 2020-03-22:
On my laptop with Kubuntu 19.10 and 5.3.0-42-generic kernel I had to disable xHCI option in BIOS.

EloyAnguianoRey, 2020-03-23:
I don't have that option in BIOS, however as you can see the device is in USB2.0.

I have configured xorg with the correct vendor and product and yet when I look at the x11 events that are generated with xinput no event is generated.

lsusb info:

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:b023 Realtek Semiconductor Corp. Bluetooth Radio 
Bus 001 Device 003: ID 04f2:b604 Chicony Electronics Co., Ltd Integrated Camera
Bus 001 Device 015: ID 3938:1032 MOSART Semi. 2.4G RF Keyboard & Mouse
Bus 001 Device 018: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 001 Device 017: ID 256c:006e  USB2.0Hub
Bus 001 Device 016: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

EloyAnguianoRey, 2020-03-29:
Hello,

It's finally working properly.In the end my solution for Ubuntu is to downgrade the kernel to 5.0.0, reinstall digimend-dkms-9 and in BIOS put all the options available to Legacy.

Thank you very much for your invaluable help.

closed 2020-03-29 by EloyAnguianoRey