DIGImend/digimend-kernel-drivers #362: HUION HS611

rsmozos, 2020-03-23:
Please add support for HUION HS611

https://www.huion.com/pen_tablet/huion-hs611.html

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4371943/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4371944/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4371945/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4371946/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4371947/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4371948/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4371949/pen_tilt.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4371950/probe.txt)

spbnick, 2020-03-24:
Thank you for the diagnostics, @rsmozos. I have recently merged support for HS611 into master and it will be a part of the next release. Meanwhile you can try installing the [latest development version](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip).

spbnick, 2020-03-24:
Scratch that, I'm still working on HS611, I got HS610 merged. I always confuse them :)
Anyway, try the latest master it will mostly work.

rsmozos, 2020-03-24:
Many thanks!
It worked beautifully.  
The pen buttons also works.  Its buttons are now central mouse button and right button.
I don't know whether the buttons in the tablet work. But it is something unimportant.
How do I configure the buttons of the tablet?

rsmozos, 2020-03-24:
On the other hand,
When I recorded the buttons in the frame, I skipped the ones controlling sound volume, play, pause, show desktop, etc.
I could record those also repeating frame_buttons.txt  file. Please, let me know if that's helpful.
Many Thanks.

spbnick, 2020-03-25:
Thank you, @rsmozos! I now have the tablet myself, and won't need anymore diagnostics.

spbnick, 2020-04-05:
@rsmozos, I've worked out a way to make all the buttons on the frame work. That is to get them out of the kernel and to the appropriate X.org driver. Please try the latest development version of the driver again.

rsmozos, 2020-04-06:
Dear Nikolai,

Thanks so much!! Now the keys to control volume, playlist and some window
control work.
However, I'm not able to assign the keys in the left side of the table.
I checked using showkey command. When I press the keys on the top some
keycodes are shown.
However, when I press the keys on the left side, nothing happens.

Should I configure them some other way?
Thanks again!
Best Regards,
Ricardo.

spbnick, 2020-04-26:
@rsmozos, you need to configure them using xsetwacom, as pointed out in the README.md.

diplomatichal, 2020-04-27:
Hi! Just for info, is the HS611 supposed to work correctly with the latest development version? Thanks for the great service to the community! :)

spbnick, 2020-04-27:
@diplomatichal, yes it's supposed to work with the development version and the next release v10.

spbnick, 2020-05-17:
Fixed in latest master.

closed 2020-05-17 by spbnick

sebgre, 2020-06-06:
@rsmozos 
Hi, sorry to disturb you, I also have a HS611. I have installed the master branch and the tablet is working. I can assign the left button, but the volume/windows buttons and scroll bar aren't working. I saw in one of your post that you succeeded to have those buttons working. Did you do something special for that ? Thanks a lot

rsmozos, 2020-06-06:
I just compiled the drivers from master.
I didn't modify x.org files. I'm not using wacon drive. Just installed the
drivers.

Uradamus, 2020-06-14:
So I picked one of these tablets up. As of now, on Manjaro, I can plug this in and most things work, including the media keys up top. I suppose that is with the xinput kernel driver, as I haven't installed digimend on this system at this time. But if I follow the directions on the readme for this github and add the file to get the wacom stuff to recognize this tablet, I can then get the tablet at least partly recognized by the kde settings GUI for wacom tablets. But I also lose the use of the media buttons. Is there a remedy for this, bit of a best of both worlds with the GUI convenience but still having a fully functioning tablet?

Bit of a rant, but I really like being able to use the GUI as it makes having different profiles for different software so much easier to manage and experiment with, as there is at least one program where I like to use the tablet in relative coordinates mode like a mouse, as it is 3D and wasn't made with absolute coordinates in mind and causes the camera to flake out any time I try to move it around with the stylus without switching modes. Not to mention I use Blender and Krita extensively as well and I have very different express key preferences for both of those.

But it sucks losing the use of those handy media buttons. This is one of those areas where Linux has been really lacking for entirely too long, we really need better support for non-wacom tablets when it comes to GUI configuration tools, so much better than having to mess around with scripts for your average user. Sucks having to switch to the incomplete wacom drivers with poor feature support just to gain access to such basic useful tools. /rant

sebgre, 2020-06-15:
Hi,

I first take the opportunity of this email to thank @rsmozos for his
answer, it solved my issues.

To come back to your question, I'm not on Manjaro, but on Arch, so it
should be pretty close. No KDE for me but Cinnamon, I don't know if it
makes a difference or not.

I was in pretty much the same situation as you. What I did:
- install the last version of the digimen drivers from github. It didn't
worked with the aur package.
- do not update the xorg config. Also check that there isn't any reference
to the tablet in the xorg.conf file or xorg.conf.d directory.
- you should now be able to see the tablet with the command xsetwacom list
devices. You should have 3 lines.

I don't use any GUI, but I configure the buttons with the xsetwacom command.
Media buttons are working as well as the zoom bar.

Let me know if this works. I'm on mobile right now, I can have a deeper
look later when I'm on m computer.

Regards

diplomatichal, 2020-07-26:
Hi all! 
I finally managed to get my hands on a HS611. It works mostly out of the box (pen, buttons on the pen and pressure sensitivity), although I am having trouble assigning the top touch buttons and touch strips. 
I'm running ArchLinux, kernel 5.7.10-arch1-1, and installed the latest digimend driver from source. I tried messing around a bit with the xsetwacom commands but honestly I have no idea how to even set the touch strip to scroll up/down. (Also, are the hardware button numbers assigned with any logic? It seems some numbers are skipped...)
@sebgre any comment on how you managed to make the strips work? Or did they work out of the box?

sebgre, 2020-07-29:
Hi @diplomatichal.
I just tested, and apparently the touch strip isn't working anymore. It works well out of the box on my other computer which is running linux mint, but doesn't react on my Arch computer. On the Linux Mint, is zooms in and out. I don't really use it, so I haven't looked further why it is not working anymore.
The hardware buttons are numbered 1 to 3, then 8 to 14. Following the wiki
> Because X uses buttons 4-7 as the four scrolling directions, physical buttons 4 and higher are mapped to mouse buttons 8 and higher by default. While xorg.conf uses the actual button numbers and only lets you map to mouse buttons, xsetwacom uses the translated mouse button numbers and allows mapping to multiple keycodes (but not keysyms). 
Sorry that I can't help you further.

I-dont-need-name, 2020-08-07:
> Hi, I first take the opportunity of this email to thank @rsmozos for his answer, it solved my issues. To come back to your question, I'm not on Manjaro, but on Arch, so it should be pretty close. No KDE for me but Cinnamon, I don't know if it makes a difference or not. I was in pretty much the same situation as you. What I did: - install the last version of the digimen drivers from github. It didn't worked with the aur package. - do not update the xorg config. Also check that there isn't any reference to the tablet in the xorg.conf file or xorg.conf.d directory. - you should now be able to see the tablet with the command xsetwacom list devices. You should have 3 lines. I don't use any GUI, but I configure the buttons with the xsetwacom command. Media buttons are working as well as the zoom bar. Let me know if this works. I'm on mobile right now, I can have a deeper look later when I'm on my computer. Regards

I did what you described however buttons on the left and touchstrip don't work. You mentioned that "it isn't working anymore" why? is that a problem with kernel update or something else?

sebgre, 2020-08-08:
Hi. I'm certainly not an expert, so my knowledge on the subject is limited to my experience in this specific case. What I know:
- On my ubuntu based distro (linux mint), zoom bar and media players are working out of the box with my procedure above.
- On my Arch distro, the zoom bar doesn't work and media players work out of the box. It seems to me that the zoom bar worked before, but I'm not sure at all, as I don't normally use it.
- To configure the buttons on the left, you have to do that manually with the xsetwacom function. They are plenty of tuto on the wab on how to use it, so I will not go into details. I copy below the script that I use for the shortcut for Krita.

`#!/bin/bash
#Use xev command to find keycode for special keys
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 1 key +Ctrl +Z				#undo
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 2 key +Ctrl +Shift +Z		#redo
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 3 key +K					#plus foncé
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 8 key +L					#plus clair
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 9 key +ctrl					#prendre couleur
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 10 key +T					#bouger objet
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 11 key +B					#brosse
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 12 key +E					#gomme
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 13 "key +ISO_Level3_Shift +0xfe52"					#diminue taille brosse
xsetwacom set "HUION Huion Tablet_HS611 Pad pad" button 14 "key +ISO_Level3_Shift +0x24"					#augmente taille brosse
`

I hope this helps.

I-dont-need-name, 2020-08-09:
> I hope this helps.
Thank you, it did help and now the buttons work.

I also tried Linux mint live xfce iso and zoom bar doesn't work there (out of the box at least). when run `evtest` on my arch linux there is a device `/dev/input/event10:     HUION Huion Tablet_HS611 Touch Strip` and output is as follows:
`Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6d version 0x111
Input device name: "HUION Huion Tablet_HS611 Touch Strip"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
    Event code 8 (ABS_WHEEL)
      Value      0
      Min        0
      Max        7
      Resolution       3
    Event code 40 (ABS_MISC)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)
Event: time 1596970375.678162, type 3 (EV_ABS), code 40 (ABS_MISC), value 15
Event: time 1596970375.678162, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 4
Event: time 1596970375.678162, -------------- SYN_REPORT ------------
Event: time 1596970375.866168, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 3
Event: time 1596970375.866168, -------------- SYN_REPORT ------------
Event: time 1596970375.982162, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 2
Event: time 1596970375.982162, -------------- SYN_REPORT ------------
Event: time 1596970376.760154, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 3
Event: time 1596970376.760154, -------------- SYN_REPORT ------------
Event: time 1596970376.798163, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 4
Event: time 1596970376.798163, -------------- SYN_REPORT ------------
Event: time 1596970376.874166, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 5
Event: time 1596970376.874166, -------------- SYN_REPORT ------------
Event: time 1596970376.914164, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 6
Event: time 1596970376.914164, -------------- SYN_REPORT ------------
Event: time 1596970376.952159, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1596970376.952159, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 0`
I believe there is a way to get it to work but I don't know how.

I-dont-need-name, 2021-02-27:
So today I reinstalled drivers again and now even two buttons on right don't seem to work (did they?).
I'm also frustrated that Touch Strip doesn't work in spite of showing output in evtest.
The log shows this:
[image]
I remember seeing some fixes for "resolution" but idk what to do with pen and btn_stylus

Speaking of keys on the right:
[image]
Why does it believe them to be tablet? this is just a keyboard with 2 keys..

If someone has any idea on what can be done to improve support of HS611 I would very grateful.

albfan, 2022-02-08:
About strip, from point of view of driver, work is done. Then is up to tech stack consuming it to do something with that, like libwacom -> libinput -> graphical session

I open this issue for GNOME: https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/1630

MikiVanousek, 2022-04-02:
Hey, the tablet is not among [supported devices](https://digimend.github.io/tablets/). Do I understand it correctly, it is indeed supported? 
Thanks, Miki

albfan, 2022-04-02:
Depends what means supported because this driver is uses by other software. If you use X11 probably all works. If you use wayland (and libinput) scroll and config is not in place, but mostly all básico things work (buttons, Pen, pressure)

MikiVanousek, 2022-04-02:
@albfan What does it mean, 'scroll and config are not in place'?

albfan, 2022-04-03:
this kernel implementation maps the scrollbar with a no standard events, see https://github.com/linuxwacom/libwacom/pull/288#issuecomment-1035688116

about config, standards like GNOME OS relay on libwacom to read tablet config 

https://github.com/linuxwacom/libwacom/pull/288/files

but as these Chinese tablets reuse the vendor and productid it is not about put that files in proper place, you need to remove other config for same vid:pid

That means that only ad-hoc implementations (per app) can use this device entirely, as it needs to deal with it's quirks.

I think krita and other apps can config directly devices anyway.

linuxgirl22, 2022-05-01:
Hello, may I ask if the pen already has the upper button set to right click and colour sample? or as non-good-at-tech way to configure it (Debian-based/Mate/no access to a gui/no wayland)? Considering purchasing this, but already issues with not able to configure xp-pen Deco L right click; arthritis in left hand, so using the pen in other hand for right-click colour sampling is vital. Thanks.

albfan, 2022-05-01:
@linuxgirl22 that depends on what software you use.

this is the GNOME config for tablet, where you can configure buttons. 

[image]

If that raises a colour sample depends on final app you are using (krita, gimp).

linuxgirl22, 2022-05-01:
Thank you @albfan for responding. Much appreciate. I'd looked previously to see if I could find the gnome config gui download and run it within Mate, but I think it's built into gnome. I previously needed to add an .hwdb file and a .tablet file to fix right click (xp-pen) for any softwares, so was wondering if similar would be needed for the HS611, while not knowing what the contents would need to be. But am not sure if Huion already sets the pen to work without needing such a workaround.

albfan, 2022-05-01:
All that is related with how tablet is used in an operative system, not about this, which are the kernel drivers (how signals from tablet are turned into input events)

https://ubuntu-mate.org/about/ looks mate is a continuation of GNOME 2 desktop, probably that do not offer the GNOME settings where you setup your tablet.

That probably still uses x11 config, and you need to check how to config it, but is outside scope of this project

albfan, 2022-05-01:
@linuxgirl22 I suggest you to hide your comments as offtopic

linuxgirl22, 2022-05-01:
With respect, no need for me to hide comments, as am asking on the HS611 thread if the HS611 pen's right click works to sample colour by default. I'd said a particular reason for asking too. I'm not sure if you're using a language translator, or there's some confusion generally? Just to clarify, others are discussing general support for the HS611, and I'm posting about that too.

albfan, 2022-05-01:
Your asking questions about OS and final software on a project for kernel drivers.

Once this driver reports button X is pressed, is up to your technology stack do something with it. Totally offtopic.

I get you are a little bit lost on where is the correct place to ask for help. But it is problably a forum on Ubuntu mate or whatever final app your using for drawing. Not here

linuxgirl22, 2022-05-01:
I'm not asking about OS or software. Tablets can need workarounds for pen buttons to work, and that's what's discussed in a thread such as this one about the HS611 ie what works and how. I'm not at all lost about where to ask and why, mainly because I'm not stupid. I don't want to discuss this more, as it's getting more surreal with each post.

albfan, 2022-05-02:
I have hs611. Pen works with kernel default driver. This driver fixes extra buttons and scrollbars still not merged into upstream driver.

Your questions are all about using a legacy OS (Ubuntu mate stick to GNOME 2) that can be more or leds friendly to setup your tablet. Here there will be no answers for your questions because concepts like color sample are for final app. On e button is reportes is Up to OS or final app to do something with it.

We both agreed you're not stupid, but insist to win this discussion make you look like more and more. You get your answers, hide this offtopic and find your answers elsewhere.