DIGImend/digimend-kernel-drivers #363: Huion 1060 Plus doen't works

EloyAnguianoRey, 2020-03-24:
Hello,
I have done all the actions to install my tablet huion 1060 plus in ubuntu on a lenovo 330 laptop and it does not work. I have scanned with wireshark the usb port and attached it here. While scanning I used the pen by pressing one button, then the other and then no buttons. No data is received from the tablet.

Can someone help me? In another issue I provided information about dmesg, lsusb, etc but I can provide it again here.
[huion-usb-scan.pcapng.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/4375292/huion-usb-scan.pcapng.zip)

prostylinux, 2020-03-24:
I agree. I have the same issue with Linux Mint 19.3. Computer identify huion via lsusb, but it command 'sudo usbhid-dump -es -m 256c:006e' results message:
Failed to find HID interfaces: Entity not found

prostylinux, 2020-03-24:
I've already tried with older kernels. Unfortunately doesn't help...

fernandodandrea, 2020-03-24:
Manjaro running on 5.5 kernel.
Device shows on `xinput list` shows it as HID 256c:006d Pen Pen (0).
It worked even before installing digimend drivers, but there's nowhere I can configure it, and I'm left-handed, so I kind of need the drivers.

Garbaz, 2020-03-25:
Same issue here for a few months now. Found some mentions of this issue in other places on the internet. It seems to be a problem with some changes in the Kernel, at least that is what I've gathered so far. I don't know whether this even is a problem you can solve in the digimend drivers.

Some observations:

I get in `dmesg`
```
[  604.789418] usb 1-1.1: new full-speed USB device number 14 using xhci_hcd
[  604.920457] usb 1-1.1: New USB device found, idVendor=256c, idProduct=006e, bcdDevice=30.00
[  604.920466] usb 1-1.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  604.927569] usb 1-1.1: can't set config #1, error -32
```
when plugging in the tablet and the same result as @prostylinux when running his `usbhid-dump` command and `Bus 001 Device 014: ID 256c:006e` (No name) in `lsusb`. This does not change with or without the digimend drivers.

```
Distro: Linux Mint 19.3
Kernel: 5.3.0
Device: Lenovo E560
```
**My current solution:**
When booting with Kernel 5.0.0 `dmesg` changes to:

```
[  236.349294] usb 1-3.1: new full-speed USB device number 16 using xhci_hcd
[  236.498335] usb 1-3.1: New USB device found, idVendor=256c, idProduct=006e, bcdDevice=30.00
[  236.498343] usb 1-3.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  236.514283] input: HID 256c:006e Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:256C:006E.000D/input/input36
[  236.573702] input: HID 256c:006e Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:256C:006E.000D/input/input37
[  236.574057] uclogic 0003:256C:006E.000D: input,hidraw0: USB HID v1.11 Keypad [HID 256c:006e] on usb-0000:00:14.0-3.1/input0
[  236.578633] uclogic 0003:256C:006E.000E: No inputs registered, leaving
[  236.578963] uclogic 0003:256C:006E.000E: hidraw1: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:14.0-3.1/input1

```

(independed of digimend being installed or not)

**and the device works** with digimend drivers installed (don't know if important, but I reinstalled the driver while booted in 5.0.0. Not sure whether the driver gets installed for all installed kernels or just the current one) and does nothing without digimend. `lsusb` still shows no name (I'm pretty certain it used to, but isn't important of course) and `usbhid-dump` works as expected.

EloyAnguianoRey, 2020-03-26:
My dmesg when I plug in the tablet:

[14975.510248] usb 1-1: new high-speed USB device number 11 using xhci_hcd
[14975.530946] usb 1-1: New USB device found, idVendor=058f, idProduct=6254, bcdDevice= 1.00
[14975.530952] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[14975.530955] usb 1-1: Product: USB2.0Hub
[14975.531917] hub 1-1:1.0: USB hub found
[14975.532042] hub 1-1:1.0: 4 ports detected

Only identifies flash storage. I will try to downgrade kernel to 5.0.0

EloyAnguianoRey, 2020-03-26:
Rebooting I found that dmesg tells me the following:

[  217.509790] usb 1-1: USB disconnect, device number 2
[  217.509795] usb 1-1.1: USB disconnect, device number 4
[  217.510248] usb 1-1.4: USB disconnect, device number 6
[  230.832753] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[  230.853381] usb 1-1: New USB device found, idVendor=058f, idProduct=6254, bcdDevice= 1.00
[  230.853386] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  230.853389] usb 1-1: Product: USB2.0Hub
[  230.855928] hub 1-1:1.0: USB hub found
[  230.856065] hub 1-1:1.0: 4 ports detected
[  231.144741] usb 1-1.1: new full-speed USB device number 9 using xhci_hcd
[  231.275529] usb 1-1.1: New USB device found, idVendor=256c, idProduct=006e, bcdDevice=30.00
[  231.275535] usb 1-1.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  231.281577] usb 1-1.1: can't set config #1, error -32

There's probably a rule missing from the udev or a rule that's wrong.

I don't know if any developer can help me. I'm a university professor and I urgently need the tablet for the distance classes caused by the COVID-19

EloyAnguianoRey, 2020-03-26:
@spbnick Can you help me?

spbnick, 2020-03-26:
Hi @EloyAnguianoRey, do you have another machine to test your tablet on? Maybe Windows, or another Linux distro? This looks a little like a hardware issue. If your tablet's USB cable is removable, try it with another cable.

spbnick, 2020-03-26:
Another thing you could try, if you know how to do that, is to install another Linux distro in a VM, and redirect your tablet there. If the problem is the kernel's USB stack, it could work with another distro in VM, as it will do (some of) its own initialization.

spbnick, 2020-03-26:
Oh, and of course, try with another kernel version, if you can.

EloyAnguianoRey, 2020-03-26:
In my daughter's windows it works perfectly. However, given the criticality of the moment and that I have to use the Tablet with Teams to teach, I cannot risk a reinstallation and a VM is obviously not an option.

Any more ideas? Any data you might need? Ask me what you need.

spbnick, 2020-03-26:
Can you try an older kernel version? Your distribution probably has multiple.

spbnick, 2020-03-26:
Also, try another USB port.

spbnick, 2020-03-26:
Or try connecting through a USB hub.

prostylinux, 2020-03-28:
Maybe this information help: As I said earlier, in Linux Mint 19.3 Huion 1060P stopped work couple weeks ago. I tried older version of Kernel (even 4.19) and nothing. I have another computer with Linux Mint 18.3 with kernel 4.15 and everything works perfectly.

Garbaz, 2020-03-28:
>  [ 231.281577] usb 1-1.1: can't set config #1, error -32

That is the same issue I got on Kernel 5.3, so downgrading to 5.0.0 (and make sure to install digimend again after downgrading) certainly would be worth a try.

EloyAnguianoRey, 2020-03-29:
Hello,

It's finally working properly.In the end my solution for Ubuntu is to downgrade the kernel to 5.0.0, reinstall digimend-dkms-9 and in BIOS put all the options available to Legacy.

Thank you very much for your invaluable help.

closed 2020-03-29 by EloyAnguianoRey