DIGImend/digimend-kernel-drivers #365: UGEE M708 Buttons not working.

emwjacobson, 2020-03-28:
OS: Manjaro - GNOME
Digimend Version: 9.r75.gcaf0fdd-1 ([Built from AUR](https://aur.archlinux.org/packages/digimend-kernel-drivers-dkms-git))

The problem I am having is actually getting the button presses to do anything.

Xorg config:
```
$ cat /etc/X11/xorg.conf.d/52-tablet.conf 
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "28bd:0907"
EndSection
```

USB ID is matching:
```
$ lsusb
...
Bus 002 Device 007: ID 04ca:7035 Lite-On Technology Corp. Integrated Camera
** Bus 002 Device 010: ID 28bd:0907  10 inch PenTablet **
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
...
```

Tablet shows up in the xsetwacom list.
```
$ sudo xsetwacom --list
10 inch PenTablet Mouse touch   	id: 9	type: TOUCH     
10 inch PenTablet stylus        	id: 10	type: STYLUS
```

The config file I made:
```
$ cat configTablet.sh 
#!/bin/sh

xsetwacom set "10 inch PenTablet Mouse touch" button 1 "key b"
```

My problem is just having having buttons do anything. Something as easy as just typing the letter "b". But when I click any button at all, nothing happens. I have attempted numerous things, but nothing has worked.

Things do happen **BEFORE** I run configTablet.sh (eg. whatever defaults the tablet has) and after running it, nothing happens at all.

I **do** have _xf86-input-wacom_ and _libwacom_ installed. Could there be a possible conflict?

nic3-14159, 2020-03-28:
The M708 had a hardware refresh, which is not fully supported at the moment. See #227 and [M708 v2 tablet status](https://digimend.github.io/tablets/Ugee_M708_v2/).
By the way, xf86-input-wacom doesn't conflict with libwacom.

teras, 2021-10-02:
Just to make things clear, what works for me and what not, the pen down/movement etc works, as well the lower button. Only the upper button + the 8 onboard buttons don't work.

leorv, 2024-08-23:
It happens the same way for me as it does for my companions. The 8 buttons on the table are not working. Does anyone have a solution?
My tablet: UGEE M708 V1.
Debian 12 GNU/Linux.