DIGImend/digimend-kernel-drivers #366: Support for XP-Pen Deco Pro M Tablet

rastislav-janci, 2020-03-30:
I wold like to properly use [XP-Pen Deco Pro](https://www.xp-pen.com/product/432.html) tablet. Is there work being done on the XP-Pen Deco Pro Medium? The tablet, touchpad and stylus are recognized, and I can draw with pressure in gimp and krita, but I cannot assign any buttons, the wheel or the touchpad. Here are the debug outputs:

[XP-PEN_DecoPro-descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402580/XP-PEN_DecoPro-descriptors.txt)
[XP-PEN_DecoPro-hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402582/XP-PEN_DecoPro-hid_report_descriptors.txt)
[XP-PEN_DecoPro-keys_top_left_to_bottom_right_bottop_rows.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402583/XP-PEN_DecoPro-keys_top_left_to_bottom_right_bottop_rows.txt)
[XP-PEN_DecoPro-keys_top_left_to_bottom_right_top_rows.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402585/XP-PEN_DecoPro-keys_top_left_to_bottom_right_top_rows.txt)
[XP-PEN_DecoPro-pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402586/XP-PEN_DecoPro-pen_coords.txt)
[XP-PEN_DecoPro-pen_input_keys.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402587/XP-PEN_DecoPro-pen_input_keys.txt)
[XP-PEN_DecoPro-pen_input_no_touch.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402588/XP-PEN_DecoPro-pen_input_no_touch.txt)
[XP-PEN_DecoPro-pen_input_touch.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402589/XP-PEN_DecoPro-pen_input_touch.txt)
[XP-PEN_DecoPro-pen_presure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402591/XP-PEN_DecoPro-pen_presure.txt)
[XP-PEN_DecoPro-pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402592/XP-PEN_DecoPro-pen_tilt.txt)
[XP-PEN_DecoPro-phy_wheel_scrolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402594/XP-PEN_DecoPro-phy_wheel_scrolling.txt)
[XP-PEN_DecoPro-probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402595/XP-PEN_DecoPro-probe.txt)
[XP-PEN_DecoPro-touch_pad_circular_scrolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402596/XP-PEN_DecoPro-touch_pad_circular_scrolling.txt)
[XP-PEN_DecoPro-touch_pad_left_right.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402597/XP-PEN_DecoPro-touch_pad_left_right.txt)
[XP-PEN_DecoPro-touch_pad_up_down.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4402598/XP-PEN_DecoPro-touch_pad_up_down.txt)

OdinVex, 2021-08-09:
I have the Small. They now also have a Bluetooth version of S and M, SW and MW.

JoseExposito, 2022-07-10:
Hi!

I just added support for the XP-PEN Deco Pro Small tablet to mainline kernel:
https://lore.kernel.org/linux-input/20220710175043.192901-1-jose.exposito89@gmail.com/T/

I don't own the tablet. Instead, I built a modified version of DIGImend and the user who reported the issue tested it for me:
https://gitlab.freedesktop.org/libinput/libinput/-/issues/738#note_1453926

Your tablet (XP-Pen Deco Pro M) should be very similar. If you could test the code, I'll add support for it in the kernel as well.

Could you test these drivers?

First, download the source code: [digimend-kernel-drivers.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/9079721/digimend-kernel-drivers.tar.gz)
The source code contains the code from DIGImend plus the changes required to add support for your tablet.

In order to install it, you'll need to install the kernel headers:

```
$ sudo apt-get install -y "linux-headers-$(uname -r)"
```

After extracting the code, cd into the source code and:

```
$ make
$ sudo make install
```

You can uninstall it running:

```
$ sudo make uninstall
```

More detailed instructions are available in the [DIGImend README file](https://github.com/DIGImend/digimend-kernel-drivers#digimend-kernel-drivers).

JoseExposito, 2022-07-11:
Ooops, I forgot to ping the people who reacted to the first message and won't get notified: @rastislav-janci @OdinVex @yohanboniface @nek0 @fabienengels @BloodyMess @skandalfo @okitoxo @kanashimia 

Please check my previous message

OdinVex, 2022-07-11:
> Please check my previous message
The UEFI on my AIO doesn't allow adding keys (yet...), so I won't be able to test until the mainline hits Manjaro for my home PCs.

OdinVex, 2022-07-11:
@JoseExposito, I've tested it, and it shows my device under xsetwacom list, works in Krita, pressure, et cetera, all buttons too, but the ring and pad do not. I was hoping for a mouse-mode, but that too doesn't work (or is that off out-of-box?). I did need to add my lsusb to my 70-wacom.conf (Linux Mint) so that it would be handled.

JoseExposito, 2022-07-11:
Hi @OdinVex , thanks a lot for the quick test.

> the ring and pad do not [work]

Both the dial (ring) and touchpad should work. If not, something unexpected happened.

Could you run `sudo libinput record` and copy/attach the output after selecting the interface that ends in "Pad", please?

While running the command, if you press the tablet buttons or use the dial you should see the events on the terminal.

The XP-PEN firmware expose a string indicating the tablet capabilities, including the kind of frame. It could possible that the firmware is not setting a byte as expected.

> Edit: Ring and pad works with xppentablet app.

Is the device working then? Or did you need to tweak some settings or similar?

OdinVex, 2022-07-11:
> Hi @OdinVex , thanks a lot for the quick test.
>
>
> Both the dial (ring) and touchpad should work. If not, something unexpected happened.
>
> Could you run `sudo libinput record` and copy/attach the [...]

I have a ton of devices that interact with libinput all the time, too difficult to sort (hardware developer). As for the ring and pad, it only worked with the xppentablet app *.deb from XP-Pen. If I remove the digimend driver compile from above, I can use the ring and pad. This could be something on my end, I do a ton of libinput and event overrides for so many things.

OdinVex, 2022-07-11:
I'm using the non-BT version via USB.

OdinVex, 2022-07-11:
@JoseExposito, I did, the Ring and Pad do not work *as well*. :/ I may or may not have once reverse-engineered the XP-Pen Windows software and found abs/rel wasn't controlled by the device, but rather the driver. As for why mine may not work, it could be the various libinput configs, maybe a stray udev, or the xppentablet app.

Danthe-GH, 2023-01-20:
@JoseExposito 
Hello. I use an XP-Pen Deco Pro M
It works well for me, but xsetwacom doesn't detect it
when I set "/etc/X11/xorg.conf.d/50-tablet.conf"  
`xsetwacom list
11 inch PenTablet stylus                id: 13  type: STYLUS    
11 inch PenTablet Pad pad               id: 14  type: PAD  
`
the pad works, but when I use the touch the whole tablet stops working, I have to disconnect it and it just works again until the touch is used.

JoseExposito, 2023-01-21:
Hi @Danthe-GH ,

Support for the XP-Pen Deco Pro M has been queued to be released for Linux kernel 6.3:
https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/log/?h=for-6.3/uclogic

You can try to take those patches and compile your own kernel with them applied. Hopefully, your device should be fully supported. There are multiple guides online explaining how to do it, it changes a bit depending on the distro you are using.

Danthe-GH, 2023-01-22:
> Hola@Danthe-GH,
> 
> El soporte para XP-Pen Deco Pro M se ha puesto en cola para su lanzamiento para Linux kernel 6.3: [https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/log/?h =para-6.3/uclogic](https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/log/?h=for-6.3/uclogic)
> 
> Puede intentar tomar esos parches y compilar su propio núcleo con ellos aplicados. Con suerte, su dispositivo debe ser totalmente compatible. Hay varias guías en línea que explican cómo hacerlo, cambia un poco dependiendo de la distribución que estés usando.

thanks for the reply, but I only see support for Deco Pro S (0x0909), Deco Pro Sw (0x0933) and Deco Pro MW (0x0934).
No support for Deco Pro M (0x0904)

> Hi @Danthe-GH ,
>
> Support for the XP-Pen Deco Pro M has been queued to be released for Linux kernel 6.3: https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/log/?h=for-6.3/uclogic
>
> You can [...]

Thanks for the reply, but I only see support for Deco Pro S (0x0909), Deco Pro Sw (0x0933) and Deco Pro MW (0x0934).
No support for Deco Pro M (0x0904)

JoseExposito, 2023-01-23:
Ah sorry, the names are too similar and I mixed them in my head.

Could you test this driver, please? [digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/10482375/digimend-kernel-drivers.zip).

It includes the latest changes from the mainline kernel and support for your tablet.

You can follow the instructions available in [this comment](https://github.com/DIGImend/digimend-kernel-drivers/issues/366#issuecomment-1179784243).

I added your tablet ID to the xorg.conf file present in DIGImend, so hopefully you won't need to change your local config.

> but when I use the touch the whole tablet stops working, I have to disconnect it and it just works again until the touch is used

Does this happen when you touch the tablet with the pen? Or when you touch the circular touchpad present on the left hand side of the tablet?

This issue reminds me to https://gitlab.freedesktop.org/libinput/libinput/-/issues/839. The difference is that this user was experimenting issues after using the stylus button.

Could you attach the output of `sudo libinput record` as a text file, please? You should see 3 interfaces for your tablet (Pen, Pad and Mouse) and events when interacting with them.

Danthe-GH, 2023-01-23:
The problem happens when I touch the touchpad, the whole tablet stops responding, however, when libinput is executed, the tablet reacts and the crash does not happen.
[Mouse.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10483653/Mouse.txt)
[Pad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10483654/Pad.txt)
[Pen.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10483655/Pen.txt)

JoseExposito, 2023-01-28:
The kernel side seems to be working as expected. The only problem that I noticed is that the pad udev properties:

```yaml
  udev:
    properties:
    - ID_INPUT=1
    - ID_INPUT_KEY=1
    - LIBINPUT_DEVICE_GROUP=3/28bd/904:usb-0000:0a:00.3-2
```

Should include:

```yaml
    - ID_INPUT_TABLET=1
    - ID_INPUT_TABLET_PAD=1
```

You can solve it adding your device to libwacom:
https://github.com/linuxwacom/libwacom#adding-tablet-descriptions-to-libwacom

About the issue with the tablet not responding after using the touchpad, it'd be interesting to test it again after adding your device to libwacom. Also, if you could run `sudo libinput debug-events --verbose`, press a pad button, use the touchpad and then press again the same button and use the stylus and attach the output, we'll have more information about what is going on.

For reference, this is a simplified overview about how libinput's tools work:

```
kernel -> libinput record -> libinput -> libinput debug-events -> apps
```

You output of `libinput record` does not show any issues in the kernel output. `sudo libinput debug-events --verbose` will help us to understand if the problem is with libinput.

Anyway, I'd try to add your device to libwacom, as it might fix the issue.

Anerag, 2025-02-18:
> Could you test this driver, please? [digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/10482375/digimend-kernel-drivers.zip).

I had the same problem where the tablet would die after touching the touchpad and this fixed it. Actually, I took the relevant commit from the zip and applied it to the current master.
But now when running `libinput debug-events --verbose` and touching the touchpad I get this which is the same as turning the dial clockwise.
```
event21  TABLET_PAD_DIAL             +1.958s   dial 0 delta -120.00 (mode 0)
```

This might be relevant.
```
event19 - 11 inch PenTablet Mouse: is tagged by udev as: Mouse Tablet
event19 - 11 inch PenTablet Mouse: device "11 inch PenTablet Mouse" (28bd:0904) is not known to libwacom
event19 - 11 inch PenTablet Mouse: libinput bug: missing tablet capabilities: xy pen btn-stylus resolution. Ignoring this device.
event19 - 11 inch PenTablet Mouse: device is a tablet
event19 - failed to create input device '/dev/input/event19'
```

I then used `xsetwacom` to disable the dial which also disabled the touchpad which I was going to do anyway since I keep bumping into them accidentally. Everything else seems to working fine. Would like to see this merged eventually so that I don't have to deal with dkms.