DIGImend/digimend-kernel-drivers #368: Feature support for the HUION 430P

simos, 2020-04-04:
Thanks for your work on DIGImend and the support for the HUION drawing tablets!

I want to write an article specifically on using the 430P in Linux, on Ubuntu 20.04 (with Linux kernel 5.4).
The 430P appears to work out of the box with this kernel (`uclogic` driver), thanks to patches in 2015. 

Should I suggest to the users that for this specific model, the Linux support in 5.4 is sufficient when using the 430P? And they do not need to install DIGImend? Does DIGImend have newer end-user features that would benefit users of that drawing tablet?

spbnick, 2020-04-05:
Hello @simos, yes, no additional drivers need to be installed for H430P, the stock kernel should support it. You might want to install the Wacom drivers for X.org, if you want extra features, like button assignment, pressure curves, and so on. See https://github.com/DIGImend/digimend-kernel-drivers#enabling-wacom-xorg-driver for a few tips.

closed 2020-04-26 by spbnick

simos, 2020-05-15:
Here is my post on using this tablet in Ubuntu 20.04 LTS,
https://blog.simos.info/how-to-setup-the-huion-430p-drawing-tablet-on-ubuntu-20-04/