DIGImend/digimend-kernel-drivers #370: Mention the need for Wacom Suppress 0 setting for Huion touch ring/strip in README.md

spbnick, 2020-04-05:


p-velasco, 2020-04-18:
Would be great to also mention that gnome doesn't currently work because it is intercepting the button 4 and 5 actions, and try to find a workaround for it...

spbnick, 2020-04-18:
Yeah, that thing. I'm not sure if there's a workaround for that :(

closed 2020-04-26 by spbnick

different55, 2020-05-07:
Could there be a userspace utility for it? I have no idea what I'm talking about here but Steam Input is able to generate scroll events with the Steam (or any other) Controller.

p-velasco, 2020-05-10:
I've been looking into this with some detail using a Huionl HS610 with a Debian Buster and gnome 3.30, and after some testing:
evtest as root shows four devices:
-"HUION Huion Tablet Pen"
-"HUION Huion Tablet Pad"
-"HUION Huion Tablet Touch Ring"
-"HUION Huion Tablet Dial"

evtest gets wheel events from the "HUION Huion Tablet Touch Ring", button events from the "Pad" device and motion events from the "Pen" device, but no events from the "Tablet Dial" device.

xinput lists the four devices, but xsetwacom doesn't list the "Dial" device.

what's interesing is that running evtest without root privileges sees the "Touch Ring" device and its events, but no other input device. An "ls" on /dev/input shows that it's the only device with a "+" marking the access mode (ACL??)

"libinput debug-events" only lists the "Pen" and "Dial" devices, and also shows in red that "event11 - HUION Huion Tablet Pad: libinput bug: missing tablet capabilities: pen resolution. Ignoring this device.".
The same happens with "libinput list-devices"

I also see that while the other three devices get this kind of message on Xorg.log:
"HUION Huion Tablet Pad: Applying InputClass "evdev tablet catchall""
and
"HUION Huion Tablet Pad: Applying InputClass "libinput tablet catchall""
The Touch Ring only gets the "Huion tablets with Wacom driver", meaning (I think) that libinput and evdev are not matching the "Touch Ring" based on "MatchIsTablet" on the xorg.conf files

So (and I'm just speculating) its seems that libinput is not seeing the touch ring device, and so ignores the "button" mappings of the wacom driver, maybe due to some strange thing happening with the /dev/input files or some udev rule?

Moreover, I have been unable to make the Touch Ring work in xfce, Ubuntu focal, altought I only spent a few minutes on it, but I see the same behaviour as described for the Debian Buster host.

commit fe0f38c8 (2020-04-26): uclogic: Set Suppress to zero for touch controls

Fixes #370