DIGImend/digimend-kernel-drivers #372: HUION 610proV2, no answer in events

JMMouchel, 2020-04-06:
Hi, first thanks for this excellent work.
I recently bought a tablet HUION 610proV2 from Amazon, and it seems I face some problems.
I have a Debian 10 distribution and I installed version 9 of the drivers using the .deb file.  Results of most tests compare well with those of Browncoat97 (Dec 2018), with a difference in usb numbers (256c:006d instead 256c:006e) which should not be a problem, and some other differences I guess minor but which I could describe in more details if needed.
A first problem is that the kernel drivers do not load at boot (although the tablet is connected), so I used modprobe -a  to load them so that they properly appear in lsmod
lsmod | grep hid gives :
hid_kye                16384  0
hid_polostar           16384  0
hid_viewsonic          16384  0
hid_uclogic            28672  0
hid_logitech_hidpp     40960  0
hid_logitech_dj        24576  0
hid_generic            16384  0
usbhid                 57344  0
hid                   135168  8 hid_polostar,usbhid,hid_uclogic,hid_kye,hid_generic,hid_viewsonic,hid_logitech_dj,hid_logitech_hidpp
usbcore               294912  6 ohci_hcd,ehci_pci,usbhid,hid_uclogic,ehci_hcd,ohci_pci

Then I have no signal from "/sys/kernel/debug/hid/0003:256C:006D.0027/cat events"  while moving the pen on the tablet, although "/sys/kernel/debug/hid/0003:256C:006D.0027/cat rdesc answers properly.
I also have no signal when using evtest and the tablet is not seen in xinput.

Any idea ? Is more information needed ?
Best regards,
JMM

vkurup, 2020-04-11:
Hi, I have the same tablet and had a similar problem on Ubuntu 18.04. I edited the file `/usr/share/X11/xorg.conf.d/50-digimend.conf` and changed 006e to 006d on line 9, then rebooted and things worked. Not sure if that was the same issue you were facing or not, though.

JMMouchel, 2020-04-11:
Oh yes, excellent. This worked. Thank you very much vkurup !!
This means that there is no need of the additional kernel modules proposed by digimend for this tablet.
I added 256c:006d to 50-digimend.conf on line 9 : `MatchUSBID "5543:006e|256c:006e|256c:006d"`

closed 2020-04-11 by JMMouchel