DIGImend/digimend-kernel-drivers #374: Artisul D16 [5543:0047]

phelipefox, 2020-04-09:
I want to help with the project and intercepted the information from an Artisul D16 tablet [5543: 0047] UC-Logic via usbhid-dump!

[usbhid-dump Artisul D16 [5543;0047].tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/4453398/usbhid-dump.Artisul.D16.5543.0047.tar.gz)

**Notes:** The tablet works with a generic driver, but only the pressure and button below the pen. I use the command "**_xsetwacom set" UC-Logic ARTISUL D16 Pen (0) "MapToOutput HEAD-1_**" to lock the pen only on the tablet screen and it works. When I install the Digimend drivers the tablet starts to appear in the list of "**_xsetwacom --list devices_**", but it stops working! :(

Official WebSite of Tablet: https://www.artisul.com/

beyondoblivion, 2020-05-25:
I have the same tablet, and I've mostly decoded the raw data from usbhid-dump last night. I have noticed that the tablet works out of the box under Ubuntu 16.04 (buttons weren't tested), but not under Ubuntu 18.04, nor Linux Mint 19.3. I haven't tested any other distros.

```
Let's call the raw data from usbhid-dump 'unsigned char* rawData'

rawData[0] == Always 0x07??
rawData[1] are flags which break down to:

1 << 0 == Stylus down (Left click)
1 << 1 == Stylus button 1 (Middle click)
1 << 2 == Stylus button 2 (Right click)
1 << 3 == Always 0??
1 << 4 == Always 0??
1 << 5 == isButtonOrDialActive
1 << 6 == isStylusNearTablet?? or isActive?? (not entirely sure what this is for)
1 << 7 == Always 1??

if((rawData[1] & (1 << 5)) != 0){
	button or dial is active

	rawData[2] == Always 0x02??
	rawData[3] == 0x01 if button, or 0x10 if dial

	dial: (the dial itself doesn't have any buttons except for at its center)
	rawData[4] == 0x01 if ticked clockwise, or 0xff if ticked counter-clockwise (0 after the tick)

	buttons:
	All of following bits are found in byte: rawData[4]

	I numbered the buttons from the top to the bottom (if the buttons are on the left)

	button 1: 1<<0
	button 2: 1<<1
	button 3: 1<<2
	button 4: 1<<7 (center of dial)
	button 5: 1<<3 (square button)
	button 6: 1<<4
	button 7: 1<<5
	button 8: 1<<6

	The bits are set when pressed (or dial starts to spin) and reset when released (or dial stops)

	The rest of the bytes after seem to all be 0
}else{
	Information about the stylus (16 bit values) (values are little endian)

	(Assuming the buttons are on the left)

	The values in the parenthesis' are the upper bounds, the lower bounds are all 0
	(The upper bounds for the x and y coordinates may go a bit above these bounds)

	rawData[2] and rawData[3] == x coordinates (>=0x866E (right))
	rawData[4] and rawData[5] == y coordinates (>=0x4B9D (bottom))
	rawData[6] and rawData[7] == pressure (==0x1fff)
}
```

I'm not sure what rawData[8] and rawData[9] are.

I don't have any experience writing drivers for Linux, but I'm thinking of at least attempting it.

beyondoblivion, 2020-05-26:
I turned on debugging earlier and this is what I found in '/sys/kernel/debug/dynamic_debug/control'

```
/home/atmos/Downloads/digimend-kernel-drivers-master/hid-uclogic-params.c:590 [hid_uclogic]uclogic_params_frame_init_v1 =_ "invalid response to enabling generic buttons: \042%s\042\012"
/home/atmos/Downloads/digimend-kernel-drivers-master/hid-uclogic-params.c:584 [hid_uclogic]uclogic_params_frame_init_v1 =_ "generic button -enabling string descriptor not found\012"
/home/atmos/Downloads/digimend-kernel-drivers-master/hid-uclogic-params.c:424 [hid_uclogic]uclogic_params_pen_init_v2 =_ "string descriptor with pen parameters seems to contain only text, assuming not compatible\012"
/home/atmos/Downloads/digimend-kernel-drivers-master/hid-uclogic-params.c:407 [hid_uclogic]uclogic_params_pen_init_v2 =_ "string descriptor with pen parameters is too short (got %d, expected at least %d), assuming not compatible\012"
/home/atmos/Downloads/digimend-kernel-drivers-master/hid-uclogic-params.c:399 [hid_uclogic]uclogic_params_pen_init_v2 =_ "string descriptor with pen parameters not found, assuming not compatible\012"
/home/atmos/Downloads/digimend-kernel-drivers-master/hid-uclogic-params.c:262 [hid_uclogic]uclogic_params_pen_init_v1 =_ "string descriptor with pen parameters has invalid length (got %d, expected %d), assuming not compatible\012"
/home/atmos/Downloads/digimend-kernel-drivers-master/hid-uclogic-params.c:254 [hid_uclogic]uclogic_params_pen_init_v1 =_ "string descriptor with pen parameters not found, assuming not compatible\012"
/home/atmos/Downloads/digimend-kernel-drivers-master/hid-uclogic-params.c:176 [hid_uclogic]uclogic_params_get_str_desc =_ "string descriptor #%hhu not found\012"
```

I don't know how to read the device descriptors, nor the UC-Logic parameters. I tried using hidrd-convert for the device descriptors, but I can't seem to get it to work.

As a side note, I have another yet different tablet with the same vendorID and productID, and that tablet works with the Digimend.

beyondoblivion, 2020-10-24:
I've been looking into fixing the bug that prevents the D16 from working with Digimend again, and so far, I haven't found anything that should cause any issues. The parameters (frame v1) for my D16 and UG-1910B are virtually identical. (Note: The UG-1910B works without a problem with Digimend) The only difference is the descriptors.

I have noticed that when ever I use the pen or press a button on the D16 (with Digimend installed), and check the debug output I find "usb 2-1.1: input irq status -75 received"

Any input would be greatly appreciated.

beyondoblivion, 2020-11-19:
It now works flawlessly with both of my tablets. Now I just need to figure out how to do a pull request

beyondoblivion, 2020-11-19:
#483 

Fixed

phelipefox, 2020-11-19:
Hello again! I performed a new test with the latest driver added **v10 Oct 4 via DKMS**, the kernel I am using is the **Linux 4.15.0-124-generic** (Ubuntu 18.04.4 LTS). I got the same results as before The pen and a button normally work, but the buttons on the pen display do not work. When I install the driver the pen display is displayed in the list ...

$ xsetwacom --list devices
UC-Logic ARTISUL D16 stylus     	id: 8	type: STYLUS    
UC-Logic ARTISUL D16 Pad pad    	id: 9	type: PAD  

.. but nothing works, not even the pen!

If you need any kind of information I am available! 😀

beyondoblivion, 2020-11-19:
I have a fork of Digimend on my profile with the fix applied, so you could use that until the fix has been officially applied.

phelipefox, 2020-11-19:
I tested the **beyondoblivion** modification and it worked great! All pen and display buttons! To lock the pen cursor only on the display, use **xsetwacom set "UC-Logic ARTISUL D16 stylus" MapToOutput HEAD-1**, because now the name has been changed from **UC-Logic ARTISUL D16 Pen (0)** to **UC-Logic ARTISUL D16 stylus**! 👍👍👍👍👍👍👍👍👍👍👍👍👍👍👍👍👍👍

mjdolanart, 2021-02-07:
beyondoblivion: Thank you so much for this ...works flawlessly ... thought this tablet was heading for e-waste :)