DIGImend/digimend-kernel-drivers #376: Gaomon s620 ubuntu

de1uxe, 2020-04-11:
Is this tablet supported on ubuntu 18.04? I want to use it for osu!

DavidHusicka, 2020-06-17:
I use Gaomon S620 for osu on my Linux system and it works. The care of this tablet is taken by one of Huion drivers in this repo. This tablet is reported to the kernel as a Huion tablet.

spbnick, 2020-08-01:
Closing, since the tablet is supported, and there's no response from the reporter.

closed 2020-08-01 by spbnick

attila-i-szabo, 2020-10-14:
> I use Gaomon S620 for osu on my Linux system and it works. The care of this tablet is taken by one of Huion drivers in this repo. This tablet is reported to the kernel as a Huion tablet.

Could you give some instructions? I installed the repo from the .deb file, plugged in the tablet, but it doesn't work. I get a USB device ID 256c:006d without any description, and it isn't recognised by xinput or anything else...

bigbigmdm, 2021-10-02:
Please see my repository in the [https://github.com/bigbigmdm/GAOMON-S620-in-LINUX](github)