DIGImend/digimend-kernel-drivers #378: Huion H610Pro v2 - pad buttons are not working

NicoPL, 2020-04-16:
Hi there,

I've just bought a Huion H610Pro v2, and try to make it work.
I installed the drivers, and wrote the `52-tablet.conf` file as detailed in there:
https://github.com/DIGImend/digimend-kernel-drivers/issues/26#issuecomment-263492989

However, before trying to map the pad keys I ran `xsetwacom --list` and got 
```
HUION Huion Tablet stylus               id: 10  type: STYLUS    
HUION Huion Tablet eraser               id: 16  type: ERASER
```
instead of
 ```
HUION PenTablet Pad pad id: 11 type: PAD
HUION PenTablet Pen stylus id: 12 type: STYLUS
```

I noticed a similar issue encountered by the HS610 users: https://github.com/DIGImend/digimend-kernel-drivers/issues/275#issuecomment-484893599

I don't know if this is related. As anybody manage to make the H610ProV2 work completely?

So far, in GIMP, with the Stylus tool I can draw only (no pressure sensitivity), and when pressing one of the 2 buttons on the pen it acts as the 'hand' grabbing my sheet to move it.

Thanks in advance for your help!

spbnick, 2020-04-17:
Could you try installing the [latest master](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip) instead?

agolks, 2020-04-18:
have used the latest master, and i get now:
HUION Huion Tablet Pen stylus           id: 9   type: STYLUS    
HUION Huion Tablet Pad pad              id: 10  type: PAD       
HUION Huion Tablet Touch Strip pad      id: 11  type: PAD       
HUION Huion Tablet Dial pad             id: 12  type: PAD

spbnick, 2020-04-18:
Now you can configure the buttons as hinted in the README.md.

agolks, 2020-04-18:
yes, works, too.
i've an issue with '[' and ']', though, which is replaced by 8 and 9 respectivly...

spbnick, 2020-04-18:
@agolks, you probably need to add a `Shift` to the sequence. There should be an example, either in the man page, or online.

spbnick, 2020-04-18:
Although, I'm not sure what keyboard layout you have, on US layout you don't need Shift, of course.

agolks, 2020-04-19:
german layout, would mean altgr. but altgr is not a valid modifier:
```
ich@surfer4:~$ xsetwacom --set 'HUION Huion Tablet Pad pad' Button 10 "key +altgr 8 -altgr"
Invalid key '+altgr'.
Cannot parse keyword '+altgr' at position 2
```
some research suggests that i have to modify my layout using setxkbmap. otherwise i'll just set another shortcut in krita.
at least, the driver works for ;)

NicoPL, 2020-04-19:
Thank you @spbnick and @agolks , I tried the latest master driver and now I get:
```
HUION Huion Tablet Pen stylus           id: 15  type: STYLUS    
HUION Huion Tablet Pad pad              id: 16  type: PAD       
HUION Huion Tablet Touch Strip pad      id: 19  type: PAD 
```
without the need to create the file `/etc/X11/xorg.conf.d/50-tablet.conf`

Strangely I do not get `HUION Huion Tablet Dial pad             id: 12  type: PAD` like @agolks , but when I open GIMP, I get 4 entries in 'Configure Input Device': Pen stylus, Pad pad, Touch Strip pad, and Dial pad.

Now I have to understand what is what (Pad pad, Touch strip pad, and Dial pad) and what are the numbers for each of the buttons.
I believe I have to follow https://github.com/DIGImend/digimend-kernel-drivers#configuring-wacom-xorg-driver and '[2] CONFIGURING AND MAPPING TABLET:' on https://github.com/DIGImend/digimend-kernel-drivers/issues/26#issuecomment-263492989 is that right?

spbnick, 2020-04-20:
@NicoPL, glad to hear it worked for you!
The H610 will only have the "Pen stylus" and "Pad pad" devices working, please ignore the rest. You can configure the buttons on the "Pad pad" device, according to tips you mention.

@agolks, I can see "altgr" in xsetwacom source code, not sure why it's not picked up and not e.g. listed in the output of `xsetwacom list modifiers` output. I'd suggest you ask the developers.

spbnick, 2020-04-20:
@agolks, the "altgr" has been added in xf86-input-wacom-0.35.99.1, your distribution might be using an older version.

agolks, 2020-04-20:
i'm running debian testing, bullseye/sid.  you're right, my xsetwacom is
```
ich@surfer4:~$ xsetwacom --version
0.34.99
```
will try to get latest version

NicoPL, 2020-04-21:
@spbnick Thank you

As I'm on a French layout keyboard (AZERTY), I mapped the keys as follow to make 7 of the 8 buttons work:
```
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 1 "key +ctrl +z -z -ctrl" # Back button
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 2 "key +shift e" # Eraser
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 3 "key n" # Pen
#xsetwacom --set 'HUION Huion Tablet Pad pad' Button 3 "key p" # Paintbrush
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 8 "key +shift +" # Zoom in
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 9 "key -" # Zoom out
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 10 "key +altgr ]" # Increase brush size
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 11 "key +altgr [" # Decrease brush size
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 12 "button +12" # Grab tool?
```

I still have a couple of questions:
* what is the button 12 for? (I do not find the right name of the keyboard shortcut in GIMP to make it work properly)
* the pressure sensitivity does not work for me. In GIMP it seems activated though (I can modify the curve in Input devices). And even when trying `xsetwacom --set 'HUION Huion Tablet Pen stylus' PressureCurve "5 10 90 95"` it does nothing. Is there anything I can do to find out what is the issue?
* how are the soft keys working? Do I have to map them as well using the same technique? or is it impossible to use them as for the H1060 https://github.com/DIGImend/digimend-kernel-drivers/issues/322#issuecomment-558141653 ?

kortschak, 2020-04-25:
Was having same problem. Also fixed by building from master.

spbnick, 2020-05-17:
@NicoPL, button 12 is for tablets which have it, yours doesn't so please just ignore it.
Soft keys are not supported, because they're mostly are a marketing gimmick, and are useless. They take your attention away from the screen, where your work is. There's a reason why Wacom tablets don't have them.

NicoPL, 2020-05-18:
I confirm that my tablet has got the 12th button, I just don't know which key it corresponds to.
There's a hand drawn on it, so I assume it should be attached to a "grab" tool, or the "hand" tool. That's why I tried to map it as the "mouse middle click" without success.
How guys to you usually use it?
[image]

Ok for the soft keys, understood then. Cheers

spbnick, 2020-05-18:
@NicoPL, ah indeed, I miscounted. The middle mouse button would be number two. Try this:

    xsetwacom --set 'HUION Huion Tablet Pad pad' Button 12 "button +2"

spbnick, 2020-06-06:
Closing due to no response from reporter. Please reopen if the issue is still present.

closed 2020-06-06 by spbnick