DIGImend/digimend-kernel-drivers #379: Huion H640P - wrong PID name

canorve, 2020-04-21:
I just bought a HUION H640P for my Dell laptop with elementary OS. 

The driver didn't work at first, but following your tutorial 
"How to locate a failure", I notice that the PID is wrong. Your tablet driver support says that
for my tablet is  256c:006e, instead of 256c:006d after checking with "lsusb" and "usbhid-dump".

So, I changed line 9 in **xorg.conf** file to  
`MatchUSBID "5543:006e|256c:006d"`
and the tablet worked

By the way, the issue is resolved, but I wanted you guys to notice this problem.

spbnick, 2020-04-21:
Thank you for the report, Chris. The 6d PID is used by newer models, and the latest development version of the drivers should know about it. Which version were you using?

canorve, 2020-04-21:
I assume it is the last one:  Version 9

chexbox, 2020-04-22:
Thank You! On Ubuntu (Digimend v9) It didn't have a section for the Huion H640P at all. I got the PID set, and it seems to be working. My tablet, however doesn't seem to report the x-coordinate on about 1/2 the tablet.

ruhsontorugut, 2020-04-23:
> Thank You! On Ubuntu (Digimend v9) It didn't have a section for the Huion H640P at all. I got the PID set, and it seems to be working. My tablet, however doesn't seem to report the x-coordinate on about 1/2 the tablet.
I don't really understand English
But maybe your tablet on phone mode so only half of tablet register your input correctly
Try press button 1 + button 5 to enter computer mode

closed 2020-04-24 by canorve

JoseKling, 2020-08-15:
I have the same problem. But I don't know how to change this configuration.
My OS is Manjaro KDE.
Thanks