DIGImend/digimend-kernel-drivers #381: Huion H950P on dual-monitor

adriansnetlis, 2020-04-26:
Hey,
I just recently decided to try my Huion Inspiroy H950P on my Arch Linux installation. Got this package from aur and installed it, so far it's working well. Pressure sensitivty and tilt work just fine and it feels just as accurate and nice as on Windows. However, there are a couple problems. One of them is that I don't know how to configure the 8 buttons, but that's less of an issue since I can comfortably use keyboard for shortcuts while drawing. However, the bigger issue is the monitor surface mapping on the tablet. I have two 1080p monitors next to each other, and the tablet stretches it's surface to fit both monitors, meaning that on horizontal axis the movement is twice as sensitive. I would love to be able to set the tablet to only work on the primary monitor.

Here's some data that might be useful:
`uclogic-probe`
```
S C8 13 03 A8 AC 00 D0 6B 00 FF 1F D8 13 03 08 00 00 04 00 40
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 54 00 31 00 37 00 32 00 5F 00 31 00 39 00 30 00 33 00 32 00 35 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00
```

`libinput list-devices`
```
event25 - HID 256c:006d Pad: libinput bug: missing tablet capabilities: pen resolution. Ignoring this device.
```
(this kind of seems concerning, btw)

spbnick, 2020-04-27:
Hi @adriansnetlis, I suggest you use the Wacom X.org drivers with your tablet, and configure your buttons and display mapping with xsetwacom. See https://github.com/DIGImend/digimend-kernel-drivers/#xorg-drivers

adriansnetlis, 2020-04-27:
Thank you for pointing this out to me. It absolutely worked. I am now able to freely map the tablet to certain parts of screen and adjust the express keys and stylus settings (I am also using the GUI tool to make this more simple which works, too). Seems like this tablet works just fine with the wacom drivers.