DIGImend/digimend-kernel-drivers #383: Tablet pressure stops working after a re-plug

hyiltiz, 2020-04-27:
### Goal
Tablet works after every re-plug after ...
- Ideally: some one-time only configuration has been done (maybe after writing some `udev` rules). 
- Compromise: running some command that doesn't kill X11 before/after the re-plug (maybe manual unloading then reloading some kernel modules/drivers, flushing `udev` rules)
- Acceptable: running some command (including killing X11) before/after the re-plug (maybe manual unloading then reloading some kernel modules/drivers, flushing `udev` rules)
- Unacceptable: a reboot

### Problem
I was able to set up my tablet (documented #296) without `digimend` drivers; even after uninstalling the `digimend` drivers, simply including the appropriate `50-tablet-xorg.conf` file is sufficient to make it work, i.e. enable pressure sensitivity for the tablet after a fresh reboot [Xorg.0.log](https://termbin.com/4nzq)  in Debian Testing (tried kernels from 4.9 to 5.5.0). However, once I plug out the tablet USB, then there is nothing I can do to make it work upon plugging back in. 

### Things I have tried 
Things I have tried before and after the re-plug in each combination:

1. Tried unloading / unload then reloading back in kernel drivers using `modprobe -a/-r`: `hid`, `usbhid`, `hid-generic`, `hid-uclogic`, `usb-common`, `usbcore` (Warning: most of them also leaves the keyboard and mouse unusable so be prepared);
2. Tried commenting out `50-tablet-xorg.conf` file
3. Tried removing `xserver-xorg-input-wacom` package
4. Tried installing `digimend` drivers
5. Flush `udev` rules with: `udevadm control --reload-rules && udevadm trigger`

My hypothesis is that apparently the order that loads the drivers is important. `hid-uclogic` driver from `digimend` is not required for my tablet as simply including the `50-tablet-xorg.conf` file is sufficient. However, even during a working state, `usbhid` seems to be driving the tablet rather than `wacom` regardless of what `Xorg.0.log` file included above indicates:

```
$ /sys/kernel/debug/usb/devices
...
T:  Bus=03 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  5 Spd=12   MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=5543 ProdID=0061 Rev= 0.00
S:  Manufacturer=UC-Logic
S:  Product=A640
C:* #Ifs= 3 Cfg#= 1 Atr=a0 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=5ms
I:* If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
E:  Ad=82(I) Atr=03(Int.) MxPS=   8 Ivl=4ms
I:* If#= 2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=83(I) Atr=03(Int.) MxPS=   8 Ivl=8ms
...
```

[Here](https://termbin.com/km6d) is a `dmesg` after a reboot (tablet working) then a re-plug (tablet stops working).

Would greatly appreciate if anyone could help.

hyiltiz, 2020-05-01:
After some extensive research on kernel modules `hid-uclogic` `hid-generic` `wacom`, USB input interfaces `usbhid`, X input drivers `libinput` `wacom`, I found a solution. But first, it was hard to really identify the source of the problem. Once I got that right, fixing was somewhat straightforward.

### Problem
I checked `/sys/devices` for reliable source of information. After a fresh reboot, the `eventX` file corresponding to a directory under `sysfs` which contained a device named `UC-Logic A640` responded to physical stylus input events, when inspected with `xxd /dev/input/eventX`; however, after a re-plug (with or without killing Xorg), a different `eventY` file corresponding to a different   directory for a device named `UC-Logic A640 Mouse` responded instead. Notice the change of the device name and sysfs directory for the same physical stylus.

### Solution
I created a `udev` rules by repeatedly re-plugging/killing Xorg/rebooting while checking which `/dev/input/eventX` responded to the stylus inputs, and traverse the `sysfs` to locate its directory, then logged down `udevadm info -a -p /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:5543:0061.000B/input/input31/event10` (note the path name changes every time), finally comparing them until I could find the following matching rules:

```

```

p.s. `wdaemon` may offer an alternative solution, but I'd rather keep one software out of the stack.
p.p.s `hid-uclogic` was never loaded somehow; it seems installing `digimend` via `dkms` was unnecessary.

### References
On manually configuring X input driver for the tablet:
https://github.com/linuxwacom/xf86-input-wacom/wiki/Configuring-X

On manually detecting the input (sub)devices:
https://github.com/linuxwacom/input-wacom/wiki/Fixed-device-files-with-udev
http://reactivated.net/writing_udev_rules.html

If you feel advantageous, maybe try:
https://github.com/linuxwacom/xf86-input-wacom/wiki/wdaemon

hyiltiz, 2020-05-01:
Can we learn something from this, and maybe abstract it into a Wiki tutorial page?

spbnick, 2020-05-17:
@hyiltiz, I did a quick change to try and support your tablet with DIGImend drivers, please see if it works: https://github.com/spbnick/digimend-kernel-drivers/archive/add_support_for_parblo_a640.zip

PR #393 is opened for this.

hyiltiz, 2020-05-17:
Thanks @spbnick ! I'll test it out and let you know. It turns out that my solution above wasn't robust at all; my `udev` rules wasn't robust against plugging it into a different USB port or the order of plugging in other USB devices (e.g. mouse). Glad to see it supported by `digimend` drivers.