DIGImend/digimend-kernel-drivers #385: Diagnostics for Manufacturer "UC-Logic", Model name "Parblo A640"

hyiltiz, 2020-04-29:
According to the following [project diagnostics](http://digimend.github.io/support/howto/trbl/diagnostics/) page, I have collected the diagnostics. I also included an Xorg.0.log and other system log files after installing this project's dkms from source, setting up the `wacom` X driver and *before* doing a hotplogging (replug) test. The tablet has a pad plus a stylus. There are four buttons on the pad, and two buttons on the stylus, plus the pressure sensitive stylus tip. The frame buttons are pressed in top left, bottom left, top right, bottom right order. 

Kernel `usbhid` driver (version `5.5.0-2-amd64`) successfully creates `event` files in the `/sys/kernel/debug/hid/XXXX/` directories for both the stylus and the pad that responds correctly to all physical input actions. 

Not specifying any X driver, which defaults to `libinput`, leaves stylus pressure unusable; specifying `wacom` as the X driver will make it "partially work". Partially work means: 

- `xinput test` (or any application such as Krita, MyPaint, GIMP etc.) doesn't respond to the upper button on the stylus;
- `xinput list` shows both a Mouse and a Stylus devices, and the Stylus device doesn't respond to any input action when tested with `xinput test`;
- No items shown in `xinput list` responds to button presses on the pad when tested with `xinput test` or other apps
- hotplugging doesn't work; 
  a) after plugging out then re-plugging back in, kernel module `usbhid` changes the stylus input events from the original `/dev/input/event6` to `/dev/input/event8` (sometimes to `/dev/input/event10`) although the same USB port was used and no additional USB devices were inserted.
  b) Krita and any KDE apps will crash instantly when the mouse pointer was on them and any stylus action was performed
  c) `xev` and `xinput test` stops responding to stylus pressure inputs
[UC-Logic-ParbloA640-5543_0061.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/4555222/UC-Logic-ParbloA640-5543_0061.tar.gz)

Product page: https://www.parblo.com/products/a640?variant=9694024106031

Manufacturer's provided driver for Windows and Mac:
https://support.parblo.com/hc/en-us/articles/360011942153-A640-Driver
https://support.parblo.com/hc/en-us/articles/360020073973-A640-Driver

DONofDESTROY, 2020-05-14:
I never achieved the pressure senstivity of this tablet but as You said When I was using manjaro kde version when I use the xorg wacom conf and with any action from the pen causes Black screen sometimes the just another workspace now I switched to arch and that black screen problem doesn't occur but it slows them down and the it is not reporting the pen hovering atleast for me

eduard-sukharev, 2020-11-20:
I've got this tablet too and it works fine on Windows, but on Ubuntu 20.04 LTS it didn't track stylus hovering motions up until the stylus touches the surface, but then it registers it as immediate pointer movement and click, which results in choppy motion.

Weirdly enough, it is kind of capable of tracking hovering motions right after it's connected to my laptop, but after the first "click" it seem to stick down to surface again and no hovering motions tracking.

I have installed your device drivers via v10 .deb package but it didn't change anything in that behavior. I then decided to collect all data required for debugging and adding support for this tablet (during which the tablet started working flawlessly up until I've re-plugged it).
Anyways, here goes the all required probes, in case you might need them:

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5574108/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5574109/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5574110/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5574111/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5574112/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5574113/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5574114/pen_tilt.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5574115/probe.txt)