DIGImend/digimend-kernel-drivers #386: 1060PLUS(2048) Huion H610pro can't set config #1, error -32

americapat, 2020-04-30:
Using the same digimend-driver git pull as of 4/29/2020 (version 10) for kernels younger than  approximately 5.2 the device probe fails with the error -32.  I was only able to check that Debian (SID) kernel 4.19.0-1-amd64 works, and that any kernel younger than 5.3.0-trunk-amd64 fail (the latest being 5.5.0-2-amd64 for me). Let me know if you need more details.
digimend, 10, 4.19.0-6-amd64, x86_64: installed
filename:       /lib/modules/4.19.0-6-amd64/updates/dkms/hid-uclogic.ko
srcversion:     5385100FF1B55429F5A439A
⎜   ↳ HID 256c:006e Pen stylus                	id=13	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad pad                   	id=14	[slave  pointer  (2)]

americapat, 2020-05-01:
5.6.0-1-amd64 kernel has same issue
[  160.780290] usb 2-10.1: New USB device found, idVendor=256c, idProduct=006e, bcdDevice= 0.00
[  160.780295] usb 2-10.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  160.786190] usb 2-10.1: can't set config #1, error -32

Iwyil, 2020-05-03:
Looks like a duplicate of #330 which is a duplicate of #318 
The problem seems resolved in kernel version 5.6.8. In the referenced issues you may find useful information. I resolved the same problem with an update. Don't know of which could be your best way to resolve the problem beyond updating or compiling the kernel.

americapat, 2020-05-15:
Hi Iwyil, thanks, 5.7.0-rc4 fixes things, but indeed you have to compile the kernel. Sorry for the delay. - Pat.

spbnick, 2020-05-17:
Closing as a duplicate.

closed 2020-05-17 by spbnick