DIGImend/digimend-kernel-drivers #387: Install fails on Ubuntu 20.04

yntelectual, 2020-04-30:
Hi installing from deb fails on latest Ubuntu LTS

```
Selecting previously unselected package digimend-dkms.
(Reading database ... 222993 files and directories currently installed.)
Preparing to unpack digimend-dkms_9_all.deb ...
Unpacking digimend-dkms (9) ...
Setting up digimend-dkms (9) ...
Loading new digimend-9 DKMS files...
Building for 5.4.0-28-generic
Building initial module for 5.4.0-28-generic
ERROR (dkms apport): unable to determine source package for digimend-dkms
Error! Bad return status for module build on kernel: 5.4.0-28-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: error processing package digimend-dkms (--install):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms
```

and the dkms log:

```
DKMS make.log for digimend-9 for kernel 5.4.0-28-generic (x86_64)
Št 30. apríl 2020, 09:32:00 CEST
make -C /lib/modules/5.4.0-28-generic/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
  LEX     scripts/kconfig/lexer.lex.c
/bin/sh: 1: flex: not found
make[3]: *** [scripts/Makefile.host:9: scripts/kconfig/lexer.lex.c] Error 127
make[3]: *** Waiting for unfinished jobs....
  YACC    scripts/kconfig/parser.tab.[ch]
/bin/sh: 1: bison: not found
make[3]: *** [scripts/Makefile.host:17: scripts/kconfig/parser.tab.h] Error 127
make[2]: *** [Makefile:594: syncconfig] Error 2
make[1]: *** [Makefile:704: include/config/auto.conf.cmd] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
make: *** [Makefile:22: modules] Error 2
```

Linux header were installed previously 
Tried installing from source, but run to similar error.

binarydepth, 2020-04-30:
Same here:  `installed digimend-dkms package post-installation script subprocess returned error exit status 10`

spbnick, 2020-04-30:
Looks like Ubuntu messed up `dkms` package dependencies, since they are no longer installing kernel build dependencies.

Take a look at https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel for instructions on installing build dependencies.

spbnick, 2020-04-30:
They give the following command for "Disco Dingo":
    sudo apt-get install libncurses-dev flex bison openssl libssl-dev dkms libelf-dev libudev-dev libpci-dev libiberty-dev autoconf

spbnick, 2020-04-30:
Would be great if you raised this issue with Ubuntu maintainers.

binarydepth, 2020-04-30:
The manual build works fine `sudo make dkms_install`, still have to test the driver though

yntelectual, 2020-05-11:
For me the build from sources fails as well. 
```
make -j8 KERNELRELEASE=5.4.0-29-generic KVERSION=5.4.0-29-generic....(bad exit status: 2)
ERROR (dkms apport): binary package for digimend: 9 not found
Error! Bad return status for module build on kernel: 5.4.0-29-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
make: *** [Makefile:75: dkms_modules_install] Error 10
```

And the relevant part of the make log:
```
/usr/src/linux-headers-5.4.0-29-generic/tools/build/Makefile.build:37: /usr/src/linux-headers-5.4.0-29-generic/tools/build/Build.include: No such file or directory
make[6]: *** No rule to make target '/usr/src/linux-headers-5.4.0-29-generic/tools/build/Build.include'.  Stop.
make[5]: *** [Makefile:43: /usr/src/linux-headers-5.4.0-29-generic/tools/objtool/fixdep-in.o] Error 2
make[4]: *** [/usr/src/linux-headers-5.4.0-29-generic/tools/build/Makefile.include:5: fixdep] Error 2
make[3]: *** [Makefile:67: objtool] Error 2
make[2]: *** [Makefile:1823: tools/objtool] Error 2
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-29-generic'
make[1]: *** [Makefile:22: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/digimend/9/build'
```

TSGut, 2020-05-15:
The default kernels Ubuntu 20.04 and by extension Pop OS 20.04 currently ship with (some version of 5.4) is causing this issue. All usual safety precautions in mind, you can fix this issue (and this worked flawlessly for me), by updating to a post 5.6 version kernel. You can do this easily by following e.g. this guide:
https://linuxconfig.org/how-to-upgrade-kernel-to-latest-version-on-ubuntu-20-04-focal-fossa-linux

On Pop OS / Ubuntu 20.04 I had to tell the bash script in the above link to manually download the .deb files and install them one after another, followed by running update-grub so that you can select which kernel to start from on your grub boot menu. The drivers then install correctly and work just fine (my test device is a Huion New 1060 Plus) if you build them from master (didn't try to use the release version, but that may also work on on 5.6+).

I can't tell you exactly what causes this instability with the fresh install 5.4 kernels but this should at least help with the issue.

seasick, 2020-05-16:
> didn't try to use the release version, but that may also work on on 5.6+

I wasn't able to make it work with the release version, but installing from source worked for me with kernel v5.6.13

sephirothtbm, 2020-05-17:
I have the same issue, trying on kubuntu 20.04, Linux kernel v5.6.13, compiled and installed from sources and it looks like everything is fine, the graphics tablet program recognizes my tablet but didn't works correctly, mapping the buttons didn't works, the ring touch also doesn't works, just the stylus.

spbnick, 2020-05-17:
Thank you for the information, @TSGut!

@sephirothtbm, could you post the output of `cat /sys/module/hid_uclogic/version`?

sephirothtbm, 2020-05-17:
Sure, this is it

`sephirothtbm@sakura-hm:~$ cat /sys/module/hid_uclogic/version
10
sephirothtbm@sakura-hm:~$ `

aviskase, 2020-05-17:
Hi, I've tried installing deb on Ubuntu 20.04, it failed (as in this issue), but now I don't know how to clean up the mess.

Anytime I run apt commands, there is an output:

```
➜ sudo apt-get install -y "linux-headers-$(uname -r)"
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-5.4.0-29-generic is already the newest version (5.4.0-29.33).
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up digimend-dkms (9) ...
Loading new digimend-9 DKMS files...
Building for 5.4.0-29-generic
Building initial module for 5.4.0-29-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/digimend-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.4.0-29-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: error processing package digimend-dkms (--configure):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I've tried downloading source from release and running `sudo make dkms_uninstall`, but it didn't help. What are the proper clean up steps in this situation?

TSGut, 2020-05-17:
Yeah, don't even try to install it with the 5.4 kernels. It fails in a pretty bad way and leaves some stuff behind. You can get apt back to normal by just telling it to remove the incomplete install, though. To do that, simply run 
`sudo apt remove digimend-dkms`

Then see above for how to get it to work on 20.04.

TSGut, 2020-05-17:
@sephirothtbm How are you trying to setup the buttons to determine that they are not working? Most tablets don't come with pre-configured buttons as far as I know and thus need to be set up manually. Have you tried running
`xsetwacom --list devices`
to check what your system is recognizing and then using for example
xsetwacom --set <id> Button <#> "key +ctrl +z -z -ctrl"
to set the button of number <#> on the device with number <id> in the above list to the command for undo, i.e. Ctrl+Z?

sephirothtbm, 2020-05-17:
@TSGut Yeah I realised that, then tried using the GUI of the wacom tools, didn't worked, so used xsetwacom from the command line, as you mention, the buttons are working, but the scroll is still missing, if I run `xsetwacom --list devices` the stylus, the pad and the touch ring are listed, but the touch ring doesn't works.

maciej-jezierski, 2020-05-20:
Same issue. When installing from source

`make[1]: Entering directory '/usr/src/linux-headers-5.4.0-29-generic'
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
make[2]: *** No rule to make target 'arch/x86/tools/relocs_32.c', needed by 'arch/x86/tools/relocs_32.o'.  Stop.
make[1]: *** [arch/x86/Makefile:232: archscripts] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-29-generic'
`

TSGut, 2020-05-20:
Did you try the solution I outlined above? You seem to be running 5.4 kernels but as it stands it won't work for those.

maciej-jezierski, 2020-05-20:
Sorry, I didn't notice I have to restart the computer. I had to install additional dependencies:

`sudo apt install flex bison
`
But still have the same issue once installed

```
make[1]: Entering directory '/usr/src/linux-headers-5.6.14-050614-generic'
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support

make[2]: *** No rule to make target 'arch/x86/tools/relocs_32.c', needed by 'arch/x86/tools/relocs_32.o'.  Stop.

make[1]: *** [arch/x86/Makefile:233: archscripts] Error 2

make[1]: Leaving directory '/usr/src/linux-headers-5.6.14-050614-generic'

make: *** [Makefile:22: modules] Error 2
```
I will try to dig into to get more information.

maciej-jezierski, 2020-05-20:
It has started on installation from the source once I modify Makefile with

```
SUBDIRS=$(PWD) 
vs
M=$(shell pwd)
```

also, my tablet is working as well.

spbnick, 2020-05-24:
@maciej-jezierski, this change is in the latest development version, could you please try that?

maciej-jezierski, 2020-05-25:
@spbnick latest development version is working as expected.

spbnick, 2020-05-25:
@maciej-jezierski, thank you for testing and the confirmation! I hope to release the next version soon. Just waiting for Huion to test it with some of their tablets.

binarydepth, 2020-06-03:
How do I purge all dkms changes until next release? show error at every package I install

EGermond, 2020-06-04:
Hello, I have the same installation problem with 5.4 linux kernel. So like said in the comment : https://github.com/DIGImend/digimend-kernel-drivers/issues/387#issuecomment-629462913

> The default kernels Ubuntu 20.04 and by extension Pop OS 20.04 currently ship with (some version of 5.4) is causing this issue. All usual safety precautions in mind, you can fix this issue (and this [...]

but the script (doing it's job) installed not linux 5.6 but 5.7

Hence my situation (translation: the "digimend-dkms" package don't have a suitable version to be installed.
`E: Le paquet « digimend-dkms » n'a pas de version susceptible d'être installée `

I'll install the 5.6 linux kernel and let you know if it works, but cautionnary tales, guys :wink:.

CoreyVincent, 2020-06-08:
I'm still getting errors after updating the kernel:
```
$sudo make dkms_install
dkms add .

Creating symlink /var/lib/dkms/digimend/9/source ->
                 /usr/src/digimend-9

DKMS: add completed.
dkms build digimend/9

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.7.1-050701-generic KVERSION=5.7.1-050701-generic...(bad exit status: 2)
ERROR (dkms apport): kernel package linux-headers-5.7.1-050701-generic is not supported
Error! Bad return status for module build on kernel: 5.7.1-050701-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
make: *** [Makefile:74: dkms_modules_install] Error 10
```

spbnick, 2020-06-09:
@CoreyVincent, could you please post the `make.log` file your posted output refers to?

CoreyVincent, 2020-06-09:
@spbnick 
```
DKMS make.log for digimend-9 for kernel 5.7.1-050701-generic (x86_64)
Mon 08 Jun 2020 06:34:13 PM EDT
make[1]: Entering directory '/var/lib/dkms/digimend/9/build'
make -C /lib/modules/5.7.1-050701-generic/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[2]: Entering directory '/usr/src/linux-headers-5.7.1-050701-generic'
  LEX     scripts/kconfig/lexer.lex.c
/bin/sh: 1: flex: not found
make[4]: *** [scripts/Makefile.host:9: scripts/kconfig/lexer.lex.c] Error 127
make[4]: *** Waiting for unfinished jobs....
  YACC    scripts/kconfig/parser.tab.[ch]
/bin/sh: 1: bison: not found
make[4]: *** [scripts/Makefile.host:17: scripts/kconfig/parser.tab.h] Error 127
make[3]: *** [Makefile:588: syncconfig] Error 2
make[2]: *** [Makefile:699: include/config/auto.conf.cmd] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.7.1-050701-generic'
make[1]: *** [Makefile:22: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/digimend/9/build'
```

spbnick, 2020-06-09:
@CoreyVincent, I suspect this is a problem with dkms package dependencies in Ubuntu, and needs to be fixed there. I would suggest to raise this issue with Ubuntu support.

EGermond, 2020-06-16:
> Hello, I have the same installation problem with 5.4 linux kernel. So like said in the comment : [#387 (comment)](https://github.com/DIGImend/digimend-kernel-drivers/issues/387#issuecomment-629462913)
> 
> > The default kernels Ubuntu 20.04 and by extension Pop OS 20.04 currently ship with (some version of 5.4) is causing this issue. All usual safety precautions in mind, you can fix this issue (and this worked flawlessly for me), by updating to a post 5.6 version kernel. You can do this easily by following e.g. this guide:
> > https://linuxconfig.org/how-to-upgrade-kernel-to-latest-version-on-ubuntu-20-04-focal-fossa-linux
> 
> but the script (doing it's job) installed not linux 5.6 but 5.7
> 
> Hence my situation (translation: the "digimend-dkms" package don't have a suitable version to be installed.
> `E: Le paquet « digimend-dkms » n'a pas de version susceptible d'être installée `
> 
> I'll install the 5.6 linux kernel and let you know if it works, but cautionnary tales, guys wink.

So to make the digimend install work I've used the mainline GUI interface which allowed me to chose the kernel. (Method # 3: Update Ubuntu Kernel Version via GUI on https://linuxhint.com/update_ubuntu_kernel_20_04/)
I've tried several from the most recent (installed by the script) until I find one who worked with the DKMS install. As of now the 5.6.10. Why the 5.6.10 and not 5.6.13, your guess is as good as mine.

It's working fine now. Thanks to the digimend team and the help of the contributors. Regards

sfornengo, 2020-06-19:
Should the digimend v10 release solve the 20.04 ubuntu installation problem ?

markgaudreau, 2020-06-30:
I would also like to know if the 20.04 ubuntu installation problem is expected to be resolved by the v10 release.

clay53, 2020-07-06:
> @CoreyVincent, I suspect this is a problem with dkms package dependencies in Ubuntu, and needs to be fixed there. I would suggest to raise this issue with Ubuntu support.

Has anyone done this yet?

markgaudreau, 2020-07-07:
@clay53, from what I understand, spbnick was making that commentary to someone who was trying to update to a 5.6 kernel.  What I am personally waiting for is a fix for the 5.4 kernel.  I'd rather not upgrade to 5.6 and I wonder if the v10 release will fix the installation problem on "out of the box Ubuntu 20.04".

Am I missing something?  Could someone please explain if v10 is expected to fix this issue for a clean Ubuntu 20.04 installation?  If not, is there an issue to open with Ubuntu?

stickgrinder, 2020-07-10:
> It's working on installation from the source once I modify Makefile with
> 
> ```
> SUBDIRS=$(PWD) 
> ```
> 
> to
> 
> ```
> M=$(shell pwd)
> ```
> 
> also, my tablet is working as well now.
> 
> My setups Ubuntu 20. Tablet Huion 1060 Plus.

@maciej-jezierski sorry mate but it's unclear to me if this change makes the driver work on a "vanilla" 5.4 kernel.

Can you please clarify your steps?
Thanks a lot!

stickgrinder, 2020-07-16:
Leaving a comment for noobs like me in trouble with the v9 deb on Ubuntu 20.04.  
Installing dkms from source works out of the box on vanilla Ubuntu kernel 5.4.0-40-generic.

My steps:

- `sudo apt install dkms`
- cloned digimend repo
- issued `sudo make dkms_install`

worked out of the box. Deb is still broken it seams, even if I can't get why.
Thanks everyone for the info.

spbnick, 2020-07-17:
@markgaudreau, this is an Ubuntu kernel/dkms packaging issue - it simply doesn't specify enough dependencies. A workaround is to install those dependencies manually. The most DIGImend can do is perhaps add a note to the README.md explaining the problem and providing a command to install them. If anyone can reproduce this and figure out the exact command to install the dependencies, I would be glad to accept such a change.

stickgrinder, 2020-07-17:
> @markgaudreau, this is an Ubuntu kernel/dkms packaging issue - it simply doesn't specify enough dependencies. A workaround is to install those dependencies manually. The most DIGImend can do is [...]

Clear now, thanks... I'm about to test Pop!_OS alongside Ubuntu on a new hard drive: I'll do my research there and if I find the correct configuration I'll contribute a note in the readme.

Thanks a lot!

will-code-for-pizza, 2020-08-30:
For each new kernel-version installed by an upgrade, simply do:

```
$ sudo apt-get install git libncurses-dev flex bison openssl libssl-dev dkms libelf-dev libudev-dev libpci-dev libiberty-dev autoconf build-essential checkinstall

$ wget https://github.com/DIGImend/digimend-kernel-drivers/archive/v9.tar.gz

$ tar -xzf v9.tar.gz

$ cd digimend-kernel-drivers-9/

$ cp Makefile Makefile.org

$ sed -i 's|SUBDIRS=$(PWD)|M=$(shell pwd)|' Makefile

$ make

$ echo "digimend-kernel-drivers" > description-pak

$ sudo checkinstall -y -D --install=yes --pkgname=digimend-kernel-drivers --provides=digimend-kernel-drivers --requires=libncurses-dev,libssl-dev,libelf-dev,libudev-dev,libpci-dev,libiberty-dev --pkgversion=$(date +%Y%m%d%H%M) --pkggroup=libs --pakdir=~/tmp/ --reset-uids=yes --exclude=/home,/tmp
```

You'll get a .dpkg, which could be uninstalled by 'sudo apt-get purge digimend-kernel-drivers'

Regards

will-code-for-pizza, 2020-08-31:
The DKMS method:

`$ sudo apt-get install -y "linux-headers-$(uname -r)" build-essential dkms git

$ git clone https://github.com/DIGImend/digimend-kernel-drivers.git

$ cd digimend-kernel-drivers

$ sudo make dkms_install

$ declare -a myMODULES; myMODULES=( 'hid-kye' 'hid-uclogic' 'hid-polostar' 'hid-viewsonic' )
$ for mymodule in ${myMODULES[@]}; do sudo modprobe -r ${mymodule}; done
$ for mymodule in ${myMODULES[@]}; do sudo modprobe ${mymodule}; done

$ sudo lsmod | grep -iE "hid-kye|hid-uclogic|hid-polostar|hid-viewsonic"`

;-)

wandetri, 2020-09-01:
> The default kernels Ubuntu 20.04 and by extension Pop OS 20.04 currently ship with (some version of 5.4) is causing this issue. All usual safety precautions in mind, you can fix this issue (and this worked flawlessly for me), by updating to a post 5.6 version kernel. You can do this easily by following e.g. this guide:
> https://linuxconfig.org/how-to-upgrade-kernel-to-latest-version-on-ubuntu-20-04-focal-fossa-linux
> 
> On Pop OS / Ubuntu 20.04 I had to tell the bash script in the above link to manually download the .deb files and install them one after another, followed by running update-grub so that you can select which kernel to start from on your grub boot menu. The drivers then install correctly and work just fine (my test device is a Huion New 1060 Plus) if you build them from master (didn't try to use the release version, but that may also work on 5.6+).
> 
> I can't tell you exactly what causes this problem with the fresh install 5.4 kernels but this should at least help with the issue.

i've upgraded to 5.8.5-05 , but the problem still exists

wandetri, 2020-09-01:
> The DKMS method:
> 
> ```
> #!/bin/bash
> 
> sudo apt-get install -y "linux-headers-$(uname -r)" build-essential dkms git
> 
> # THIS IS FOR BLEEDING EDGE
> #git clone https://github.com/DIGImend/digimend-kernel-drivers.git
> #cd digimend-kernel-drivers
> 
> # THIS IS FOR STABLE v9
> wget https://github.com/DIGImend/digimend-kernel-drivers/archive/v9.tar.gz
> tar -xzf v9.tar.gz
> cd digimend-kernel-drivers-9
> 
> sudo make dkms_uninstall
> sudo make dkms_install
> 
> declare -a myMODULES; myMODULES=( 'hid-kye' 'hid-uclogic' 'hid-polostar' 'hid-viewsonic' )
> for mymodule in ${myMODULES[@]}; do echo "unloading ${mymodule}"; sudo modprobe -r ${mymodule}; done
> for mymodule in ${myMODULES[@]}; do echo "reloading ${mymodule}"; sudo modprobe ${mymodule}; done
> 
> sudo lsmod | grep -iE "kye|uclogic|polostar|viewsonic"
> ```
> 
> ;-)

woow it works on 5.8.5-05 !
thanks!

spbnick, 2020-10-05:
Should be working with the v10 release.

closed 2020-10-05 by spbnick