DIGImend/digimend-kernel-drivers #388: XP-PEN Star G430S_B doesn't work with DIGImend

Sturwandan, 2020-05-04:
This tablet is similar in appearance to the white version has same size and buttons, however, USB ID is different.
Black tablet ID is 28bd:0913
White tablet ID is 28bd:0075

When I insert the tablet, following messages appear:
```
usb 6-3: new full-speed USB device number 37 using ohci-pci
usb 6-3: New USB device found, idVendor=28bd, idProduct=0913, bcdDevice= 0.00
usb 6-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 6-3: Product: 4 inch PenTablet
usb 6-3: Manufacturer: UGTABLET
usb 6-3: SerialNumber: 000000
input: UGTABLET 4 inch PenTablet Mouse as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.0/0003:28BD:0913.0085/input/input151
input: UGTABLET 4 inch PenTablet Keyboard as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.0/0003:28BD:0913.0085/input/input152
hid-generic 0003:28BD:0913.0085: input,hidraw0: USB HID v1.00 Mouse [UGTABLET 4 inch PenTablet] on usb-0000:00:16.0-3/input0
input: UGTABLET 4 inch PenTablet as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.1/0003:28BD:0913.0086/input/input153
hid-generic 0003:28BD:0913.0086: input,hidraw1: USB HID v1.00 Device [UGTABLET 4 inch PenTablet] on usb-0000:00:16.0-3/input1
hid-generic 0003:28BD:0913.0087: hiddev96,hidraw2: USB HID v1.00 Device [UGTABLET 4 inch PenTablet] on usb-0000:00:16.0-3/input2
```

The tablet works with generic drivers, pressure sensitivity is available in GIMP, however, buttons on the pen do not work as expected: the button next to the tip works as MMB (middle mouse button), while the button further from the tip doesn't work at all. Expected behavior is: close button: RMB, far button: MMB

The far button works in principle, since this command ``usbhid-dump -m 28bd:0913 -e stream`` shows lines like ``006:036:001:STREAM             1588586846.485230
 07 A0 E0 3B 0C 22 00 00 00 00`` when no buttons are pressed, but 
A0 changes to A1 with tip touch, to A2 with close-to-tip button and A4 with far button. If I both touch the tablet and hold far button, I get A5, as expected (1+4), so we can conclude that the second byte encodes buttons, similar to G430S white, which works with DIGImend.

I have tried to change this line to 
#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_G540   0x0913
But it wasn't enough. Likely it happens because of different HID packet length, I have seen a bug report with similar problem on Windows drivers: https://github.com/hawku/TabletDriver/issues/520

Playing with the monitor shows that, I get this numbers when I bring the pen to a tablet corner slowly and then pull away, without touching:
07 A0 00 00 00 00 00 00 00 00 || top left corner
07 A0 00 00 FF 7F 00 00 00 00 || bottom left corner
07 A0 FF 7F 00 00 00 00 00 00 || top right corner
07 A0 FF 7F FF 7F 00 00 00 00 || bottom right corner

A0 changes to C0 when I remove the pen from the tablet.

I have noticed that white tablet output is like this:

013:002:001:STREAM             1588592013.478316
 07 A0 58 4F B1 39 00 00

So the HID packet length is indeed different.

Sturwandan, 2020-05-04:
So, I might be wrong, but it seems that first byte is 07, second byte encodes Button state, then there is horizontal coordinate (2 bytes), vertical coordinate (2 bytes) and pressure (2 bytes), then zero (2 bytes).

Sturwandan, 2020-05-06:
I have tried to fix it myself with this patch, but did achieve nothing:

```
diff --git a/hid-ids.h b/hid-ids.h
--- a/hid-ids.h
+++ b/hid-ids.h
@@ -52,6 +52,8 @@
 #define USB_VENDOR_ID_UGEE			0x28bd
 #define USB_DEVICE_ID_UGEE_XPPEN_TABLET_G540	0x0075
 #define USB_DEVICE_ID_UGEE_XPPEN_TABLET_G640	0x0094
+// XP-PEN G430S_B stands for black version
+#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_G430S_B	0x0913
 #define USB_DEVICE_ID_UGEE_XPPEN_TABLET_DECO01	0x0042
 #define USB_DEVICE_ID_UGEE_TABLET_G5		0x0074
 #define USB_DEVICE_ID_UGEE_TABLET_EX07S		0x0071
diff --git a/hid-uclogic-core.c b/hid-uclogic-core.c
--- a/hid-uclogic-core.c
+++ b/hid-uclogic-core.c
@@ -529,6 +529,8 @@
 	{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
 				USB_DEVICE_ID_UGEE_XPPEN_TABLET_G540) },
 	{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
+				USB_DEVICE_ID_UGEE_XPPEN_TABLET_G430S_B) },
+	{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
 				USB_DEVICE_ID_UGEE_XPPEN_TABLET_G640) },
 	{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
 				USB_DEVICE_ID_UGEE_XPPEN_TABLET_DECO01) },
diff --git a/hid-uclogic-params.c b/hid-uclogic-params.c
--- a/hid-uclogic-params.c
+++ b/hid-uclogic-params.c
@@ -1178,6 +1178,9 @@
 		     USB_DEVICE_ID_UCLOGIC_UGEE_TABLET_45):
 	case VID_PID(USB_VENDOR_ID_UCLOGIC,
 		     USB_DEVICE_ID_UCLOGIC_UGEE_TABLET_47):
+	// an attempt to hack G430S_B here
+	case VID_PID(USB_VENDOR_ID_UGEE,
+		     USB_DEVICE_ID_UGEE_XPPEN_TABLET_G430S_B):
 		rc = uclogic_params_huion_init(&p, hdev);
 		if (rc != 0)
 			goto cleanup;
```

DIGImend detects the tablet, but the code makes it change USB ID to 28bd:1227 and the tablet doesn't work anymore until it is reconnected.

When I add the new ID near G430S white (which shares ID with G540), I have same effect as I had when I tried to just replace USB ID.
The console is flooded with
usb 6-3: input irq status -75 received
When I bring the pen close to the tablet, and the tablet doesn't work, here is dmesg:
```
usb 6-3: new full-speed USB device number 53 using ohci-pci
usb 6-3: New USB device found, idVendor=28bd, idProduct=0913, bcdDevice= 0.00
usb 6-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 6-3: Product: 4 inch PenTablet
usb 6-3: Manufacturer: UGTABLET
usb 6-3: SerialNumber: 000000
input: UGTABLET 4 inch PenTablet Mouse as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.0/0003:28BD:0913.00DC/input/input238
input: UGTABLET 4 inch PenTablet Keyboard as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.0/0003:28BD:0913.00DC/input/input239
hid-generic 0003:28BD:0913.00DC: input,hidraw0: USB HID v1.00 Mouse [UGTABLET 4 inch PenTablet] on usb-0000:00:16.0-3/input0
input: UGTABLET 4 inch PenTablet as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.1/0003:28BD:0913.00DD/input/input240
hid-generic 0003:28BD:0913.00DD: input,hidraw1: USB HID v1.00 Device [UGTABLET 4 inch PenTablet] on usb-0000:00:16.0-3/input1
hid-generic 0003:28BD:0913.00DE: hiddev96,hidraw2: USB HID v1.00 Device [UGTABLET 4 inch PenTablet] on usb-0000:00:16.0-3/input2
usb 6-3: USB disconnect, device number 53
uclogic 0003:28BD:0913.00DC: failed retrieving string descriptor #200: -62
uclogic 0003:28BD:0913.00DC: failed retrieving pen parameters: -62
uclogic 0003:28BD:0913.00DC: failed probing pen v2 parameters: -62
uclogic 0003:28BD:0913.00DC: failed probing parameters: -62
uclogic: probe of 0003:28BD:0913.00DC failed with error -62
uclogic 0003:28BD:0913.00DD: No inputs registered, leaving
uclogic 0003:28BD:0913.00DD: hidraw0: USB HID v1.00 Device [UGTABLET 4 inch PenTablet] on usb-0000:00:16.0-3/input1
uclogic 0003:28BD:0913.00DE: interface is invalid, ignoring
usb 6-3: new full-speed USB device number 54 using ohci-pci
usb 6-3: New USB device found, idVendor=28bd, idProduct=1227, bcdDevice= 0.00
usb 6-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 6-3: Product: TABLET
usb 6-3: Manufacturer: UGTABLET
hid-generic 0003:28BD:1227.00DF: hiddev96,hidraw0: USB HID v1.11 Device [UGTABLET TABLET] on usb-0000:00:16.0-3/input0
input: UGTABLET TABLET Mouse as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.1/0003:28BD:1227.00E0/input/input242
input: UGTABLET TABLET Keyboard as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.1/0003:28BD:1227.00E0/input/input243
hid-generic 0003:28BD:1227.00E0: input,hidraw1: USB HID v1.00 Mouse [UGTABLET TABLET] on usb-0000:00:16.0-3/input1
input: UGTABLET TABLET as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.2/0003:28BD:1227.00E1/input/input244
input: UGTABLET TABLET as /devices/pci0000:00/0000:00:16.0/usb6/6-3/6-3:1.2/0003:28BD:1227.00E1/input/input245
hid-generic 0003:28BD:1227.00E1: input,hiddev97,hidraw2: USB HID v1.00 Device [UGTABLET TABLET] on usb-0000:00:16.0-3/input2
hid-generic 0003:28BD:1227.00E2: hiddev98,hidraw3: USB HID v1.00 Device [UGTABLET TABLET] on usb-0000:00:16.0-3/input3
```

I guess, either I am doing something wrong or adding support for this tablet is not that easy.
I have tried to put the code near white tablet, then I didn't have the device to change ID, but it didn't work either, emitting ``interface is invalid, ignoring``  and flooding the logs with ``usb 6-3: input irq status -75 received`` when pen is in range.

Sturwandan, 2020-05-08:
I have fixed the tablet behavior with this patch against the kernel source:
```
--- a/drivers/hid/hid-input.c
+++ b/drivers/hid/hid-input.c
@@ -828,8 +828,9 @@
                         * Actual eraser (BTN_TOOL_RUBBER) is set by Invert usage when
                         * tool gets in proximity.
                         */
-                       map_key_clear(BTN_TOUCH);
-                       break;
+                       // A dirty hack to make G430S_B work as it should
+                       //map_key_clear(BTN_TOUCH);
+                       //break;
 
                case 0x46: /* TabletPick */
                case 0x5a: /* SecondaryBarrelSwitch */
```

This patch does, of course, break support for all tablets which do have an actual eraser switch on the back, but it works as temporary fix for people who only have one tablet.

Sturwandan, 2020-05-10:
Next step is to try and switch the tablet to its native mode with coordinates going up to (10160, 7620) instead of (32767, 32767) and HID reports being 12 bytes long instead of 10. 
In order to do this, I had to read some articles regarding USB protocol and monitoring.
First of all, I have tried to record the text dump of exchange, this is done simply by this command:
> cat /sys/kernel/debug/usb/usbmon/4u | tee [usbmon_4u.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4605114/usbmon_4u.txt)

This method works, but it's kinda hard to analyze. If you want anyway, [read this](https://www.kernel.org/doc/Documentation/usb/usbmon.txt).

There is a program, [vusb-analyzer](http://vusb-analyzer.sourceforge.net/), which is supposed to help decoding the text dumps, however I didn't get it to work on a modern system. There is [a newer version](https://github.com/scanlime/vusb-analyzer) moved from sourceforge to github, however it's also old and I didn't try it.

So, I have had to use ``tcpdump -i usbmon4 -w usbmon4.pcap`` to record USB events in a different format, which can be fed into wireshark.
Later on, comparing both text and binary output I was able to pinpoint two exchanges, which are of interest. The first one:
```
ffff99e840165480 3047498234 S Ci:3:007:0 s 80 06 0364 0409 0200 512 <
ffff99e840165480 3047499053 C Ci:3:007:0 0 14 = 0e03b027 c41d0600 ff1fec09 0000
```
This is request and answer to get string descriptor 0x64, which is similar to what is done by uclogic-probe above:
```
S 64 0E 03 B0 27 C4 1D 06 00 FF 1F EC 09 00 00
 64 |        Params block #1: ??????
    |                          Max X: 10160
    |                          Max Y: 7620
    |                   Max pressure: 8191
    |                     Resolution: 2540
```
The next echange is this one:
```
ffff99e840165b40 3047581421 S Io:3:007:3 -115:8 3 = 02b004
ffff99e840165b40 3047589051 C Io:3:007:3 0:8 3 >
ffff99e834042000 3047591046 C Ii:3:007:3 0:2 10 = 02b10400 00000000 0000
```
The first line means sending three bytes {0x02, 0xB0, 0x04} as interrupt transfer to endpoint 0x03

I have tried to change uclogic-probe.c to make it send this interrupt transfer after it's done with probing. This task proved quite challenging because even after I had the code in place, it would not work, returning -1 which means I/O error. After a while I have looked in dmesg and noticed this message: ``usb 4-5: usbfs: process 1080 (uclogic-probe) did not claim interface 1 before use``
This have explained why I wasn't able to send the packet and now I made it work:
```
--- uclogic-probe.c.orig
+++ uclogic-probe.c
@@ -127,8 +127,9 @@
     libusb_device_handle   *handle      = NULL;
     unsigned char           buf[257];
     int                     len;
-    uint8_t                 idx_list[]  = {0x64, 0x65, 0x6E, 0x79, 0x7A,
-                                           0x7B, 0xC8, 0xC9, 0xCA};
+//    uint8_t                 idx_list[]  = {0x64, 0x65, 0x6E, 0x79, 0x7A,
+//                                           0x7B, 0xC8, 0xC9, 0xCA};
+    uint8_t                 idx_list[] = {0x64};
     uint8_t                 idx;
 
     struct libusb_device_descriptor     dev_desc;
@@ -197,6 +198,25 @@
         /* Print the descriptor */
         print_chunk('S', buf, len + 1);
     }
+
+/* Trying to send some bytes to USB device
+ *int libusb_interrupt_transfer(struct libusb_device_handle *devh,
+ *                                            unsigned char endpoint, 
+ *                                            unsigned char *data, 
+ *                                                      int length, 
+ *                                                      int *transferred, 
+ *                                             unsigned int timeout)
+ */
+    int bytes_transferred = 0;
+    unsigned char interrupt_data[3] = {0x02, 0xB0, 0x04};
+    int driver_active = libusb_kernel_driver_active(handle, 1);
+    int claim_status = libusb_claim_interface(handle, 1);
+    int interrupt_status = libusb_interrupt_transfer(handle, 0x03, interrupt_data, sizeof(interrupt_data), &bytes_transferred, 500);
+    int release_status = libusb_release_interface(handle, 1);
+    printf("INTERRUPT: DRV:%d CLAIMED:%d INT:%d RELEASED:%d Bytes:%d | %p, %08X, %d\n", driver_active, claim_status, interrupt_status, release_status,
+          bytes_transferred, handle, *(uint32_t*)interrupt_data, sizeof(interrupt_data));
[... 13 lines cut ...]
S 64 0E 03 B0 27 C4 1D 06 00 FF 1F EC 09 00 00
INTERRUPT: DRV:0 CLAIMED:0 INT:0 RELEASED:0 Bytes:3 | 0x167d9d0, 0304B002, 3
# modprobe usbhid hid_generic
```
After it's done, I start to get 12 bytes HID reports:
```
# usbhid-dump -es | head -30
Starting dumping interrupt transfer stream
with 1 minute timeout.

004:021:002:STREAM             1589085235.477630
 02 B1 04 00 00 00 00 00 00 00

004:021:002:STREAM             1589085239.289547
 02 A0 26 15 22 16 00 00 00 00 00 00

004:021:002:STREAM             1589085239.293632
 02 A0 27 15 28 16 00 00 00 00 00 00
```

The first report corresponds to USB dump from exchange with proprietary driver, other reports have similar formats to 10-bytes reports above, but the first byte is 02 instead of 07 and coordinates are going from (0, 0) to native tablet resolution. The last four bytes remain 00.

However, tablet does not work anymore after that, and HID reports from it are only recognized by usbhid-dump and usbmon. ``/sys/kernel/debug/hid/*/events`` do not register any events anymore.

The same thing can be seen if you start and then stop proprietary driver. The tablet is sending events, but they are not processed. The likely reason is that the HID descriptor is still the same, so the hid driver cannot decode the tablet events anymore after their format is changed.

Sturwandan, 2020-05-12:
So, the next step is to make the kernel driver.  There are two possibilities: easier and harder.
The easy way is to model the new driver after existing hid-redragon.c in the kernel source. 
It does only change only one byte in HID report descriptor of that keyboard and it's very short.

The harder way involves sending bytes 0x02, 0xB0, 0x04 via an interrupt transfer, similar to what modified uclogic-probe.c does, as described above. The driver will also need a freshly written HID Report descriptor which allow the kernel to understand 12-bytes long reports, which in this mode, are sent from interface 2, rather than interface 1 (starting from 0).

The result of easy way is very short and easy to understand:

```C
/*
 *  HID driver for XP-PEN G430S_B, simple version
 *  Based on hid-redragon.c Copyright (c) 2017 Robert Munteanu
 *  Copyright (c) 2020 Sturwandan
 *  SPDX-License-Identifier: GPL-2.0+
 */

/*
 * This program is free software; you can redistribute it and/or modify it
 * under the terms of the GNU General Public License as published by the Free
 * Software Foundation; either version 2 of the License, or (at your option)
 * any later version.
 */

#include <linux/device.h>
#include <linux/hid.h>
#include <linux/module.h>

#include "hid-ids.h"

static __u8 *g430sb_report_fixup(struct hid_device *hdev, __u8 *rdesc, unsigned int *rsize) {
	if (*rsize == 109 && rdesc[16] == 0x09 && rdesc[17] == 0x45) {
		dev_info(&hdev->dev, "Fixing XP-PEN Star G430S_B report descriptor.\n");
		rdesc[17] = 0x46;
	}
	return rdesc;
}

static const struct hid_device_id g430sb_devices[] = {
	{HID_USB_DEVICE(USB_VENDOR_ID_UGEE, USB_DEVICE_ID_UGEE_XPPEN_TABLET_G430SB)},
	{}
};

MODULE_DEVICE_TABLE(hid, g430sb_devices);

static struct hid_driver g430sb_driver = {
	.name = "g430sb",
	.id_table = g430sb_devices,
	.report_fixup = g430sb_report_fixup
};

module_hid_driver(g430sb_driver);

MODULE_LICENSE("GPL");

```

However in order for this driver to work, this patch is necessary so it's build together with other DIGImend drivers:

```patch
diff -ru digimend-kernel-drivers-master/Makefile digimend-kernel-drivers-modified/Makefile
--- digimend-kernel-drivers-master/Makefile
+++ digimend-kernel-drivers-modified/Makefile
@@ -1,4 +1,4 @@
-obj-m := hid-kye.o hid-uclogic.o hid-polostar.o hid-viewsonic.o
+obj-m := hid-kye.o hid-uclogic.o hid-polostar.o hid-viewsonic.o hid-g430sb.o
 hid-uclogic-objs := \
 	hid-uclogic-core.o \
 	hid-uclogic-rdesc.o \
@@ -44,7 +44,8 @@
 	rm -vf /lib/modules/*/extra/hid-kye.ko \
 	       /lib/modules/*/extra/hid-polostar.ko \
 	       /lib/modules/*/extra/hid-uclogic.ko \
-	       /lib/modules/*/extra/hid-viewsonic.ko
+	       /lib/modules/*/extra/hid-viewsonic.ko \
+	       /lib/modules/*/extra/hid-g430sb.ko
 
 install: modules_install udev_rules_install depmod_conf_install xorg_conf_install
 	udevadm control --reload
diff -ru digimend-kernel-drivers-master/dkms.conf digimend-kernel-drivers-modified/dkms.conf
--- digimend-kernel-drivers-master/dkms.conf
+++ digimend-kernel-drivers-modified/dkms.conf
@@ -6,7 +6,9 @@
 BUILT_MODULE_NAME[1]="hid-uclogic"
 BUILT_MODULE_NAME[2]="hid-polostar"
 BUILT_MODULE_NAME[3]="hid-viewsonic"
+BUILT_MODULE_NAME[4]="hid-g430sb"
 DEST_MODULE_LOCATION[0]="/extra"
 DEST_MODULE_LOCATION[1]="/extra"
 DEST_MODULE_LOCATION[2]="/extra"
 DEST_MODULE_LOCATION[3]="/extra"
+DEST_MODULE_LOCATION[4]="/extra"
diff -ru digimend-kernel-drivers-master/hid-ids.h digimend-kernel-drivers-modified/hid-ids.h
--- digimend-kernel-drivers-master/hid-ids.h
+++ digimend-kernel-drivers-modified/hid-ids.h
@@ -52,6 +52,8 @@
 #define USB_VENDOR_ID_UGEE			0x28bd
 #define USB_DEVICE_ID_UGEE_XPPEN_TABLET_G540	0x0075
 #define USB_DEVICE_ID_UGEE_XPPEN_TABLET_G640	0x0094
+// XP-PEN G430S_B stands for black version
+#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_G430SB	0x0913
 #define USB_DEVICE_ID_UGEE_XPPEN_TABLET_DECO01	0x0042
 #define USB_DEVICE_ID_UGEE_TABLET_G5		0x0074
 #define USB_DEVICE_ID_UGEE_TABLET_EX07S		0x0071
```

I have tested this driver with my tablet and it seems to work fine.

Sturwandan, 2020-05-23:
I have also made a native mode driver, it works, however there is an unsolved problem still. 
[The source code is here](https://github.com/DIGImend/digimend-kernel-drivers/blob/58b321338fdbb43fbc3f97c64a2890ff8ed3ad2b/hid-g430sb.c)
Along with this file you also need to apply the patch from comment above, to make the build system recognize it.

This driver switches the tablet to native mode first, sending bytes 0x02, 0xB0, 0x04 via an interrupt transfer to interface 0x03. This is different from all other tablets supported by DIGImend, so it's easier to write this driver in a separate file.

In order to make this driver, I have taken hid-viewsonic.c driver and replaced the text strings with appropriate. It turns out that all the drivers have similar layout and I could have started from hid-redragon.c too, but I didn't discover it until much later.

In order to send interrupt transfer, I needed to know the name of kernel function (which is not same as name of libusb function above) and how to use it.
It was easy to figure out the name: First of all by reading uclogic driver, I have found that  usb_control_msg was used to get string descriptors. Then grepping it in linux kernel source, I have found the include file, which it is from: /usr/src/linux/include/linux/usb.h
Then, figuring out the other USB functions are there too, I have found this definition:
```
extern int usb_interrupt_msg(struct usb_device *usb_dev, unsigned int pipe,
        void *data, int len, int *actual_length, int timeout);
```
The same include file usb.h has explanation, how to make a pipe knowing the endpoint: ``usb_sndintpipe(udev, endpoint)``, and other arguments are similar to what is found in libusb. 

However there is a pitfall: for some reason this function can't just send data from regular variable, it needs that data to be located in a specific kind of memory, while local variables in functions are in kernel stack and not suitable. So I had to learn how to use kmalloc and kcalloc and copy those three bytes into a newly created buffer. Otherwise I was getting error  -11 (EAGAIN) and wasn't able to figure out where it's coming from, since in wireshark it seemed that the USB packet was sent correctly and then the device responded with this error.

The next problem was that there were no hid-reports from the device in evtest or /sys/kernel/debug/hid/*/events but usbhid-dump tool was able to demonstrate 12 byte reports, so the kernel was sending the correct packets.

I have thought that was because of incorrect HID Report descriptor and tried to debug it.
It turns out that the best tool to check whenever it's correct or not is to load it into the kernel and check corresponding rdesc file. For example with the final version of the report I am getting this:
```
==> /sys/kernel/debug/hid/0003:28BD:0913.0003/rdesc <==
05 0d 09 02 a1 01 85 02 09 20 a1 00 09 42 09 44 09 46 15 00 25 01 75 01 95 03 81 02 95 02 81 03 09 32 95 01 81 02 95 02 81 03 a4 05 01 09 30 15 00 26 b0 27 65 11 55 0d 35 00 46 b0 27 75 10 95 01 81 22 09 31 26 c4 1d 46 c4 1d 81 22 b4 09 30 15 00 26 ff 1f 75 10 95 01 81 02 95 02 81 03 c0 c0 

  INPUT(2)[INPUT]
    Field(0)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(3)
        Digitizers.TipSwitch
        Digitizers.BarrelSwitch
        Digitizers.TabletPick
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(3)
      Report Offset(0)
      Flags( Variable Absolute )
    Field(1)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        Digitizers.InRange
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(1)
      Report Offset(5)
      Flags( Variable Absolute )
    Field(2)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        GenericDesktop.X
      Logical Minimum(0)
      Logical Maximum(10160)
      Physical Minimum(0)
      Physical Maximum(10160)
      Unit Exponent(-3)
      Unit(SI Linear : Centimeter)
      Report Size(16)
      Report Count(1)
      Report Offset(8)
      Flags( Variable Absolute NoPreferredState )
    Field(3)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
[... 16 lines cut ...]
        Digitizers.TipPressure
      Logical Minimum(0)
      Logical Maximum(8191)
      Report Size(16)
      Report Count(1)
      Report Offset(40)
      Flags( Variable Absolute )

Digitizers.TipSwitch ---> Key.Touch
Digitizers.BarrelSwitch ---> Key.Stylus
Digitizers.TabletPick ---> Key.Stylus2
Digitizers.InRange ---> Key.ToolPen
GenericDesktop.X ---> Absolute.X
GenericDesktop.Y ---> Absolute.Y
Digitizers.TipPressure ---> Absolute.Pressure
```

As you can see, unlike [this tool](http://eleccelerator.com/usbdescreqparser/) or [hidrd](https://github.com/DIGImend/hidrd), kernel provides full info about every field you are trying to interpret, including its bit-offset in hid reports. 

While trying to catch this problem, I have added check for interface number in both fixup and probe functions, so that will be only run once and added some more minor improvements.
It didn't help at first, because I haven't noticed immediately, that the HID reports are emitted from the third interface in native mode, while they were emitted from the second one in default mode. 

So after all those problems were fixed, I got a working driver for my tablet. However, one problem is still unsolved: 
The driver works only if it is loaded after the tablet plugged in, and doesn't work, if you replug the table, however reinserting the driver with modprobe makes it work again.
I would be grateful if anybody provided me any insights why it could be happened.

alfazaz, 2020-05-23:
Hello @Sturwandan, 

Thank you for all these informations.

I have no idea about your problem unsolved but I see that in https://github.com/DIGImend/digimend-kernel-drivers/pull/291 , there is also the same init of tablet in the function `uclogic_params_init_ugee_xppen_a156p` and may be there is somewhere an answer to your problem ?

As for me, I'm still blocked for my XP PEN Artist 12 Pro (https://github.com/DIGImend/digimend-kernel-drivers/issues/351). usbhid-dump tool is able to demonstrate 12 byte reports, so the kernel sends the correct packets but... still no success with evtest. May be a small detail but... :-(

Sturwandan, 2020-05-23:
@alfazaz, actually, usbhid-dump bypasses kernel drivers and talks with hardware directly. So if you see something in it, it only means that the tablet sends reports correctly, but it doesn't mean that it is interpreted correctly. I suggest to try doing the same thing I did: find out encoding of your tablet reports and make sure that its description in rdesc is correct, especially bit offsets and usages. And make sure that it's coming from interface you expect.

alfazaz, 2020-05-23:
So, if I understand things well, I need to write a good rdesc. I'm working on it (I'm not an expert of these things so it's hard for me... If you have concrete tricks, you can write them on https://github.com/DIGImend/digimend-kernel-drivers/issues/351). Thanks.