DIGImend/digimend-kernel-drivers #389: Huion 1060 Plus - problems patching ?

Kan-Rup, 2020-05-06:
Hello

I have a Huion 1060 Plus tablet that I would like to use on my Linux machine, and I have installed the Digimend drivers, I have tried dkms install, the deb package and installing by source. However when the tablet is connected it does not detect the pen. 

Looking into the problem I came to two bugzilla posts ( https://bugzilla.kernel.org/show_bug.cgi?id=207065#c7 ) which indicate that the Huion 1060 Plus tablet MAY work after applying the two patch files. But it seems the patches are applied to a file (if I have identified this correctly I have little experience with patching) called 'xhci-ring.c'. 

Googling this I found the file is present in Linux kernel source tree ( https://github.com/torvalds/linux ). I could not find the file in my own machine (is it for installing purposes only?). So if I understand correctly I have to build a new kernel from these files and update? I had to assume that this upto date kernel version would get tablet working. But then I am NOT sure it is the actual solution.

Current kernel version is : 5.3.0-51-generic

I started to build the new kernel but as always with that process met string of obstacles. I am not sure if even successful with the build the thing might affect other parts of system adversely. I've stopped at the point where I installed the whole Qt framework, then having to set up an environment variable, then having to install moc, after that ???. I've stopped as I have to find clues and problem-solve and it's taking forever. Am waiting some change in updates for now.

Sturwandan, 2020-05-07:
If that's indeed the case you can download the latest kernel source, patch it, compile, install and then boot with new kernel to see if it works. Though if the patch was against a kernel module specific to the device then it would be enough to rebuild only the module.

bsamorodov, 2020-06-03:
Untill the kernel is fixed one may be interested at the [workaround](https://github.com/DIGImend/digimend-kernel-drivers/issues/405#issuecomment-637844659).

HTH, bsam