DIGImend/digimend-kernel-drivers #390: i cant install the drivers in pop os (ubuntu) 20.04

yosoyfreeman, 2020-05-10:
https://imgur.com/50bBX08  anyone knows how to installl the drivers in the new lts? thank you!

SylvainAir, 2020-05-10:
Hello 

I have the same issue, and we are not alone : see https://github.com/DIGImend/digimend-kernel-drivers/issues/387

yosoyfreeman, 2020-05-10:
> Hello
> 
> I have the same issue, and we are not alone : see #387

Hello mate! im a bit new. Can you please explain me a little more what its happening? it looks like i should compile the kernel, right? but that looks pretty complicated... You think will be a easier solution?

Sturwandan, 2020-05-11:
Compiling kernel is fairly easy, but it takes some time if the CPU is weak. However in case of digimend kernel drivers you don't need to compile entire kernel, you only need the source of the kernel running on your machine to be referenced by /lib/modules/kernel.version/build where kernel.version is the version of kernel you run, such as 4.4.44.

yosoyfreeman, 2020-05-11:
Okay mates thank you all. I guess my only option its go back to 19.01 or windows.  So sad. Thank you anyway :)

Sturwandan, 2020-05-12:
You just need to execute one command. It is much easier than to install a different OS.

yosoyfreeman, 2020-05-12:
> You just need to execute one command. It is much easier than to install a different OS.

Im going to take a look on it. Its a bit scary for a newbiew you know. Thank you very much for your kindness and sorry for the newbie questions!

TSGut, 2020-05-15:
See my reply here for Pop OS 20.04: https://github.com/DIGImend/digimend-kernel-drivers/issues/387#issuecomment-629462913

yosoyfreeman, 2020-05-15:
> See my reply here for Pop OS 20.04: [#387 (comment)](https://github.com/DIGImend/digimend-kernel-drivers/issues/387#issuecomment-629462913)

Hi, this is a bit difficult for me, im just an artist with low experience in linux and never compiled anything. But im sure that your reply will be so useful for a bit more experienced users.  Thanks a lot for all your replys mates!

TSGut, 2020-05-15:
I understand. Unfortunately for now updating the kernel was the only way that I could find to fix this in Ubuntu 20.04. If nothing else, this should make you feel confident that this will be automatically fixed when kernels are updated in newer upcoming Ubuntu releases (e.g. when Ubuntu 20.10 comes around).

yosoyfreeman, 2020-05-15:
> I understand. Unfortunately for now updating the kernel was the only way that I could find to fix this in Ubuntu 20.04. If nothing else, this should make you feel confident that this will be [...]

Oh, that exactly what i was going to ask for. I just going to be patient i think. Thank you again! (This is my first github issue, should i mark this as closed?)

TSGut, 2020-05-15:
I'm not a maintainer on this project so it's not for me to say but since it's effectively a mirror of the tagged issue at https://github.com/DIGImend/digimend-kernel-drivers/issues/387 it should be fine to close it, I think.

closed 2020-05-15 by yosoyfreeman