DIGImend/digimend-kernel-drivers #391: Add support for G640S

DarkWarrior703, 2020-05-13:
If G640 is working, how G640S isn't?

Sturwandan, 2020-05-15:
XP-pen has completely different hardware variations even between same model number (G430S white and black). See my bug report (#388 ) for example.
If you wish for support for G640S to be added, you have to provide full diagnostic information, similar to my report.

Franzcoz, 2021-12-18:
Hello! I own an XP-Pen Star G640S and after a lot of time trying to find a suitable sollution to use it on my laptopt and not finding it, I decided to help make it, since I don't know how to write a linux driver.
I have the [diagnostic](https://github.com/DIGImend/digimend-kernel-drivers/files/7738394/diagnostics_digimend.zip) information asked on the digimend website, and also I would like to help writing IF i get enough help :).
But I guess just the diagnostics will do for now
[diagnostics_digimend.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/7738394/diagnostics_digimend.zip)

Franzcoz, 2022-02-24:
Hi again! Trying to help a little, but with almost zero knowledge, I modified some files to incorporate this model and its descriptor. I do not know if what I did is correct, I tried to follow what was done with other models in past commits. And also I wouldn't know how to do a merge request so I am letting what I did here if someone can tell I am heading in the right direction or not. Hope it helps in anything. Also, my descriptors turned out different from those in the repository that holds tablet diagnostics, Idk why.

[xp-pen-g640s.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/8136620/xp-pen-g640s.zip)

Franzcoz, 2022-02-26:
Following recommendations from people in the IRC channel (thanks!) I have run the [uclogic-new-init](https://github.com/nic3-14159/uclogic-new-init) script and then again the rest of the [diagnostics](https://github.com/DIGImend/digimend-kernel-drivers/files/8145810/xp-pen-g640s-diagnostics.zip). I divided the hid-report-descriptors file into 3 parts, and its corresponding 3 code translations.

Franzcoz, 2022-02-27:
I have run the script again, but this time ommiting the step involving uclogic-probe before collecting samples. This time the reports start with 02, as was mentioned in the chat that was expected to. Also, worth to mention that this time, if I run uclogic-probe and then try to collect samples, it says "No matching HID device", which didn't happen before. Maybe it has to do with some config files I found in /etc for xorg and udev, from previous attempts to make it work, and that I removed before preoceeding with this diagnostic attempt.

[second_new_diag.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/8149304/second_new_diag.zip)

Franzcoz, 2022-03-03:
Report descriptors translated as code with the hidrd-convert tool. Second attempt obtaining reports this time after executing the uclogic-new-init script.

[descriptors_as_code.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/8180500/descriptors_as_code.zip)

nic3-14159, 2022-03-03:
@Franzcoz Could you collect and post the raw input samples for this tablet after running the uclogic-new-init script? Those are the things that should change after running the script. They are the ones you dump using `usbhid-dump -es -m 28bd:0906` while moving the pen around and pressing buttons. The descriptors that you posted shouldn't change, whether or not the script was run.

nic3-14159, 2022-03-03:
> I think those are the ones I posted here: [#391 (comment)](https://github.com/DIGImend/digimend-kernel-drivers/issues/391#issuecomment-1053655579) ?

Yeah, sorry about that, I didn't notice those

ezequielv, 2022-04-11:
Hello. I'm considering buying this device (on offer on [amazon.co.uk](https://www.amazon.co.uk/Graphics-Digital-Pressure-Shortcut-Signature/dp/B07GGN6YY4/) today), but it seems that digimend's kernel driver support is not quite there yet for this model.

I already own an XP-PEN Deco 01 V2, and it seems to be supported perfectly by digimend's kernel drivers, so I'm not sure it'd be worth to buy this cheaper one as a second tablet for my kids, since digimend kernel driver support is now a must in my household :)

I've checked the latest "list" posted [here](https://github.com/DIGImend/digimend-kernel-drivers/issues/424), which only seems to include a "versionless" G640S: `PID_0926=G640S`. 

And I've also seen [this comment (#253)](https://github.com/DIGImend/digimend-kernel-drivers/issues/253#issuecomment-663935969), in which it seems that the tablet (V1?) might be supported by the `libinput` driver.

Are there any news regarding support for the XP-Pen G640S V2? Or are users supposed to be using the `libinput` driver instead?

nic3-14159, 2022-04-18:
> Hello. I'm considering buying this device (on offer on [amazon.co.uk](https://www.amazon.co.uk/Graphics-Digital-Pressure-Shortcut-Signature/dp/B07GGN6YY4/) today), but it seems that digimend's [...]

At the moment the G640S V2 is not supported, but I think all the information needed to support it has been posted already. I've been meaning to take a look at it and write some code, but I haven't found time/motivation to do it. The maintainer for DIGImend hasn't touched the project in a while, so there's quite a few unmerged pull requests and I don't know if and/or when that will change. Some of those requests contain code that would be needed to support the G640S, so it would be nice for those to get merged to avoid having to duplicate code or maintain a fork with all the changes.

The number of new requests seems to be relatively constant, so there is still community interest in DIGImend, but there aren't nearly enough people working on it to keep up with the demand.

> And I've also seen [this comment (#253)](https://github.com/DIGImend/digimend-kernel-drivers/issues/253#issuecomment-663935969), in which it seems that the tablet (V1?) might be supported by the [...]
The libinput driver is not the same sort of driver as DIGImend. DIGImend is a kernel driver, and takes in the raw USB HID events from the tablet and makes it so that the generic HID kernel driver properly recognizes the events. That HID driver then outputs generic input events that can be used by userspace. Drivers like libinput or xf86-input-wacom are drivers for Xorg/Wayland, and take in those generic input events and send them to the applications.

So supporting a tablet in Linux involves making sure it works with the kernel drivers, as well as the Xorg/Wayland input drivers.

Franzcoz, 2022-07-28:
Oh, ok, understandable. Is there something one could do to help?